B-Lang-org/bsc #20: `/build/source/inst/bin/bsc -stdlib-names -bdir /build/source/build/bsvlib -p . -vsearch /build/source/build/bsvlib -no-use-prelude PreludeBSV.bsv`

NinjaTrappeur, 2020-02-06:


flokli, 2020-02-06:
FYI, I managed to get past that failure with GCC 7.5.0 and GHC 8.0.2 and pushed my current WIP to my [bluespec-bsc nixpkgs branch](https://github.com/flokli/nixpkgs/tree/bluespec-bsc).

Note GHC 8.0.2 is way out of support, see https://github.com/B-Lang-org/bsc/issues/16#issuecomment-582665265

Whoever is interested, I joined the `#bluespec` channel on freenode.

flokli, 2020-02-08:
I'm able to trigger this with GHC 8.6.5 as well, with 8.4.4 it seemed to still work.

dramforever, 2020-02-09:
**This is untested** but I think I have found the issue. The culprit is here:

https://github.com/B-Lang-org/bsc/blob/a82d720f3fe0d0d634fed5d14a0a7a628a6904a6/src/Parsec/ParsecPrim.hs#L239-L249

Which seems to be have been written intending to take care of the transitioning regarding `MonadFail`. However, `fail` of the `Monad` class is *still* (basically) `error` ([see source](http://hackage.haskell.org/package/base-4.12.0.0/docs/src/GHC.Base.html#fail)); it's just discarded. Which means that given GHC `>= 8.6`, a call to `fail` will not trigger a backtrack and instead will crash the whole program with the message, which seems to be happening here:

https://github.com/B-Lang-org/bsc/blob/03af17e62dc8de34cd15b5b89a10497422aeb466/src/comp/Parser/BSV/CVParser.lhs#L2815

---

As of how to fix this, I'm honestly not sure. `grep`ping for the `#if`, I see three Monads affected by this:

- `src/Parsec/ParsecPrim.hs:242`
- `src/comp/ErrorMonad.hs:28`
- `src/comp/SEMonad.hs:23`

The [migration guide](https://prime.haskell.org/wiki/Libraries/Proposals/MonadFail#Adaptingoldcode) says the `fail` in `Monad` should just be `Control.Monad.Fail.fail` which should work until maybe a 'future release' where `fail` in `Monad` is removed (mentioned [here](http://hackage.haskell.org/package/base-4.12.0.0/docs/Control-Monad.html#v:fail))

My take is that we probably just write `fail = Control.Monad.Fail.fail` until it breaks, if ever, which is going to be a compile-time error so it's not hard to catch.

quark17, 2020-02-09:
Ah, thank you!  This makes sense.  I added these conditionals recently, and tested that it compiled on various 8.x versions, but I didn't have working installations so I hadn't tested that it ran properly.  (Both Debian10 and MacPorts on Catalina have issues with GHC that I hadn't gotten past.)  I've installed GHC 8.6.5 on Ubuntu 16.04 (xenial) and confirmed that the repo builds and checks, so I'll commit these changes.  Thanks!

commit 7b3538c9 (2020-02-09): Fix Monad instances affected by MonadFail, fix #20

Extend the #if around fail in Monad to include 8.6 (< 806 -> < 808).
This way, everyone is happy:

- For 8.6 or earlier, Prelude.fail is Monad.fail, which goes to the
  custom function (8.6 would get errorWithoutStackTrace otherwise)
- For 8.8 or later, Prelude.fail is MonadFail.fail, which also goes to
  the custom function.
- Also, there is no fail in Monad in 8.8+, so the #if removes it.
commit c9bfdce5 (2020-02-09): Merge pull request #42 from dramforever/dramforever-monadfail

Fix Monad instances affected by MonadFail, fix #20
closed 2020-02-09 by quark17