B-Lang-org/bsc #22: Improve `src/comp` build time

NinjaTrappeur, 2020-02-06:
Building `src/comp` takes about 20 minutes on my beefy machine.

This long build time is mostly due to the fact `GHC` builds the whole 229 targets in a single core, hence not leveraging parallel nature of modern CPUs.

One way to improve this situation would be to divide this massive 229 targets package into several smaller independent libs, allowing a potential Haskell-specific build tool (cabal? stack?) to parallelize a bit more the overall build.

The solution to this will mostly depend on the build system you're (ie. ppl working full-time on this) envisioning. 

Do you plan to keep this makefile + bare ghc custom build system or move to a Haskell-specific build tool for the Haskell bits?

christiaanb, 2020-02-06:
`ghc` takes a `-j` flag for building modules on parallel threads, e.g. `-j4` to build on 4 threads. No need for Haskell-specific build tools if you want to explore compile parallelism.

NinjaTrappeur, 2020-02-06:
@christiaanb The GHC build paralelization won't necessarily lead to a linear build time improvement, see https://gitlab.haskell.org/ghc/ghc/issues/9221

bpfoley, 2020-02-06:
@NinjaTrappeur, it's true that ghc --make -jN won't lead to linear speedups, but in practice it does make comp build substantially faster. I tried -j6 (because I've a 6 core machine) and at times it did sustain 600% CPU usage.

bpfoley, 2020-04-02:
Another note on this: GHC 8.10.1 has some substantial compile-time performance improvements in pattern matching exhaustiveness checking, and this speeds up the compilation time of a few of our compiler modules quite a bit.