B-Lang-org/bsc #23: make aborts with error

mxsrv, 2020-02-06:
Every time I start the make file, the program aborts after about 10 minutes and I don't understand why. Here's the extract from my terminal: 
```
...
[220 of 229] Compiling LambdaCalcUtil   ( LambdaCalcUtil.hs, /var/tmp/pamac-build-max/bluespec-git/src/bsc/src/comp/../../build/LambdaCalcUtil.o )
[221 of 229] Compiling SAL              ( SAL.hs, /var/tmp/pamac-build-max/bluespec-git/src/bsc/src/comp/../../build/SAL.o )
[222 of 229] Compiling LambdaCalc       ( LambdaCalc.hs, /var/tmp/pamac-build-max/bluespec-git/src/bsc/src/comp/../../build/LambdaCalc.o )
[223 of 229] Compiling ADumpSchedule    ( ADumpSchedule.hs, /var/tmp/pamac-build-max/bluespec-git/src/bsc/src/comp/../../build/ADumpSchedule.o )
[224 of 229] Compiling DisjointTest     ( DisjointTest.hs, /var/tmp/pamac-build-max/bluespec-git/src/bsc/src/comp/../../build/DisjointTest.o )
[225 of 229] Compiling ASchedule        ( ASchedule.hs, /var/tmp/pamac-build-max/bluespec-git/src/bsc/src/comp/../../build/ASchedule.o )
[226 of 229] Compiling AState           ( AState.hs, /var/tmp/pamac-build-max/bluespec-git/src/bsc/src/comp/../../build/AState.o )
[227 of 229] Compiling AAddSchedAssumps ( AAddSchedAssumps.hs, /var/tmp/pamac-build-max/bluespec-git/src/bsc/src/comp/../../build/AAddSchedAssumps.o )
[228 of 229] Compiling ACleanup         ( ACleanup.hs, /var/tmp/pamac-build-max/bluespec-git/src/bsc/src/comp/../../build/ACleanup.o )
[229 of 229] Compiling Main_bsc         ( bsc.hs, /var/tmp/pamac-build-max/bluespec-git/src/bsc/src/comp/../../build/Main_bsc.o )
Linking bsc ...
Using flags:  @/tmp/ghc5165_0/ghc_839.rsp
Using flags:  @/tmp/ghc5165_0/ghc_842.rsp
bsc done Do 6. Feb 14:03:16 CET 2020

make[2]: Verzeichnis „/var/tmp/pamac-build-max/bluespec-git/src/bsc/src/comp“ wird verlassen
make[1]: *** [Makefile:15: install] Fehler 2
make[1]: Verzeichnis „/var/tmp/pamac-build-max/bluespec-git/src/bsc/src“ wird verlassen
make: *** [Makefile:24: install] Fehler 2
==> FEHLER: Ein Fehler geschah in build().
    Breche ab...

...
```

The german text says:  "... ==> ERROR: an error occured in build(). Aborting... "

jameyhicks, 2020-02-06:
What OS are you building on? Which version?

mxsrv, 2020-02-07:
Manjaro Linux 18.1.5

flokli, 2020-02-08:
```
[221 of 224] Compiling AState           ( AState.hs, /build/source/src/comp/../../build/AState.o )
[222 of 224] Compiling AAddSchedAssumps ( AAddSchedAssumps.hs, /build/source/src/comp/../../build/AAddSchedAssumps.o )
[223 of 224] Compiling ACleanup         ( ACleanup.hs, /build/source/src/comp/../../build/ACleanup.o )
[224 of 224] Compiling Main_bsc         ( bsc.hs, /build/source/src/comp/../../build/Main_bsc.o )
Linking bsc ...
Using flags:  @/tmp/ghc26512_0/ghc_818.rsp
Using flags:  @/tmp/ghc26512_0/ghc_821.rsp
bsc done Sat Feb 8 10:44:41 UTC 2020

make[2]: Leaving directory '/build/source/src/comp'
make[1]: *** [Makefile:13: install] Error 2
make[1]: Leaving directory '/build/source/src'
make: *** [Makefile:24: install] Error 2
builder for '/nix/store/krbznhgazzqjmjb57hgg8zbk3npdm3n3-bluespec-unstable-2020.02.07.drv' failed with exit code 2
```

I had the same output once. In my case, triggering the build again in the same environment did work.
It was built in a sandbox, meaning the next run had the same conditions, without any artifacts from the previous run around.

Until we know better, I'd suspect a parallel build problem - cc @thoughtpolice https://github.com/NixOS/nixpkgs/pull/79468#discussion_r376703525

quark17, 2020-02-09:
A fix (#40) for parallel build conflicts (#38) was merged.  Does the issue still occur with the latest?

mxsrv, 2020-02-10:
No, thanks a lot. The latest version worked fine.

flokli, 2020-02-10:
In that case, can you close the issue?

closed 2020-02-11 by mxsrv