B-Lang-org/bsc #29: There is no test suite

thoughtpolice, 2020-02-07:
From what I can tell, at the moment there is no test suite for the compiler -- which makes me extremely anxious to make any changes. For example, #16 suggests that STP can be removed, except for one niggling case in `SATPred.hs`, which could be replaced with Yices. The thing is, how do I even test that change? Why was STP used in the first place? Are there cases it succeeds where Yices fails?

Perhaps for now the right thing to do is test changes by doing full compilation flows e.g. against Piccolo or Flute, and ensuring the end-to-end simulation runs turn out the same, or something.

In the long run, I think there would be a significant amount of value to be had by simply adding tests that can be run using some "golden testing" infrastructure, that simply test typechecking/compiling/etc all work as expected, on very small and minimal examples. (Most compilers do something like this, of course.)

quark17, 2020-02-07:
Yes, as the README says, we hope to add a test suite.  One exists, but it is not yet public.  If you submit a PR, I will run the changes on the private test suite -- though I recognize that it would help to have regression tests prior to submitting the PR, so hopefully we can have something soon.

thoughtpolice, 2020-02-07:
Thanks, sorry I missed that! That's great to know. Right now I have some changes queued up for doing builds with Nix, possibly some CI with GitHub actions, etc. Nothing material, we're just trying to get all the dependencies/build requirements sorted out reliably. I just wanted to try and fix our Nix builds to work with a more modern GHC -- since only 8.0.x seems to work, which we long since deprecated -- and was wondering how to test anything.

I don't know what y'alls plans are yet for some of the stuff like CI, but I can just submit some pulls and hash it out I suppose...

thoughtpolice, 2020-02-07:
I'll also write a patch to at least run the smoke test under `make check` for now (and hopefully any future public test suite can be under that target name!) Many packaging systems use this as an implicit default target for doing integration tests (including ours), so making the default test do something keeps things simpler for everyone.

bpfoley, 2020-02-07:
For what it's worth, it definitely builds (and runs the smoke test) with GHC 8.4.4 for me, so you can upgrade at least that far. I haven't tried anything newer yet.

thoughtpolice, 2020-02-07:
Thanks, I have a working Nix build using 8.0.2, so I'll try an upgrade.

quark17, 2020-02-07:
I've compiled with 8.8.2, and added in the necessary CPP macros to account for library versions and feature changes, so I think it should compile with later versions (I had a bad installation that wouldn't link, so I didn't run it, but it compiles).  I think I also did the same with 8.6.5.  And certainly with 8.4.4.

thoughtpolice, 2020-02-07:
I've successfully built everything with 8.4.4, and removed STP support as well (which I suppose did fix things, I suppose!) So I think everything is in pretty good shape.

thoughtpolice, 2020-02-07:
(Also: the patch for putting the smoke test inside `make check` is defined in #30, as well.)

quark17, 2020-10-14:
Oh, I forgot to update this issue!  We do now have an open testsuite, at [bsc-testsuite](https://github.com/B-Lang-org/bsc-testsuite)

closed 2020-10-14 by quark17