B-Lang-org/bsc #38: sporadic compilation failures with parallel builds enabled

flokli, 2020-02-08:
For time to time, building `bsc` with a huge number of jobs `-j48` fails.

I managed to get a log output for one of these failures:
[bluespec-bsc-parallel-build-failure.txt](https://github.com/B-Lang-org/bsc/files/4175175/bluespec-bsc-parallel-build-failure.txt)

The interesting lines:

```
[  4 of 149] Compiling EitherUtil       ( EitherUtil.hs, /build/source/src/comp/../../build/EitherUtil.o )
[  4 of 149] Compiling EitherUtil       ( EitherUtil.hs, /build/source/src/comp/../../build/EitherUtil.o )
[  5 of 149] Compiling ErrorTCompat     ( ErrorTCompat.hs, /build/source/src/comp/../../build/ErrorTCompat.o )
[  5 of 149] Compiling ErrorTCompat     ( ErrorTCompat.hs, /build/source/src/comp/../../build/ErrorTCompat.o )
[  1 of 224] Compiling BDD              ( BDD.hs, /build/source/src/comp/../../build/BDD.o )
/build/source/src/comp/../../build/ErrorTCompat.o: getModificationTime:getFileStatus: does not exist (No such file or directory)
[  6 of 149] Compiling Eval             ( Eval.hs, /build/source/src/comp/../../build/Eval.o )
make[2]: *** [Makefile:403: bluetcl] Error 1
make[2]: *** Waiting for unfinished jobs....
```

It seems there's at least two builds ghc builds going on in parallel (one with 149 targets, and one with 224 targets), and they both manage to stumble over each others feets trying to read/write to/from `ErrorTCompat`.

We should probably run only one of these builds at once, have intermediate build artifacts in separate scratch dirs, or further split things into libraries (as suggested by @NinjaTrappeur in https://github.com/B-Lang-org/bsc/issues/22)

bpfoley, 2020-02-08:
What I've found that works for me is to make src/comp/Makefile itself non-parallel (by replacing the `$(MAKE) -C comp` in src/Makefile with `make -C comp`) and then just adding `-j$(ncores)`   to the GHC options in src/comp/Makefile.

flokli, 2020-02-08:
That's probably a good measure for now, until the Haskell code can get a bit restructured and split. Would you mind filing a PR for that, with some comments in the makefile on why we do that?

flokli, 2020-02-08:
I opened #40.

PR #40 (2020-02-09): don't build targets in `comp` in parallel
Otherwise, src/comp/Makefile invokes multiple ghc builds, all in the
same workspace, which break each other sometimes.

We still get some parallelization, by passing -jN to GHC itself, but
reading what number of jobs is passed to make, and passing it down to
GHC doesn't really work
[without a lot of hackery](https://stackoverflow.com/questions/5303553),
and there seems to be a
[upper limit](https://gitlab.haskell.org/ghc/ghc/issues/9221) as well,
so let's pick a "reasonable number", and allow overriding it via
GHCJOBS=n, if people run it on a very beef [... cut]
commit ca2cc7bf (2020-02-09): don't build targets in `comp` in parallel

Otherwise, src/comp/Makefile invokes multiple ghc builds, all in the
same workspace, which break each other sometimes.

One way of still getting some parallelization there would be to pass -jN
to GHC itself.

Reading what number of jobs is passed to make, and passing it down to
GHC doesn't really work
[without a lot of hackery](https://stackoverflow.com/questions/5303553),
and there seems to be a
[upper limit](https://gitlab.haskell.org/ghc/ghc/issues/9221) as well,
so I decided to introduce a GHCJOBS=n variable.

However, compilation seems to be some [... cut]
closed 2020-02-09 by quark17