B-Lang-org/bsc #47: bsc accepts, ignores -o option with -verilog

cbiffle, 2020-02-11:
Steps to repro:

`bsc -verilog -o Top.v Top.bs`

Expected result: output goes into `Top.v`

Expected result, slightly darker timeline: `-o` flag is rejected as incompatible with Verilog output, compile fails.

Actual result: `-o` flag silently ignored, output goes into `mkTop.v` (as that is the name of the module in `Top.bs`).

quark17, 2020-02-11:
I apologize that the user guide document is not yet available.

The `-o` flag is for naming the simulation executable generated during a linking call to BSC (`bsc -e <topmod>`).  It would be reasonable to have BSC warn if you specify `-o` and are not linking into an executable.

Most tools that take Verilog files as input expect that a module M will be defined in a file called `M.v`, and BSC respects this convention by generated one file per module and named for the module.  A single BSC package (like `Top.bsv`) can define many modules and therefore generate many Verilog outputs.  A single `-o` would not be sufficient.  In theory, a pragma in the source code could be used to rename the generated Verilog (or maybe a more complicated flag) but if someone really wants defy this convention, it's probably best that they rename the outputs themselves.

cbiffle, 2020-02-11:
Rejecting the unused `-o` flag as incompatible with `-verilog` would be fine by me.