B-Lang-org/bsc #48: Multiple -p options to bsc clobber one another

cbiffle, 2020-02-11:
Steps to repro:
`bsc -p dir1 -p dir2 Top.bs`

Expected result, ideal world: search path is composed of `dir1:dir2` (by analogy to how `-L` / `-I` work on gcc, for example). (Note: this option would substantially simplify build system integration.)

Expected result, less ideal world: compiler notes that `-p` was given twice and that was almost certainly a mistake, compile fails.

Actual result: final `-p` on the command line silently erases all knowledge of previous occurrences.

quark17, 2020-02-11:
This is the intended behavior.  The -p flag can be used to override previous values on the command-line, allowing one to set defaults in an alias, say.  To add additional directories to an existing path, you can use `+`:

`bsc  -p +:dir1  -p +:dir2  -p dir3:+`

The path already starts pre-populated with some default library locations, so using `+` is generally necessary.  If you want to redefine the path from scratch, you can use `%` to refer to the BSC installation directory (BLUESPECDIR) under which the libraries are installed.

Again, I apologize that the user guide document is not yet available.  Some of this described at the end of the `-help` message:

`` Flags later on the command line override earlier ones.``
`` Path strings such as the import path may contain the character``
`` `%' representing the current Bluespec directory, as well as``
`` `+' representing the current value of the path.``

The default import path (the initial value of `+` for `-p`) is given below that.

cbiffle, 2020-02-11:
Fair enough, treating this as working-as-intended.

closed 2020-02-11 by cbiffle