B-Lang-org/bsc #49: Concurrent bsc invocations using same build dir interfere with one another

cbiffle, 2020-02-11:
**Steps to repro:**

Obtain a machine with 2+ CPU cores, just to be safe.

Check out the tutorials, specifically `Tutorials/Bluespec_Classic_Training/Example/Eg06a_Mergesort` -- the mergesort isn't special, it's just the one I'm hacking on.

Place the following `build.ninja` file in the tutorial directory:

```ninja
bsc_flags = -vdir verilog_RTL_ -bdir build_ -info-dir build_ -keep-fires -aggressive-conditions -no-warn-action-shadowing -check-assert -cpp +RTS -K128M -RTS -show-range-con
flict -p src:../Resources:%/Libraries

rule bsc_o
  command = bsc $bsc_flags -o $out $in

rule bsc_v
  command = bsc $bsc_flags -verilog $in

build build_/Req_Rsp.bo: bsc_o ../Resources/Req_Rsp.bs
build build_/Fabric_Req_Rsp.bo: bsc_o ../Resources/Fabric_Req_Rsp.bs | build_/Req_Rsp.bo build_/Fabric_Defs.bo

build build_/Fabric_Defs.bo: bsc_o src/Fabric_Defs.bs
```

Run: `ninja build_/Fabric_Req_Rsp.bo`

Now run: `ninja -t clean && ninja -j1 build_/Fabric_Req_Rsp.bo`

**Expected results:** the parallel build produces results equivalent to the serial build.

**Actual results:**

On the initial, parallel build:

```
[1/3] bsc -vdir verilog_RTL_ -bdir buil...uild_/Fabric_Defs.bo src/Fabric_Defs.b
FAILED: build_/Fabric_Defs.bo 
bsc -vdir verilog_RTL_ -bdir build_ -info-dir build_ -keep-fires -aggressive-conditions -no-warn-action-shadowing -check-assert -cpp +RTS -K128M -RTS -show-range-conflict -p src:../Resources:%/Libraries -o build_/Fabric_Defs.bo src/Fabric_Defs.bs
Error: Unknown position: (S0084)
  Could not remove the file `_t_o_p.c':
    does not exist
[2/3] bsc -vdir verilog_RTL_ -bdir buil...ild_/Req_Rsp.bo ../Resources/Req_Rsp.b
ninja: build stopped: subcommand failed.
```

Using `-j1`: works fine.

Is it possible that every bsc invocation is assuming it can generate a file called `_t_o_p.c` in the build directory, regardless of source/destination filenames? This makes it difficult to drive bsc from high-concurrency build systems. (I can arrange for every bsc invocation to get a *separate* build directory, but that's kind of messy.)

quark17, 2020-02-12:
I see that you've submitted a PR for this. Thank you, I'll review it.  BSC should definitely be generating a unique name.

For years, the only supported syntax was BSV, because the target users were hardware designers that were familiar with Verilog and SystemVerilog, not Haskell.  The Haskell-based "Classic" syntax (which we're now calling BH) was maintained, but new features weren't added to it and nearly no one wrote designs in it.  The BSV syntax natively supports SystemVerilog preprocessor directives (it preserves position information etc) and so I don't think anyone ever used CPP, instead preferring the SV PP.  The Classic/BH syntax obviously doesn't have SV preprocessor support, so you have to use CPP.  Recently, there's been interest in using the Classic/BH syntax, and those Classic/BH tutorial examples were added to help those folks.  I assume that as more people start using the BH front-end, we'll see requests for fixes like this and for features (to make the BH front-end catch up with the BSV front-end).

cbiffle, 2020-02-12:
That's a great point that this affects only the "classic" frontend. I didn't realize that (as I've been working with classic).

I *think* I've seen this bug manifest without `-cpp` on the command line, but I'm going to have to check my notes carefully before I assert that more forcefully.

quark17, 2020-02-15:
The `_t_o_p.c` name should be fixed with commit e5af869.  Do any other interferences occur?

bpfoley, 2020-04-02:
@cbiffle Gentle poke... is this working for you now?

quark17, 2020-04-03:
For reference: Above, I mentioned that I'd fixed it with commit e5af869, but I reverted that (with commit e0aab355) and accepted @cbiffle 's original solution for this (PR #52) with merge commit 8bcfc181.

closed 2020-04-13 by quark17