B-Lang-org/bsc #50: Unvendor yices2

bpfoley, 2020-02-11:
...eventually, at a time where it's suitable for us.

Yices have
* formulas suitable for use with Homebrew on MacOS X,
* PPA packages suitable for use on Debian/Ubuntu
* Binary releases statically linked against the necessary libraries for other x86 Linuxes.

https://yices.csl.sri.com/doc/install-binaries.html

Un-vendoring means we won't have to maintain yices's dependencies, and probably makes bsc more suitable for packaging on Debian/Homebrew. We may have to do some legwork to get yices made into non-third party packages on Debian/Homebrew.

bpfoley, 2020-02-11:
NB also that yices2 is GPLv3.

cbiffle, 2020-02-11:
Just noticed that license myself -- any sense for how long Bluespec has been in violation of the GPL? I assume this component wasn't added in the open source release.

quark17, 2020-02-11:
As part of making this BSC repo, to put on GitHub, I extracted code from various places and created Makefiles and cleaned things up, etc.  I also decided to update the use of Yices, which predated Yices being open source (and GPLv3); so this was in fact just added, for this release.  (I had also wanted to update STP, but didn't get around to it.)  And, yes, it does mean that any distribution of BSC from this repo needs to respect the GPLv3 license or excise the use of Yices.

cbiffle, 2020-02-11:
Thanks for the clarification!

thoughtpolice, 2020-02-13:
Note that it's unclear if a newer version of Yices is needed or not for Bluespec -- without the test suite or real users of the FOSS version reporting in just yet, it's hard to know what behavioral differences there may be between now and then. They may or may not need to do a release for us to reliably unvendor things.

That said I'm hopeful if we needed it, and someone asked them nicely, they would cut a new release. :)