B-Lang-org/bsc #51: Unvendor stp

bpfoley, 2020-02-11:
...and make it an optional dependency of Bluespec.

NB stp's build system is based on CMake, and currently there aren't packages for Debian/Ubuntu or Homebrew for Mac OS X. I'm working on the latter at the moment.

bpfoley, 2020-02-11:
STP homebrew Formula: https://github.com/Homebrew/homebrew-core/pull/50020

cbiffle, 2020-02-11:
Note that STP *also* appears to be GPL as a result of its Bit::Vector dependency. Unless I've missed something.

quark17, 2020-02-11:
I have not looked to see if the license situation has changed in the new STP available on GitHub, but for the snapshot here, the Bit::Vector library uses LGPL (as indicated in `extlib-constbv/constantbv.h` and reproduced in `LICENSES/LICENSE.stp_components`).  The BSC binaries dynamically link with the library built from the STP snapshot.

quark17, 2020-02-11:
PR #31 is related to this and has discussion.

cbiffle, 2020-02-11:
You are probably correct, I may have misread the LGPL as the GPL.