B-Lang-org/bsc #68: build command not compiled.

rajat-agarwal-, 2020-02-18:
While doing labs from 6.375 course(MIT), there was a mention of the build command as a part of the bluespec package, which i didn't find in the bin folder. Can anyone please help?

quark17, 2020-02-18:
`build` is not part of this repository.  It was a script and its primary purpose was for designs that used Bluespec's host-to-FPGA communication mechanisms (BlueNoC and SceMi via BlueNoC).  That mechanism was built using BSC: Bluespec provided BSV files and pre-generated IP for the FPGA side and C/C++ files for the host side, and the `build` script knew where to find those specific files and compile them and synthesize an FPGA bitfile.  None of those files have been included in this repo and so the `build` script hasn't been included either.

My understanding is that the 6.375 labs don't use Bluespec's communication mechanism but instead use the Connectal mechanism (which has its own FPGA-side and host-side files).  This shouldn't require the `build` script.

rajat-agarwal-, 2020-02-19:
Yes, you are correct. The recent version of the course uses Connectal as the interface mechanism. However the previous offerings did use Bluespec's interface mechanism. 
Thanks anyway.

closed 2020-02-19 by rajat-agarwal-