B-Lang-org/bsc #81: Building bsc on MacOS works!

mtikekar, 2020-02-28:
After tcl/tk was removed, I am able to build bsc on MacOS (10.13) with Apple's compiler (Apple LLVM 10.0.0). There were a few hiccups which needed manual intervention and I could not build STP, bluesim and bluetcl. These are my steps:

1. Get cabal-install, tcl-tk, autoconf, icarus-verilog from Homebrew
2. `cabal v1-install` the three packages instead of `cabal install`
3. tcl-tk is at `/usr/local/opt/tcl-tk` so edit PATH and PKG_CONFIG_PATH. comp/Makefile hardcodes the paths and version number of itcl. I just manually changed them. Also, itk is not available so remove bluetcl and bluewish from build targets
4. `make STP_STUB=1 -j all`

At this point bsc is built but it cannot build Libraries as it seems to need libbdpi. To see the libs it needs, run `otool -L inst/bin/core/bsc` which shows:

```
inst/bin/core/bsc:
        /usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1252.50.4)
        libbdpi.so (compatibility version 0.0.0, current version 0.0.0)
        /Users/mtikekar/src/bsc/src/yices/v2.6/yices2-inst/lib/libyices.2.dylib (compatibility version 2.6.0, current version 2.6.2)
        /usr/lib/libiconv.2.dylib (compatibility version 7.0.0, current version 7.0.0)
```

(absolute path to yices should probably be fixed)

So I built VPI manually and pointed bsc to libbdpi.so with

```bash
install_name_tool -change libbdpi.so $(pwd)/src/VPI/libbdpi.so inst/bin/core/bsc
```

Then manually built the remaining steps. (Some of my commands above are reconstructed from memory and might be different).

mtikekar, 2020-02-28:
Btw, `make check` runs until it needs bluetcl

bpfoley, 2020-02-28:
That's great news! I haven't been actively working on a macOS build, but I'd certainly like to have one. Thanks for exploring this. At least we know we're moving in the right direction.

thoughtpolice, 2020-02-29:
The `libbdpi.so` issue is due to a bug in the build system when using the STP no-op stub. I have a fix for it.

I think having macOS (and eventually: windows!) builds is a noble goal. It's probably worth having `bluetcl` in working order. Are you using tcl from something like homebrew? I'm not sure the best way to install tcl/tk/itcl/itk is, but that's the most important step at this point...

mtikekar, 2020-02-29:
Yes, I am using the tcl-tk package from Homebrew. It includes itcl but not
itk. I tried compiling itk from source but failed at ./configure.

bpfoley, 2020-03-01:
I'll try packaging itk/itcl for homebrew.

mtikekar, 2020-03-02:
Turns out that MacOS's builtin tcl already has itk/itcl at /System/Library/Tcl/8.5. Perhaps it is best to try to use them instead of the homebrew versions.

ncihnegn, 2020-03-02:
I completely finished ```make STP_STUB=1```.
When ```make check```, run into this error:
```
Checking bsc compiles
bsc -sim FibOne.bsv
Elaborated module file created: mkFibOne.ba

Checking compile for Bluespec simulator
bsc -sim -o mkFibOne.bexe -e mkFibOne mkFibOne.ba
Bluesim object created: mkFibOne.{h,o}
Bluesim object created: model_mkFibOne.{h,o}
Simulation shared library created: mkFibOne.bexe.so
Simulation executable created: mkFibOne.bexe
./mkFibOne.bexe > smoke_test_bluesim.out
called Tcl_FindHashEntry on deleted table
called Tcl_FindHashEntry on deleted table
/bin/sh: line 1:  4762 Abort trap: 6           ./mkFibOne.bexe > smoke_test_bluesim.out
make[1]: *** [smoke_test_bluesim] Error 134
make: *** [check] Error 2
```

mtikekar, 2020-03-02:
I removed the stublibs from the Makefile to use the system itcl/itk and it compiled fine. I did run into the same error you see. But when I looked at examples/smoke_test/smoke_test_bluesim.out, the output is all there, so `mkFibOne.bexe` crashed only at the end. The error message seems quite common and people are blaming Tcl 8.5 and recommend trying 8.6 (https://groups.google.com/forum/#!topic/comp.lang.tcl/BKexjImWklo)

ncihnegn, 2020-03-02:
Now I can complete ```make```, with some small patches on Makefiles and **stp**.
It seems the only major issue is that bluetcl will report an error on exit.

bpfoley, 2020-03-03:
Just for reference, the Tcl_Exit bug in question is this one: https://sourceforge.net/p/tcl/bugs/4043/

mtikekar, 2020-03-03:
I got a Github action to build on MacOS 10.15: https://github.com/mtikekar/bsc/tree/mtikekar-macos. You can download the inst directory from https://github.com/mtikekar/bsc/actions/runs/49104659 and try it out. You'll need `brew install tcl-tk` (version 8.6.10) to run.

jameyhicks, 2020-03-04:
Nice progress!

quark17, 2020-03-08:
PR #98 gets BSC (with real STP) to compile on MacOS and adds it to CI.  I'd appreciate if folks could review that PR and add any comments.  I'd like to merge that PR, if there's consensus.

bpfoley, 2020-03-12:
This is working on HEAD now, so I'll close the issue. Thanks everyone!

closed 2020-03-12 by bpfoley