B-Lang-org/bsc #87: after commit 501cf77d4dabd4393a7de202f16f1b5a3537c479 everything is broken

postoroniy, 2020-02-29:
I can't build with success on master and it seems that after commit 501cf77d4dabd4393a7de202f16f1b5a3537c479 everything is broken.
typical error 

Linking bsc ...
/usr/bin/ld: cannot find -litcl3.4
/usr/bin/ld: cannot find -litclstub3.4
collect2: error: ld returned 1 exit status
`compiler-script.sh' failed in phase `Linker'. (Exit code: 1)
Makefile:376: recipe for target 'bsc' failed
make[2]: *** [bsc] Error 1
make[2]: Leaving directory '/home/slava/projects/bluespec/bsc/src/comp'
Makefile:11: recipe for target 'install' failed
make[1]: *** [install] Error 2
make[1]: Leaving directory '/home/slava/projects/bluespec/bsc/src'
Makefile:24: recipe for target 'install' failed
make: *** [install] Error 2

quark17, 2020-02-29:
Before that, the repo contained source for tcl/tk and itcl/itk, which were built and used.  After that commit, these sources were removed and it became a requirement that users install tcl/tk and itcl/itk first.  The failure you're seeing is because itcl is not installed.   What OS are you building on?  If you're on Debian or Ubuntu, you'll need to install the `itcl3-dev` and `itk3-dev` packages (which commit aba212e added to the README file).

postoroniy, 2020-03-01:
installing those does not help
...
bsc start Sun 1 Mar 14:26:57 AEDT 2020
./update-build-version.sh
BuildVersion.hs up-to-date
./update-build-system.sh
BuildSystem.hs up-to-date
ghc -fmax-pmcheck-iterations=6000000 -Wtabs -hidir /home/slava/projects/bluespec/bsc/src/comp/../../build -odir /home/slava/projects/bluespec/bsc/src/comp/../../build -stubdir /home/slava/projects/bluespec/bsc/src/comp/../../build -main-is Main_bsc \
	-O2 -hide-all-packages -fasm -Wall -fno-warn-orphans -fno-warn-name-shadowing -fno-warn-unused-matches -package base -package containers -package array -package mtl -package unix -package regex-compat -package bytestring -package directory -package process -package filepath -package time -package old-time -package old-locale -package split -package syb -package integer-gmp -iGHC -iGHC/posix -iLibs -i../Parsec -i../stp/include_hs -i../yices/include_hs -i../tcltk '-tmpdir /tmp' -L../tcltk -I/usr/include/tcl -I../stp/include -L../stp/lib -I../yices/include -L../yices/lib '-pgma ./compiler-script.sh' '-pgmc ./compiler-script.sh' '-pgml ./compiler-script.sh' -rtsopts --make bsc -j1 +RTS -M3G -A128m -RTS -ltcl8.6 -ltclstub8.6 -ltk8.6 -ltkstub8.6 -litcl3.4 -litclstub3.4 -lhtcl -lstp -lyices  rts_hooks.o
<command line>: cannot satisfy -package split
    (use -v for more information)
Makefile:377: recipe for target 'bsc' failed
make[2]: *** [bsc] Error 1
make[2]: Leaving directory '/home/slava/projects/bluespec/bsc/src/comp'
Makefile:11: recipe for target 'install' failed
make[1]: *** [install] Error 2
make[1]: Leaving directory '/home/slava/projects/bluespec/bsc/src'
Makefile:24: recipe for target 'install' failed
make: *** [install] Error 2

...
git status
On branch master
Your branch is up to date with 'origin/master'.

Untracked files:
  (use "git add <file>..." to include in what will be committed)

	src/tcltk/HaskellTcl/
	src/tcltk/itcl3.4/
	src/tcltk/itk3.4/
	src/tcltk/tcltk8.5.4/
	xaa

nothing added to commit but untracked files present (use "git add" to track)

quark17, 2020-03-01:
There was a commit (d3f81bfc) an hour ago that introduced a new requirement, the `split` library.  If you look at the changes to the README file by that commit, you'll see that you now need this:
`apt-get install libghc-split-dev`

quark17, 2020-03-01:
The untracked `src` files are because those directories were deleted in later commits that you pulled in, but there were generated files in those directories and they are still around.  You can safely delete those directories.

postoroniy, 2020-03-01:
ok,,checking README file now all the time and not just splitdev but mtl libraries pop up there
thank all good for now

closed 2020-03-01 by postoroniy