B-Lang-org/bsc #103: DESTDIR not used in make install and clean removes $PREFIX

jameyhicks, 2020-03-05:
I'm working on making debian packages so I do not have to recompile on all the machines on which I use bsc.

It has been an interesting and interesting process.

To install in the usual place, PREFIX=/usr. This gets exciting at the end of dpkg-buildpackage when it does a `make clean` which does `rm -fr $PREFIX`. I can see why that was in the code before, but I would like to match packager expectations.

Also, all the installs should be relative to $DESTDIR, which is used so that the packaging scripts can install to a staging area and copy the files into the package.

jameyhicks, 2020-03-05:
I meant to add that I plan to submit a PR when I get a bit farther along.

thoughtpolice, 2020-03-11:
I also plan on doing a rewrite of the build system and fixing a lot of these problems. Stay tuned! First, I've got to do a few more cleanups. Any patches are welcome though (and my patches are only making any needed changes, nothing major.)

bpfoley, 2020-03-11:
@thoughtpolice Cool! Looking forward to it.
@jameyhicks Do you plan on submitting the Debian package to Debian/Ubuntu?

jameyhicks, 2020-03-12:
@bpfoley I would be happy to do so but I'm not yet a Debian Maintainer.
@thoughtpolice I'll watch for your patches.

quark17, 2021-01-05:
I agree that we need to fix the makefiles to follow the conventions for PREFIX/DESTDIR etc.  Here are some thoughts that I had in April, but never recorded:

(1) We can change PREFIX to be DESTDIR (as described in the GNU coding standards).  The one thing to be careful about is that Yices references DESTDIR, and we want that to be BSC's build area, not BSC's eventual DESTDIR.

(2) The DESTDIR directory should only be used in install* and uninstall* targets, and we should make sure that we're doing that.  (I suspect that we're not, if we're deleting the PREFIX files in a "clean" or "full_clean" target.)

(3) We can stop creating/deleting the PREFIX (now DESTDIR).  Instead, we'll create "lib/bsc-<version>" (or whatever layout we decide in issue #85 ) and we can add an uninstall target that will delete it.  (We might want the uninstall target to only delete the "bin/" files if they are symlinks that point to the lib directory that we're deleting?  In case another version is installed?)

(4) We'll need to add a VERSION variable to the top-level Makefile.  It'll be predefined, but can be overwritten.  And it'll be passed to src/comp/ where the Makefile there will need to write it into "Version.hs" (avoiding changing the file if the version already matches, so that we're not recompiling unnecessarily).

For reference: https://www.gnu.org/prep/standards/standards.html#Makefile-Conventions