B-Lang-org/bsc #109: Build instructions not working with cabal >= 3

leonschoorl, 2020-03-10:
I was just following along with the build instructions in README.md and I it complained that it couldn't find `regex-compat`.

<details>
  <summary>Build error</summary>

```
[...]
make[2]: Entering directory '/home/leon/devel/bluespec/src/comp'
----- Normal build options -----
can't find package Itcl
bsc start di 10 mrt 2020 17:05:52 CET
./update-build-version.sh
BuildVersion.hs up-to-date
./update-build-system.sh
BuildSystem.hs up-to-date
ghc -fmax-pmcheck-iterations=6000000 -Wtabs -hidir /home/leon/devel/bluespec/src/comp/../../build -odir /home/leon/devel/bluespec/src/comp/../../build -stubdir /home/leon/devel/bluespec/src/comp/../../build -main-is Main_bsc \
	-O2 -hide-all-packages -fasm -Wall -fno-warn-orphans -fno-warn-name-shadowing -fno-warn-unused-matches -package base -package containers -package array -package mtl -package unix -package regex-compat -package bytestring -package directory -package process -package filepath -package time -package old-time -package old-locale -package split -package syb -package integer-gmp -iGHC -iGHC/posix -iLibs -i../Parsec -i../vendor/stp/include_hs -i../vendor/yices/include_hs -i../vendor/htcl '-tmpdir /tmp' -L../vendor/htcl -I/usr/include/tcl -I../vendor/stp/include -L../vendor/stp/lib -I../vendor/yices/include -L../vendor/yices/lib '-pgma ./compiler-script.sh' '-pgmc ./compiler-script.sh' '-pgml ./compiler-script.sh' -rtsopts --make bsc -j1 +RTS -M3G -A128m -RTS -ltcl8.6 -ltk8.6 -litcl -lhtcl -lstp -lyices  rts_hooks.o
<command line>: cannot satisfy -package regex-compat
    (use -v for more information)
Makefile:326: recipe for target 'bsc' failed
make[2]: *** [bsc] Error 1
make[2]: Leaving directory '/home/leon/devel/bluespec/src/comp'
Makefile:11: recipe for target 'install' failed
make[1]: *** [install] Error 2
make[1]: Leaving directory '/home/leon/devel/bluespec/src'
Makefile:24: recipe for target 'install' failed
make: *** [install] Error 2
```
</details>

I had to change `$ cabal install regex-compat syb old-time`
to `$ cabal v1-install regex-compat syb old-time split`
Because cabal >=3 defaults to `v2-install`.
And also `split` was missing.

leonschoorl, 2020-03-10:
Also it seems that one needs to install ictl3 (on Ubuntu 18.4).

<details>
  <summary>`cannot find -litcl`</summary>
  
```
[...]
make  -C comp -j1   PREFIX=/home/leon/devel/bluespec/inst  install
make[2]: Entering directory '/home/leon/devel/bluespec/src/comp'
----- Normal build options -----
can't find package Itcl
bsc start di 10 mrt 2020 17:26:29 CET
./update-build-version.sh
BuildVersion.hs up-to-date
./update-build-system.sh
BuildSystem.hs up-to-date
ghc -fmax-pmcheck-iterations=6000000 -Wtabs -hidir /home/leon/devel/bluespec/src/comp/../../build -odir /home/leon/devel/bluespec/src/comp/../../build -stubdir /home/leon/devel/bluespec/src/comp/../../build -main-is Main_bsc \
	-O2 -hide-all-packages -fasm -Wall -fno-warn-orphans -fno-warn-name-shadowing -fno-warn-unused-matches -package base -package containers -package array -package mtl -package unix -package regex-compat -package bytestring -package directory -package process -package filepath -package time -package old-time -package old-locale -package split -package syb -package integer-gmp -iGHC -iGHC/posix -iLibs -i../Parsec -i../vendor/stp/include_hs -i../vendor/yices/include_hs -i../vendor/htcl '-tmpdir /tmp' -L../vendor/htcl -I/usr/include/tcl -I../vendor/stp/include -L../vendor/stp/lib -I../vendor/yices/include -L../vendor/yices/lib '-pgma ./compiler-script.sh' '-pgmc ./compiler-script.sh' '-pgml ./compiler-script.sh' -rtsopts --make bsc -j1 +RTS -M3G -A128m -RTS -ltcl8.6 -ltk8.6 -litcl -lhtcl -lstp -lyices  rts_hooks.o
Linking bsc ...
/usr/bin/ld: cannot find -litcl
collect2: error: ld returned 1 exit status
`compiler-script.sh' failed in phase `Linker'. (Exit code: 1)
Makefile:326: recipe for target 'bsc' failed
make[2]: *** [bsc] Error 1
make[2]: Leaving directory '/home/leon/devel/bluespec/src/comp'
Makefile:11: recipe for target 'install' failed
```
</details>

leonschoorl, 2020-03-10:
Now I'm getting a lot of linking errors:
```
/usr/bin/ld: /home/leon/devel/bluespec/src/comp/../../build/BDD.o: relocation R_X86_64_32S against `.text' can not be used when making a PIE object; recompile with -fPIC
```
<details>
<summary>And the same error for a lot of .o files</summary>

```
[...]
make  -C comp -j1   PREFIX=/home/leon/devel/bluespec/inst  install
make[2]: Entering directory '/home/leon/devel/bluespec/src/comp'
----- Normal build options -----
bsc start di 10 mrt 2020 17:27:48 CET
./update-build-version.sh
BuildVersion.hs up-to-date
./update-build-system.sh
BuildSystem.hs up-to-date
ghc -fmax-pmcheck-iterations=6000000 -Wtabs -hidir /home/leon/devel/bluespec/src/comp/../../build -odir /home/leon/devel/bluespec/src/comp/../../build -stubdir /home/leon/devel/bluespec/src/comp/../../build -main-is Main_bsc \
	-O2 -hide-all-packages -fasm -Wall -fno-warn-orphans -fno-warn-name-shadowing -fno-warn-unused-matches -package base -package containers -package array -package mtl -package unix -package regex-compat -package bytestring -package directory -package process -package filepath -package time -package old-time -package old-locale -package split -package syb -package integer-gmp -iGHC -iGHC/posix -iLibs -i../Parsec -i../vendor/stp/include_hs -i../vendor/yices/include_hs -i../vendor/htcl '-tmpdir /tmp' -L../vendor/htcl -I/usr/include/tcl -I../vendor/stp/include -L../vendor/stp/lib -I../vendor/yices/include -L../vendor/yices/lib '-pgma ./compiler-script.sh' '-pgmc ./compiler-script.sh' '-pgml ./compiler-script.sh' -rtsopts --make bsc -j1 +RTS -M3G -A128m -RTS -ltcl8.6 -ltk8.6 -litcl3.4 -lhtcl -lstp -lyices  rts_hooks.o
Linking bsc ...
/usr/bin/ld: /home/leon/devel/bluespec/src/comp/../../build/BDD.o: relocation R_X86_64_32S against `.text' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/leon/devel/bluespec/src/comp/../../build/Bag.o: relocation R_X86_64_32 against symbol `ghczmprim_GHCziClasses_zdfEqInt_closure' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/leon/devel/bluespec/src/comp/../../build/BinaryIO.o: relocation R_X86_64_32S against `.text' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/leon/devel/bluespec/src/comp/../../build/BuildSystem.o: relocation R_X86_64_32S against symbol `stg_bh_upd_frame_info' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/leon/devel/bluespec/src/comp/../../build/BuildVersion.o: relocation R_X86_64_32S against symbol `stg_bh_upd_frame_info' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/leon/devel/bluespec/src/comp/../../build/EquivalenceClass.o: relocation R_X86_64_32S against `.text' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/leon/devel/bluespec/src/comp/../../build/Eval.o: relocation R_X86_64_32S against `.text' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/leon/devel/bluespec/src/comp/../../build/Exceptions.o: relocation R_X86_64_32S against `.text' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/leon/devel/bluespec/src/comp/../../build/Fixity.o: relocation R_X86_64_32S against symbol `stg_bh_upd_frame_info' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/leon/devel/bluespec/src/comp/../../build/GraphPaths.o: relocation R_X86_64_32S against `.text' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/leon/devel/bluespec/src/comp/../../build/Classic.o: relocation R_X86_64_32S against `.text' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/leon/devel/bluespec/src/comp/../../build/IOUtil.o: relocation R_X86_64_32S against `.text' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/leon/devel/bluespec/src/comp/../../build/ListMap.o: relocation R_X86_64_32 against symbol `ListMap_lookupWithDefault_closure' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/leon/devel/bluespec/src/comp/../../build/Log2.o: relocation R_X86_64_32 against symbol `ghczmprim_GHCziTypes_True_closure' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/leon/devel/bluespec/src/comp/../../build/Parse.o: relocation R_X86_64_32S against `.text' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/leon/devel/bluespec/src/comp/../../build/RealUtil.o: relocation R_X86_64_32S against `.text' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/leon/devel/bluespec/src/comp/../../build/STPFFI.o: relocation R_X86_64_32S against symbol `base_GHCziPtr_Ptr_con_info' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/leon/devel/bluespec/src/comp/../../build/Sort.o: relocation R_X86_64_32S against symbol `stg_upd_frame_info' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/leon/devel/bluespec/src/comp/../../build/SystemVerilogKeywords.o: relocation R_X86_64_32S against `.text' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/leon/devel/bluespec/src/comp/../../build/Undefined.o: relocation R_X86_64_32S against `.text' can not be used when making a PIE object; recompile with -fPIC
[... cut ...]
/usr/bin/ld: /opt/ghc/8.8.2/lib/ghc-8.8.2/rts/libHSrts.a(M32Alloc.o): relocation R_X86_64_32 against `.rodata.m32_free_internal.str1.1' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: final link failed: Nonrepresentable section on output
collect2: error: ld returned 1 exit status
`compiler-script.sh' failed in phase `Linker'. (Exit code: 1)
Makefile:326: recipe for target 'bsc' failed
make[2]: *** [bsc] Error 1
make[2]: Leaving directory '/home/leon/devel/bluespec/src/comp'
```
</details>

Is there a place where I could enable this `-fPIC`?

bpfoley, 2020-03-10:
The linking error sounds like you might be running into a mismatch between the compiler flags that GCC uses by default, and the ones your install of GHC by default. This happened on several Linux distros (including Debian, Ubuntu and Arch) when they switched their toolchains to PIE by default but GHC wasn't updated to match.

See https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=712228

As a workaround for the moment, you could try adding `-fPIC` to the following line in `src/comp/Makefile`

```GHCOPTFLAGS = $(GHCOPTLEVEL) $(GHCFLAGS)```

But the 'proper' solution is to fix your install of GHC.

leonschoorl, 2020-03-11:
Adding `-fPIC` to `GHCOPTFLAGS` does fix the errors for all of bsc's .o files.
But all of GHC's system libraries are still broken according to ld.

My current GHC is 8.8.2 from [hvr's PPA](https://launchpad.net/~hvr/+archive/ubuntu/ghc) running on Ubuntu 18.04.4.
It builds fine on its own (`ghc --make`), with cabal and with stack.
I tried to reproducing the issues in the debian bug you linked, but couldn't.

So I'm not sure whether bsc's build system is revealing some hidden issue or causing one.

leonschoorl, 2020-03-12:
I was able to get it to build by removing the [`-pgma`,`-pgmc`,`-pgml` options](https://github.com/B-Lang-org/bsc/blob/a0d01cf21fba77e0fe0af3d9cd358f2eb770f8ed/src/comp/Makefile#L195-L197).
And `make check` reports no issues with the resulting binaries.

It seems the purpose of this [compiler-script.sh](https://github.com/B-Lang-org/bsc/blob/a0d01cf21fba77e0fe0af3d9cd358f2eb770f8ed/src/comp/compiler-script.sh) is to drop some specific compiler flags from some very old version of gcc.
But it has as a side effect overwriting some of GHC's defaults for compilers, linkers and/or their flags.

bpfoley, 2020-03-12:
Now that's an interesting finding. Thanks for persevering. I'll set up a VM later on and see if I can get to the bottom of what's going on here. `compiler-script.sh` shouldn't remove anything except for those options found in the script, so I wonder what's happening.

leonschoorl, 2020-03-13:
Just using `-pgmc gcc` makes already makes ghc fail:
```
$ echo 'main = putStrLn "Hello world!"' > test.hs
$ ghc --make test.hs
[1 of 1] Compiling Main             ( test.hs, test.o )
Linking test ...
$ ./test
Hello world!
$ rm test *.o *.hi
$ ghc --make -pgmc gcc test.hs
[1 of 1] Compiling Main             ( test.hs, test.o )
Linking test ...
/usr/bin/ld: test.o: relocation R_X86_64_32S against symbol `stg_bh_upd_frame_info' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /tmp/ghc58_0/ghc_5.o: relocation R_X86_64_32 against symbol `ZCMain_main_closure' can not be used when making a PIE object; recompile with -fPIC
[...]
```

Seems using `-pgmc` changes which flags ghc passes to the C compiler.
Most notably `-no-pie`.
```
$ rm test *.o *.hi
$ ghc -v --make test.hs |& grep -B1 gcc > normal.log
$ rm test *.o *.hi
$ ghc -v --make -pgmc gcc test.hs |& grep -B1 gcc > using-pgmc.log
$ diff -u normal.log using-pgmc.log
```

```diff
--- normal.log
+++ using-pgmc.log
@@ -1,9 +1,12 @@
 *** Assembler:
-gcc -DTABLES_NEXT_TO_CODE -iquote. -no-pie -fno-PIC -x assembler -c /tmp/ghc346_0/ghc_2.s -o test.o
+gcc -DTABLES_NEXT_TO_CODE -iquote. -fno-PIC -x assembler -c /tmp/ghc367_0/ghc_2.s -o test.o
 --
 *** C Compiler:
-gcc -DTABLES_NEXT_TO_CODE -c /tmp/ghc346_0/ghc_4.c -o /tmp/ghc346_0/ghc_5.o -no-pie -fno-PIC -I/opt/ghc/8.8.2/lib/ghc-8.8.2/include
+gcc -c /tmp/ghc367_0/ghc_4.c -o /tmp/ghc367_0/ghc_5.o -fno-PIC -I/opt/ghc/8.8.2/lib/ghc-8.8.2/include
 *** C Compiler:
-gcc -DTABLES_NEXT_TO_CODE -c /tmp/ghc346_0/ghc_7.s -o /tmp/ghc346_0/ghc_8.o
+gcc -c /tmp/ghc367_0/ghc_7.s -o /tmp/ghc367_0/ghc_8.o
 *** Linker:
-gcc -DTABLES_NEXT_TO_CODE '-fuse-ld=gold' -Wl,--no-as-needed -o test -lm -no-pie -fno-PIC -Wl,--gc-sections test.o -L/opt/ghc/8.8.2/lib/ghc-8.8.2/base-4.13.0.0 [...]
+gcc -DTABLES_NEXT_TO_CODE '-fuse-ld=gold' -Wl,--no-as-needed -o test -lm -fno-PIC -Wl,--gc-sections test.o -L/opt/ghc/8.8.2/lib/ghc-8.8.2/base-4.13.0.0 [...]
+--
+Deleting: /tmp/ghc367_0
+`gcc' failed in phase `Linker'. (Exit code: 1)

```

So I guess the issue is with ghc after all.

leonschoorl, 2020-03-13:
The dropping of the `-no-pie` flag is on purpose: https://gitlab.haskell.org/ghc/ghc/commit/8d008b71db53f7a59673f894f329b8d71f84c8ee
Because of: https://gitlab.haskell.org/ghc/ghc/issues/15319

leonschoorl, 2020-03-13:
GHC issue about this: https://gitlab.haskell.org/ghc/ghc/issues/17919

quark17, 2020-03-15:
I corrected the missing `split` in the `cabal install` command in commit e7facc63.  I will also update the README to mention newer cabal; I just need to decide what wording to use (possibly the suggestion in PR #74).

bpfoley, 2020-04-02:
Now that #143 has been submitted (removing compiler_script.sh) I hope this resolves your build issues @leonschoorl...

quark17, 2020-07-11:
I'm closing this since the only remaining issue is now captured by #201 .

closed 2020-07-11 by quark17