B-Lang-org/bsc #116: Tk dependencies should be optional

thoughtpolice, 2020-03-11:
The dependencies for the Bluespec package should be *very* minimal at runtime, but because there's a hard dependency on Tk (from what I can tell), you need to pull in other things like font-config, X11, etc. This percolates into a lot of unneeded things. I think all we should really need is `libc.so`, `libstdc++.so`, `libgmp.so`, and perhaps one or two others...

My use case is that I have a [tool](https://github.com/thoughtpolice/nix-makeself) that can create distro-agnostic binary tarballs, similar to the way Bluespec was packaged when it was proprietary (but with a much more fancy usage mechanism so it works anywhere.) Right now adding Bluespec would be a pretty big overhead -- other tools like NextPNR also have this problem when packaging them (they use Qt5, not Tk, however.)

quark17, 2020-03-11:
The only tool that really needs `libX11` etc is `bluewish`, and I think it's entirely reasonable to have a compile option that avoids building that and therefore avoids the need for X11 etc.  I believe, though, that currently `bluetcl` unnecessarily has the same requirements as `bluewish`, probably because they share the same compile flags (perhaps in an effort to avoid recompilation of the Haskell modules when the the flags change).

bpfoley, 2020-03-11:
Yeah, this makes sense to me. If you want to use bluesim, tcl is currently required, but Tk should be a (build time) option.

I suspect the best way to do this would be to split the entire bluespec codebase up into several Haskell modules that can be compiled independently, and only have tk deps for bluewish, and tcl deps for bluetcl.

thotypous, 2020-04-12:
As a related question, are there plans for releasing the workstation (BDW) as open source? I was curious as to whether bluewish is used outside of BDW. Do you have any customers who wrote in-house tools using Tk?

quark17, 2020-04-12:
I've created a version of the Makefile that removes the Tcl/Tk/X11 dependencies from BSC and the TK/X11 dependencies from Bluetcl.  It's attached to PR #144 and just waiting for @bpfoley to review.

I don't know if anyone is using Bluewish aside from the Workstation GUI.  But hopefully that's moot with the above Makefile fix.

We do plan to release the GUI as open source, but I'm not sure when.  Issue #69 is open for that.

quark17, 2020-04-14:
PR #144 has been merged, so now only the `bluewish` binary should require tk or X11 to run.  The user of a release now won't need tk/X11 for anything else in the release.  However, the builder of the release will still need tk/X11 to build `bluewish`.  The Makefile also requires `wish` and `xvfb-run` just to construct variables (even if it doesn't build the `bluewish` target), but these are for itk and that might go away when issue #149 is resolved.

@thoughtpolice, I assume that you would still want a way to build without tk/X11/etc available?  For that, I think it would be reasonable to add a variable which disables the building of `bluewish` (and the construction of any itk variables, if they exist) when provided.  For example, `make NO_TK=1` or `NOWISH` or `NOX11` or something.  Does that work?  Any naming preferences?

Another option would be to create a variable for specifying a type of build.  For example, maybe `BSC_BUILD_TYPE` could be assigned `all` to turn on build/install of the developer tools (which are not currently built by default), or it could be set to `no_wish` or other values to specify different subsets of tools to build.  Or we could even allow defining `BSC_BUILD_TARGETS` with the explicit list of targets.

I'm fine with `NO_TK=1` but I'm flexible.

thotypous, 2020-04-15:
It's currently (6a8cedf89d93d3753ef118b031436400368131c6) possible to build without Tk by passing `TCLEXES=bluetcl BLUEWISH_TOOLS=` to `make`.

quark17, 2020-04-15:
Maybe we don't even need to produce a `bluewish` executable, now that we're not statically linking tcl/tk/itcl/itk.  The only difference in `bluewish` is that it is pre-linked with the `tk` library (and may possibly have pre-loaded the `Tk` package?), but `bluetcl` can dynamically load that from the user's system:

    $ bluetcl
    % package require Tk
    8.5.9
    % package ifneeded Tk 8.5.9
    load /System/Library/Frameworks/Tk.framework/Resources/Scripts/../../Tk Tk

thotypous, 2020-04-15:
> `bluetcl` can dynamically load that from the user's system

Very well noticed! One build dependency less and one 26MB binary less installed in the system 😊

bpfoley, 2020-04-17:
> Maybe we don't even need to produce a `bluewish` executable, now that we're not statically linking tcl/tk/itcl/itk. The only difference in `bluewish` is that it is pre-linked with the `tk` library

Yes! Especially since `bluewish.hs` is literally a copy of `bluetcl.hs`. Let's just remove that target and binary entirely!

PR #160 (2020-04-20): Remove bluewish, and all dependencies (including tk, x11, fontconfig)
Update the docs to reflect this, and while here, update a slightly
bitrotted vagrant provisioning script.

NB this resolves #116
closed 2020-04-20 by quark17