B-Lang-org/bsc #126: Couldn't find litk4.1.2 for Ubuntu 19.10

karthyyy, 2020-03-15:
Trying to install open source bsv on Ubuntu 19.10. live usb. Couldn't find the package tk-itcl4-dev. While making its showing error "Cannot find -litk4.1.2". Worked fine on Ubuntu 16.04. How to proceed.

bpfoley, 2020-03-15:
It's a typo in the README. See PR #123. The correct name is actually `tcl-itcl4-dev`. Sorry for the confusion!

closed 2020-03-15 by bpfoley

cyrozap, 2020-03-15:
@bpfoley And not only that, even if @karthyyy had the library installed, the "Cannot find -litk4.1.2" error is caused by the issue that was fixed in #124. But like I just said, that's been fixed, so @karthyyy you shouldn't have this problem with the build any more.

karthyyy, 2020-03-15:
Thanks a lot for the replies. I ll try installing once again and if the issue persists, ll get back. Once again thanks for the prompt response.