B-Lang-org/bsc #136: Bluespec Not Building on macOS Catalina 10.15.3

BracketMaster, 2020-03-24:
I did:

```bash
brew install cabal-install fontconfig gperf icarus-verilog pkg-config && \
    cabal update && \
    cabal v1-install old-time regex-compat split syb
make -j2 GHCJOBS=2 GHCRTSFLAGS='+RTS -M4500M -A128m -RTS' MACOSX_DEPLOYMENT_TARGET=10.13
```

Which is what is used in the GitHub CI runner and get the following error:
```ghc -Wtabs -optl -headerpad_max_install_names -hidir /Users/yehowshuaimmanuel/src/bsc/src/comp/../../build -odir /Users/yehowshuaimmanuel/src/bsc/src/comp/../../build -stubdir /Users/yehowshuaimmanuel/src/bsc/src/comp/../../build -main-is Main_bsc \
		-O2 -hide-all-packages -fasm -Wall -fno-warn-orphans -fno-warn-name-shadowing -fno-warn-unused-matches -package base -package containers -package array -package mtl -package unix -package regex-compat -package bytestring -package directory -package process -package filepath -package time -package old-time -package old-locale -package split -package syb -package integer-gmp -iGHC -iGHC/posix -iLibs -i../Parsec -i../vendor/stp/include_hs -i../vendor/yices/include_hs -i../vendor/htcl '-tmpdir /tmp' -L../vendor/htcl -I/usr/include/tcl -I/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Tk.framework/Headers -I../vendor/stp/include -L../vendor/stp/lib -I../vendor/yices/include -L../vendor/yices/lib '-pgma ./compiler-script.sh' '-pgmc ./compiler-script.sh' '-pgml ./compiler-script.sh' -rtsopts --make bsc -j2 +RTS -M4500M -A128m -RTS -ltcl8.6 -ltk8.6 -litcl4 -lhtcl -L"" -lstp -lyices  rts_hooks.o
ghc: on the commandline: missing argument for flag: -L
Usage: For basic information, try the `--help' option.
make[2]: *** [bsc] Error 1
make[1]: *** [install] Error 2
make: *** [install] Error 2
```

BracketMaster, 2020-03-24:
I got it working on a clean install of Catalina. I'll have to see why it didn't work on my old Mac.
Also note that this time I passed ``MACOSX_DEPLOYMENT_TARGET=10.15`` instead of ``10.13`` as stated above.

BracketMaster, 2020-03-24:
Ok. I found the issue. The makefile chokes if the system has a ``brew`` provided tclsh.
I can fix this by

I will fix this by and an if clause for Darwin that defaults to the system ``tclsh`` provided by Apple.

bpfoley, 2020-03-24:
I think the issue for this might be that homebrew's tcl-tk package doesn't have itk yet. I've an open PR for that: https://github.com/Homebrew/homebrew-core/pull/51151

Also, I attempted to get STP and minisat into homebrew (which I'm pretty sure we'll need to do instead of using our own copy if we want to do a homebrew Formula for bsc), and the maintainers didn't think it was suitable: https://github.com/Homebrew/homebrew-core/pull/50020

BracketMaster, 2020-03-24:
I went ahead an made a pull request [here](https://github.com/B-Lang-org/bsc/pull/138) that causes the makefile to avoid ``brews`` install of tclsh.

Do you think it should be merged? If so, I can go ahead and close this issue.

bpfoley, 2020-03-25:
> I think the issue for this might be that homebrew's tcl-tk package doesn't have itk yet. 

Actually scrap that, I just wasn't thinking straight. Homebrew's tcl-tk doesn't have itk, but that's not relate.d

BracketMaster, 2020-03-26:
I've made a tap for Homebrew Tap for Bluespec Verilog that works fine [here](https://github.com/BracketMaster/homebrew-rtl/blob/master/bsc.rb).

```bash
brew tap bracketmaster/rtl
brew install bsc
```
I can't merge into homebrew core right now because you guys don't have a tarball release yet(homebrew checks that you're using official releases when auditing for core).

Hopefully, that will change in the future.

Also, bsc fails to pass homebrew core audits because homebrew requires any libraries built have a dylib extension.

I know this is superficial - but I didn't make the rules. Maybe if someone wants to add an ``if(Darwin)`` predicate to the makefile that changes to dylib for Mac builds - then it could work.

Thoughts? Comments?

bpfoley, 2020-03-29:
OK, that's pretty easy to fix. See #140.

closed 2020-04-09 by BracketMaster

quark17, 2020-09-05:
@agoodloe If this is still a problem, can you open a new GitHub issue for it, instead of commenting on this issue?

Note that the failure is while building the third-party STP library.  In the worst case, you can omit building STP by assigning `STP_STUB` a non-empty value on the `make` command-line.

I now have macOS 10.15.6 installed, and I am able to build BSC.  I have not rebuilt my Homebrew packages, though, and I have not explicitly done anything about Xcode or the commandline tools (other than what the macOS upgrade might have done itself).  Maybe if I did that, I would see the error you're seeing?  In my build, I see this:
```
10 warnings generated.
rm -f libcppinterface.a
ar qcs libcppinterface.a cpp_interface.o
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C parser 
```
which has a different path to `make` than shown in your output.  Is it possible that you don't have Xcode installed, just the command-line tools? or, if you do have Xcode, that you installed the command-line tools afterward and so `xcode-select` is pointing to that instead of to XCode?  I don't know if this is related to your error, or even if Xcode vs command-line tools matters, but it's a difference that I notice.

It could also be that, instead of using the native macOS tools, you've installed a tool (with Homebrew or MacPorts or some other way) and that is being used instead.  BSC builds on GitHub's macOS CI environment after installing a small number of Homebrew packages (see the install script in `.github/workflows/`).  I have done that same install command, and may have a small number of other packages installed, but I have not replaced `cc`/`gcc`/`ar` or `bison`/`flex` -- the ones in my path are from `/usr/bin/`.  The STP code is an old snapshot of third-party code, and the build scripts from back then may not work with a hybrid system; a newer version of the STP code might have better build scripts.