B-Lang-org/bsc #150: Create Docker

bpfoley, 2020-04-14:
Capturing this as a separate issue suggested by @thotypous in #147 

It would be superb if you could set up [automated builds](https://docs.docker.com/get-started/part3/) on Docker Hub using the Dockerfile.

An official image on Docker Hub would be very useful for setting up automated CI builds on Bluespec designs. It would also help to attract more newcomers (as an alternative or complement to ordinary binary releases).

bpfoley, 2020-04-14:
For now, I've set up a not-quite-official container at https://hub.docker.com/r/bpfoley11/bsc-ubuntu1804 while we figure out the process.

benjaminselfridge, 2020-06-19:
@bpfoley Any chance you can link the Dockerfile used to create this image?

I'm trying to write a bsc dockerfile currently, and am getting a strange Makefile error while compiling `Parser.BSV.CVParser`. It would be nice to contrast with a successful dockerfile and see what I'm getting wrong.

thotypous, 2020-06-21:
GitHub now has its own docker registry, so now it should be pretty simple: https://github.com/features/packages

thotypous, 2020-10-27:
It seems like you have been [trying](https://github.com/B-Lang-org/bsc/runs/1312598964) to use Cirrus CI [zero-configuration](https://medium.com/cirruslabs/zero-config-docker-builds-with-cirrus-ci-703db9c1e04e) docker builds. However, it does not fetch the git submodules, therefore the build fails. There seems to be [some ways to workaround](https://github.com/cirruslabs/cirrus-ci-docs/issues/407) it, but it is not clear how to use the workaround with docker builds.

Maybe it would be easier to use [GitHub actions](https://github.com/marketplace/actions/build-and-push-docker-images) to build the container instead of Cirrus CI.

quark17, 2020-10-27:
I enabled the Cirrus CI app for this repo so that it could be used to test FreeBSD builds (see PR 260).  When I did that, docker builds started, but that was not my intention.  I have not had the time yet to learn how the Cirrus CI works, either to find out why it is doing docker builds or how to start FreeBSD builds (but I was leaving that to be done in the PR).