B-Lang-org/bsc #170: Add automated linking for Verilator

quark17, 2020-05-05:
BSC can automate the linking of a Verilog simulation for a number of simulators.  We should add Verilator to that list:
```
bsc -e topMod -verilog -vsim verilator
```

Basic support will require two things: (1) a new script called `bsc_build_vsim_verilator` in `src/exec/` and (2) a `verilator_main.cpp` file (added somewhere in the repo) to perform the function that `src/Verilog/main.v` does for most other Verilog simulators.

Adding support for a new simulator is usually as simple as adding a new script in `src/exec/` with a name starting with `bsc_build_vsim_` and followed by the name of the simulator.  The BSC executable does not contain any information specific to the simulator.  BSC merely calls the appropriate script, using arguments to pass generic information about the design.  The script contains all the knowledge about how to link the simulator, given that generic info.  It should be possible to copy one of the existing scripts and change it for the specifics of Verilator.

The one wrinkle is that Verilog needs a top-level C++ file, which most simulators do not.  For most simulators, we link the user's design with a `main.v` file that instantiates the design, provides a clock and reset, and turns on VCD dumping (if directed by plusargs).  This file is provided in the release and can instantiate any module (by using preprocessor macros to insert the user's specific module name etc).  For Verilator, we will need to a "verilator_main.cpp" that does the same thing -- instantiates a model (as specified by preprocessor macro), provides the clock and reset, enables VCD or tracing, etc.  I'm not sure where this file would live.  Perhaps we create a new directory `Verilog.Verilator` and put it there; or just in `Verilog`; or some new place for simulator files; or something else.  This file doesn't need to be written from scratch but can be adapted from an existing file for the Bluespec RISC-V processors [[link](https://github.com/bluespec/Piccolo/blob/master/builds/Resources/Verilator_resources/sim_main.cpp)].

There's actually a second wrinkle: If a BSV design contains foreign function calls, BSC will generate VPI functions (wrapped around the imported C functions), and the scripts are responsible for linking  those VPI functions into the simulator.  Verilator does not support all of VPI that BSC uses (see issue #169 ), so a first version of the script may have to give an error if VPI files are provided.  Although it may be possible for the script to auto-generate equivalent DPI imports.

jahofmann, 2020-06-18:
Instead of supporting each simulator directly in BSC the project could use a tool like [Cocotb](https://github.com/cocotb/cocotb) which does the heavy lifting.

Then there would be only one top level Python file for all simulators.

quark17, 2025-10-17:
I'm tempted to close this issue, although it could be left open while we consider whether deeper integration with Verilator's clocking mechanisms is needed.  We should at least check how many remaining tests in the testsuite still cannot build with the latest Verilator.  The suggestion of Cocotb is an interesting one, but perhaps a separate issue can be opened for that.  (I also don't personally know enough about Cocotb to understand what is being suggested or what would be needed.)

Current state of things:

BSC now supports generating verilog with either DPI or VPI calls, for imported C functions (although polymorphic functions are not yet supported for DPI, see issue #169).  To specify DPI, `--use-dpi` is provided on the compile and link command lines.  And BSC supports `bsc -vsim verilator` for linking (as long as DPI is used).  The BSC repo provides a `main.cpp` that matches the behavior of `main.v` (used for other Verilog simulators) and performs linking with a `bsc_build_vsim_verilator` script.  Verilator no longer requires a C++ main to be provided -- it can be invoked with flags that automatically generate a main, to behave like other Verilog simulators -- so in theory BSC could possibly use `main.v` with Verilator and not have to keep a separate file in sync, but that probably requires using the `--timing` flag which might have performance consequences.  Verilator hadn't been able to support many of the multi-clock primitives, way back when I last checked, so large parts of the testsuite wouldn't run with it.  But there have been many improvements to Verilator since then, so it might be worth checking how much of the testsuite runs now -- but, again, this requires the `--timing` flag which might have performance consequences.  A possibility might be for BSC to analyze the design (similar to the Bluesim backend) to identify the clock sources etc, and perhaps generate a custom `main` that does all the clock handling there, rather than in the Verilog, to avoid performance hits.  But that would be a significant project.