B-Lang-org/bsc #191: .ba and .bo files aren't fully deterministic

bpfoley, 2020-07-02:
.ba and .bo files embed absolute paths for their dependencies instead of relative ones, which breaks determinism between different compilation environments.

This hurts on build systems (such as scons, Google's Bazel, Facebook's Buck etc) that use hashes of object file contents to determine whether dependencies should be rebuilt or not.

If we can make bsc consistently use relative paths (at least as an option) that would fix this. Without package hierarchy, I'm not sure how useful the paths are in any case.

bpfoley, 2020-07-02:
TODO: Examine .ba and .bo generation to see what the contents are and how it needs to be canonicalized (eg entries may need to be sorted, dates stripped, and file paths canonicalized.)

quark17, 2020-08-17:
Was any of this improved with the addition of the `-(no-)show-version` flag and further use of the existing `-(no-)show-timestamps` flag?

quark17, 2024-09-22:
I assume that this was resolved through the mentioned flags.  If there's still an issue, feel free to re-open.

closed 2024-09-22 by quark17