B-Lang-org/bsc #205: Poor Code Generation For Non-Consecutive And/Or Non-Zero-Based Enums

jrtc27, 2020-07-15:
Consider the following code (the methods are `always_enabled` purely to remove the RDY/EN signals that aren't relevant here):

```bsv
`define CONCAT(a, b) a``b

typedef Bit #(4) Type0;
typedef enum { A, B, C, D, E, F, G, H, I } Type1 deriving (Bits, Eq);
typedef enum { A = 1, B, C, D, E, F, G, H, I } Type2 deriving (Bits, Eq);
typedef enum { A, B = 2, C, D, E = 6, F = 8, G, H = 12, I = 15 } Type3 deriving (Bits, Eq);

typedef `CONCAT(Type, `TEST) Type;

interface Test_IFC;
   (* always_enabled *)
   method Action put (Type v);
   (* always_enabled *)
   method Type get;
endinterface

(* synthesize *)
module `CONCAT(mkTest, `TEST) (Test_IFC);
   Reg #(Type) rg_v <- mkRegU;

   method Action put (Type v);
      rg_v <= v;
   endmethod

   method Type get;
      return rg_v;
   endmethod
endmodule
```

I would expect it to generate the same code across all four tests, as they're equivalent, and the values in the Enum aren't relevant unless you actually want to perform comparisons on values. For Test0 and Test1, everything is as expected, and bsc generates the following code for get and put:

```verilog
  // value method get
  assign get = rg_v ;

  // register rg_v
  assign rg_v$D_IN = put_v ;
  assign rg_v$EN = 1'd1 ;

  // handling of inlined registers

  always@(posedge CLK)
  begin
    if (rg_v$EN) rg_v <= `BSV_ASSIGNMENT_DELAY rg_v$D_IN;
  end
```

i.e., the enum is just a way to provide a type-safe set of unique `Bit #(4)` constants.

However, if the enum label values are altered in any way such that they are no longer 0 to n-1 in that order (by permuting that sequence, shifting, or introducing holes), then suddenly bsc decides it wants to enumerate all possible values when variables of that type are involved. For Test2 it gives the following:

```verilog
  // value method get
  always@(rg_v)
  begin
    case (rg_v)
      4'd1, 4'd2, 4'd3, 4'd4, 4'd5, 4'd6, 4'd7, 4'd8: get = rg_v;
      default: get = 4'd9;
    endcase
  end

  // register rg_v
  always@(put_v)
  begin
    case (put_v)
      4'd1, 4'd2, 4'd3, 4'd4, 4'd5, 4'd6, 4'd7, 4'd8: rg_v$D_IN = put_v;
      default: rg_v$D_IN = 4'd9;
    endcase
  end
  assign rg_v$EN = 1'd1 ;

  // handling of inlined registers

  always@(posedge CLK)
  begin
    if (rg_v$EN) rg_v <= `BSV_ASSIGNMENT_DELAY rg_v$D_IN;
  end
```

This introduces a whole bunch of logic to map any invalid values to the (arbitrarily-chosen?) last label in the enum. Test3 gives similar code (defaulting `I` again which is now `15`), demonstrating that the issue is for anything outside the basic enum declarations:

```verilog
  // value method get
  always@(rg_v)
  begin
    case (rg_v)
      4'd0, 4'd2, 4'd3, 4'd4, 4'd6, 4'd8, 4'd9, 4'd12: get = rg_v;
      default: get = 4'd15;
    endcase
  end

  // register rg_v
  always@(put_v)
  begin
    case (put_v)
      4'd0, 4'd2, 4'd3, 4'd4, 4'd6, 4'd8, 4'd9, 4'd12: rg_v$D_IN = put_v;
      default: rg_v$D_IN = 4'd15;
    endcase
  end
  assign rg_v$EN = 1'd1 ;

  // handling of inlined registers

  always@(posedge CLK)
  begin
    if (rg_v$EN) rg_v <= `BSV_ASSIGNMENT_DELAY rg_v$D_IN;
  end
```

This logic makes it infeasible to use enums in this way for synthesised designs. This is unfortunate, as they provide a very attractive and concise way to define a set of constants of a given type, with the ability to derive FShow, and to statically ensure values of the wrong type aren't mixed up (as might happen if you just used `Bit #(n)` instead). Especially when large enums, tagged unions and EHRs are used.

Notably, this affects MIT's RISCY-OOO and, by extension, Bluespec's own Toooba core. RISCY-OOO likes to use this pattern, making use of it for CSR numbers, major opcodes, rounding modes, exception codes and interrupt numbers. For example:

```bsv
typedef enum {
    InstAddrMisaligned  = 4'd0,
    InstAccessFault     = 4'd1,
    IllegalInst         = 4'd2,
    Breakpoint          = 4'd3,
    LoadAddrMisaligned  = 4'd4,
    LoadAccessFault     = 4'd5,
    StoreAddrMisaligned = 4'd6,
    StoreAccessFault    = 4'd7,
    EnvCallU            = 4'd8,
    EnvCallS            = 4'd9,
    EnvCallM            = 4'd11,
    InstPageFault       = 4'd12,
    LoadPageFault       = 4'd13,
    StorePageFault      = 4'd15
} Exception deriving(Bits, Eq, FShow);
```

In their reorder buffer, a tagged union of this and an equivalent `Interrupt` enum are put into an EHR, which leads to a lot of unnecessary logic. In our CHERI-Toooba fork we introduce a third member of that tagged union for additional capability exception reporting, as well as increase (by about a factor of 2) the number of ports on that EHR. This leads to an explosion of logic causing the number of LUTs in the ROB to be a factor of 2-3 bigger than the non-CHERI Toooba. After refactoring the code to avoid enums, which requires introducing more boilerplate to provide an equivalent FShow instance (and lacks the static check that a value isn't reused), all this unnecessary logic goes away and we see significantly improved area (the core's CLB LUTs utilisation on a VCU118 decreases by about 10% from this semantically-equivalent refactoring, and the ROB's specifically goes down by around 50%).

jrtc27, 2020-07-15:
For compiling all the tests I used the following Makefile:
```make
NUM_TESTS = 4

seq = $(if $(filter $1, $(words $(value 2))), $(value 2), $(call seq, $1, $(value 2) $(words $(value 2))))

TOPS = $(patsubst %, mkTest%.v, $(call seq, $(NUM_TESTS)))

all: $(TOPS)

Test%.bo mkTest%.v: Test.bsv
	bsc -verilog -D TEST=$* -o $@ $<

clean:
	rm -f *.bo *.v
```

To be clear, this issue does directly affect RISCY-OOO and Bluespec's Toooba, not just our CHERI-Toooba, and they would see area wins from this being fixed. It's just that CHERI-Toooba's richer exception types and paths exacerbate the problem.

nanavati, 2020-07-15:
https://github.com/B-Lang-org/bsc/pull/161/ may help with this if you play with the newly introduced -derive-pack-last-clause and -opt-undet-expand flags.

Those changes weren't targeted at the custom tag case so I'm not sure what the right combination of flags would be for that, but it's worth trying. Feedback on how that worked out would be very appreciated since that could point at additional changes worth adding before merging.

jrtc27, 2020-07-16:
> #161 may help with this if you play with the newly introduced -derive-pack-last-clause and -opt-undet-expand flags.
> 
> Those changes weren't targeted at the custom tag case so I'm not sure what the right combination of flags would be for that, but it's worth trying. Feedback on how that worked out would be very appreciated since that could point at additional changes worth making before merging.

Doesn't seem to make a difference with `-derive-pack-last-clause -opt-undet-expand` (using commit cca9a02).

jrtc27, 2020-07-16:
And, whatever's going on in that pull request (I don't claim to understand all the details), I think bsc's default behaviour should be to have all the examples in this bug produce the same code. Otherwise it just requires unnecessary flags to give what is the most intuitive behaviour (or at least what seems to us at Cambridge and the authors at MIT, maybe others feel differently), and I don't see a case where the current behaviour is actually useful in these situations? I can't speak for your PR though wrt that.

quark17, 2020-07-16:
@jrtc27, I think you're correct to want the Verilog to be simply wires.  I'm starting to suspect that we could make that happen by defining new primitives `PrimEnumChr` and `PrimEnumOrd` that represent the bit packing/unpacking and reduce to identity when applied to each other.  (The reason your `Test1` example reduces to wires is because the derived `Bits` instance for that kind of enum is defined using`PrimChr` and `PrimOrd`, which together reduce to identity.)

In more detail:

When a BSV design reads a value of type T from a register (or a module input port), there is an implicit call to `unpack` to convert the raw bits of that register (or port) into a value of type T.  When a BSV design puts a value of type T into a register (or an output port), there is an implicit call to `pack` to convert the value of type T into the raw bits to the register (or port).

In the attached example, the `get` method is unpacking the bits from a register into a value and then immediately packing the value to get the bits for the output port.  Similarly, the `put` method is doing unpack and then pack, from an input port to a register.  The resulting verilog comes from evaulating the expressions:
```
get_result_port = pack (unpack (rg$Q_OUT));

rg$D_IN = pack (unpack (put_arg_port));
```
As you say, when the value is just being moved in this way, we'd like the Verilog to just be wires.  No disagreement there!

When the type T is a Bit vector, the pack/unpack functions are identities, and this does become wires in the Verilog -- as your Test0 shows.  When the type T is an enum with power-of-two values (and a derived Bits instance), it also becomes an identity.  When the type T is a enum with less than power-of-two values, the expression doesn't always reduce to wires.

How the expression is transformed depends on the interaction of several things (all of which could possibly be improved):

1. The derived Bits instance for enums is constructed in a particular way.

    When the enum has a contiguous packing starting from zero, pack and unpack are defined using `PrimChr` and `PrimOrd` (functions that map between a number `n` and the nth constructor of a type).  (See `Deriving.hs` lines 429 and 443 [here](https://github.com/B-Lang-org/bsc/blob/c7eaac65f4bf52d02acef229efe08f3580ea9e7a/src/comp/Deriving.hs#L429).)

    For all other enums, pack and unpack are defined as case statements with no default clause.  (This turns out to be treated differently than a case statement with a default clause that explicitly returns a don't-care value.  During elaboration, if the fall-through case is needed, it becomes an implicit don't-care; and BSC distinguishes between these two kinds of don't-care.) As a workaround, you can always explicitly define a Bits instance instead of deriving.

2. BSC's evaluation consists of a large set of rules for transforming expressions, including rules for how don't-care values are transformed (when they appear in expressions).

    To generate Verilog for the expression `pack (unpack (x))`, BSC inlines the definitions for `pack` and `unpack` and then begins reducing the expression using rules, until the expression consists only of synthesizable primitives.  A rule says "when the expression matches this pattern, replace it with this simpler expression".

    There are rules for case-statements, rules for don't-care values, rules that only apply when logic appears once (vs when it appears multiple times and should be defined as a signal to be shared in both places), etc.  Many rules contribute to the Verilog generated in this example, so any of them could be places where improvements could be made.  Or their interaction could be improved.  Or new rules could be added.

    As I mentioned in item 1, there isn't one set of rules that applies for all don't-care values.  Some rules only apply for some kinds.  That could be a place for improvement.

3. By default, BSC hinders the optimization of don't-care values so that the Bluesim and Verilog backends match value for value (specifically, the `opt-undertermined-vals` flag is off by default).

    People who want optimized Verilog and are not concerned about the divergence from Bluesim should definitely turn this flag on.  We can consider whether this flag should be on by default.  (See also the `unspecified-to` flag, which can be set to `X` when not trying to match Bluesim, which is only a 2-value simulator.)

The PR that @nanavati mentioned (#161) is attempting improvements in areas 1 and 2 (the derived Bits definitions and the rules for don't-care, as well as making it easier for rules to know when they are dealing specifically with enums, and adding new rules that turn enum values into `PrimChr` to help with reduction).  The PR makes some changes to the default behavior and leaves some changes to be enabled by flags -- I don't know to what extent those flags are there to allow the PR to merged without breaking too much, so that the behavior behind the flags can be tested for a while, maybe refined, and then decide whether they should become the default behavior.  In any case, I think the PR is still an experiment in progress, so I would caution against judging its flags and features as it stands.

As @nanavati said, the PR was not specifically for fixing the situation with custom enums.  It is doing some things to improve the similar situation for the bit packing of tagged unions (where the tag packing can sometimes result in an unnecessary case-state [... cut ...]  the original types, and not yet converted entirely to bits.  The second elaboration stage would be allowed to replace pack/unpack with their definitions and continue until the design was converted to bits.  (The benefit of the midpoint was to be able to capture type information or to fork off a simulator backend that worked with higher types, etc.)  Anyway, that first stage could be made to cancel out pack/unpack pairs, without having to introduce new `PrimEnumChr` and `PrimEnumOrd` primitives.

nanavati, 2020-07-16:
This thread has sparked a different idea on my part. PrimOrd and PrimChr are already primarily (exclusively?) used for enums so I was thinking the trick might be changing their interpretation. Specifically, if we changed the compiler's/evaluator's internal constructor numbering to reflect the custom tagging, I think ordinary PrimOrd / PrimChr cancellation could take care of the rest.

quark17, 2021-01-26:
@jrtc27, can you check whether this is now fixed (with commit 629c10ad)?  @nanavati implemented the idea in his comment from July, and he added a test case based on your example here (which I've just merged in bsc-testsuite).  But I'd like to get your confirmation -- if you're satisfied, can you mark this ticket as closed?  (I'm excited to see this fixed!)

closed 2023-08-24 by quark17