B-Lang-org/bsc #222: Cannot synthesize: The interface method uses type `Reg#(Int#(32))' which is not in the Bits class

ssmolov, 2020-07-30:
Hello, I'm trying to write matrix multiplication on BSV. Here is my code:
```
import RegFile::*;
import Vector::*;

`define SIZE 2

typedef Vector#(`SIZE, Vector#(`SIZE, Reg#(int))) MatrixType;

interface MatMul_iface #(parameter type aType);
  method Action start(aType x, aType y);
  method aType result();
endinterface: MatMul_iface

(* synthesize *)
module matMul(MatMul_iface#(MatrixType));
  MatrixType product <- replicateM(replicateM(mkReg(0)));
  MatrixType first <- replicateM(replicateM(mkReg(0)));
  MatrixType second <- replicateM(replicateM(mkReg(0)));
  Reg#(Bool) done <- mkReg(False);
  Reg#(Bool) initialized <- mkReg(False);

  rule cycle (!done && initialized);
    for (int i = 0; i < `SIZE; i = i + 1)
      for (int j = 0; j < `SIZE; j = j + 1)
        for (int k = 0; k < `SIZE; k = k + 1)
          product[i][j] <= product[i][j] + first[i][k] * second[k][j];
    done <= True;
  endrule
  
  method Action start(MatrixType x, MatrixType y) if (!done && !initialized);
    for (int i = 0; i < `SIZE; i = i + 1)
      for (int j = 0; j < `SIZE; j = j + 1)
        first[i][j] <= x[i][j];
    for (int i = 0; i < `SIZE; i = i + 1)
      for (int j = 0; j < `SIZE; j = j + 1)
        second[i][j] <= y[i][j];
	initialized <= True;
  endmethod
  
  method MatrixType result() if (done);
    return product;
  endmethod
endmodule: matMul
```
The compiler successfully generates *.bo, but it fails to generate Verilog. Here is the error log:
```
Error: "matMul.bsv", line 14, column 8: (T0043)
  Cannot synthesize `matMul': The interface method `start' uses type
  `Reg#(Int#(32))' which is not in the Bits class.
  During elaboration of the interface method `start' at "matMul.bsv", line 14,
  column 8.
  During elaboration of `matMul' at "matMul.bsv", line 14, column 8.
```

The code looks naive and it is the first Bluespec program for me. What am I missing here?

rsnikhil, 2020-07-31:
Hello Sergey, please see the attached tar file starting with its README.txt file for comments on the compiler error, plus some fixed-up versions.
[Matrix_Multiplication.tar.gz](https://github.com/B-Lang-org/bsc/files/5004569/Matrix_Multiplication.tar.gz)

ssmolov, 2020-07-31:
@rsnikhil thank you for the great and exhaustive answer! It helps me so much:)

A couple of questions I have though. I would be very grateful if you answer them.

1. In `Top.bsv` you've changed the following line of the original design:
```
product[i][j] <= product[i][j] + first[i][k] * second[k][j];
```
into:
```
w = w + first[i][k] * second[k][j];
```
What is it for?

2. In `mkTop` module at `Top.bsv` there are two non-conditioned rules: `rl_feed` and `rl_drain`. Why the first rule will be deterministically executed before the second one?

rsnikhil, 2020-07-31:
>----------------
>    In Top.bsv you've changed the following line of the original design:
>        product[i][j] <= product[i][j] + first[i][k] * second[k][j];
>    into:
>        w = w + first[i][k] * second[k][j];
>    What is it for?

My apologies for not commenting on this in the original README; it is
an important point.

In BSV, currently, rules are compiled to digital hardware such that
each rule can 'fire' (execute) at most once in a clock cycle.  In
digital hardware, a register can be assigned at most once in a clock
cycle.  The original formulation:

>        product[i][j] <= product[i][j] + first[i][k] * second[k][j];

represents (through static for-loop expansion) into N assignments for
each register.  The new formulation:

>        w = w + first[i][k] * second[k][j];

is just a notation for constructing a combinational circuit computing
_values_ from _values_.  The 'for-loop' does not represent sequencing
in time, it 'statically elaborated' (i.e., expanded by the compiler),
resulting in a single big combinational circuit computing the
inner-product.

Finally, we do a _single_ register assignment to the (i,j)'th register:

>        product[i][j] <= w;

If you restore the original formulation and try compiling it (please
try it!), the compiler will complain about 'conflicting parallel
assignments' This is because the static elaboration opens out to N
register assignments, to the same register, semantically at 'the same
instant', which is not possible in the hardware.

>----------------
>    In mkTop module at Top.bsv there are two non-conditioned rules:
>    rl_feed and rl_drain. Why the first rule will be
>    deterministically executed before the second one?

Since we're producing clocked digital hardware, time progresses in
units of clock cycles.

A rule can only fire when it's own condition (mentioned in the rule
itself), and all conditions on the methods it invokes are true.

rl_feed invokes method 'mm.start', whose condition '(!done &&
!initialized)' is true at cycle 0, and so it can fire. rl_drain
invokes method 'mm.result' whose condition ('done') is not true at
cycle 0, and so it cannot fire.

On cycle 1, 'done' becomes true, reversing the enablement situation
for the rules, so 'rl_done' can fire but 'rl_start' cannot.

The $finish() in 'rl_done' stops the program at cycle 1.  Note:
'$finish' is a 'simulation-only' concept; there is no corresponding
concept in hardware (similarly, $write and $display only exists in
simulation).  Without $finish, the 'done' rule would fire repeatedly,
forever.

>----------------
I have attached a few more variants of the code, for comparison.

- In Top and Top2.bsv, the whole matrix multiplication happens in 1
    clock.  Thus, all multipliers and adders (for the inner-products)
    must be instantiated, giving us N^3 multipliers and N^3 adders.
    This will be expensive in hardware for larger N.

- In Top3.bsv, we perform only 1 inner-product at a time, so it will
    have only N multipliers and N adders.  Of course, this will take
    N^2 cycles to complete.

- In Top4.bsv, even the inner-product is done sequentially, so it will
    have just one multiplier and one adder.  Of course, this will take
    N^3 cycles to complete.

- In Top5.bav, we have a variation on Top4, were each inner-product is
    done sequentially (needing 1 multiplier and 1 adder), but we
    replicate it 'P' times, so we have a total of P multipliers and P
    adders.

All of this illustrates that we can design different 'micro-
architectures'--choice of state elements (registers) and the circuits
between them-- trading off space (area of circuits) and time (how long
it takes to compute).

>----------------
Comment about interface methods and scaling

Currently the 'start' and 'result' methods carry entire matrix values
(N^2 values each) in a single cycle; this will not be good hardware
for large N.  One can have a variation where the 'start' and 'done'
methods carry smaller chunks of the matrix, and are called repeatedly
(along with index arguments identifying the chunks) to deliver the
full matrix.

More generally, as N grows large:

- One probably wants to shift to a memory-to-memory design.  Using
    O(N^2) registers for the input and output matrices, where N has to
    be chosen for the worst case, will be expensive hardware.

- Instead of the classical triply-nested loop algorithm, on may want
    to shift to a 'SIMD' algorithm, and also exploit 'blocking' (i.e.,
    work on sub-blocks of matrices at a time, instead of individual
    elements.

>----------------
Note: tutorial materials for BSV are available at:
    https://github.com/BSVLang/Main
We intend to move these to, but have not yet done so.
    https://github.com/B-Lang-org

[More_Variants.tar.gz](https://github.com/B-Lang-org/bsc/files/5007998/More_Variants.tar.gz)

ssmolov, 2020-08-03:
@rsnikhil again thank you!

closed 2020-08-03 by ssmolov