B-Lang-org/bsc #226: bsc doesn't warn on non-exhaustive or overlapping pattern matches

bpfoley, 2020-07-31:
We would expect this to produce several warnings, like GHC does.

```
package ExhaustiveTest where

-- fn1 is defined duplicate and overlapping pattern matches
fn1 :: Maybe Integer -> Integer
fn1 (Valid _) = 42
fn1 (Valid 17) = 42 -- More specific than Valid _ and appears after it, so never matches. Should warn
fn1 Invalid = 43
fn1 Invalid = 44 -- duplicate pattern, but with different result. Should warn.

-- fn2's case expr contains duplicate pattern matches
fn2 :: Maybe Integer -> Integer
fn2 x = case x of
         (Valid _) -> 42
         Invalid   -> 43
         Invalid   -> 44 -- duplicate defn with different result. Should warn.

-- fn3's case expr does not exhaustively match all alternates.
fn3 :: Maybe Integer -> Integer
fn3 x = case x of
         Invalid -> 44 -- Non exhaustive. Should warn.
```

I appreciate that this gets trickier if the case includes guards (ie case <expr> of; <pattern> when <cond> -> <expr>)), but it would be good to have (optional?) compiler warnings for at least the more obviously wrong non-exhaustive and overlapping cases.

quark17, 2024-11-02:
I note that this exists in Bluespec Inc's internal bug database as issue 584 (add a warning when a default don't-care arm is used by a non-imperative case expression).  A comment on 584 says that this may be doable now that `PrimUninitialized` exists -- which I assume means that we can create an uninitialized value that is tagged as being the default of a case-expression, and emit a warning if it gets used in elaboration, however I'm unsure how to do that without giving false warnings (over-eager evaluation) or missing warnings due to optimization of the don't-care with surrounding code.  A final comment suggests that detecting the completeness of the arms may be the way to go, and points to another Bluespec Inc internal issue 568, which is more about making sure that BSC notices that a case is complete so that it can generate a Verilog enable signal that is constant 1 (rather than `x==0 || x==1 || x==2` when there is only three options, say, which could be worked around by adding a fourth dummy enum -- I'm not sure if that code-gen issue still exists, as it was from 2004).

I also note that function clauses and case statements may be handled separately, so there may be two halves to this issue.

krame505, 2025-06-25:
I may be tackling this issue soon.  

To clarify @quark17's comment above, the suggested approach is just to add a purely syntactic analysis on the CSyntax, and not at elaboration time by looking for `PrimUninitialized`?  

At what stage should this analysis take place?  It seems that case is desugared into the application of a function with multiple clauses during type checking.  I suppose that one option would be to just perform the analysis during the simplify stage after this desugaring, but I'm not sure how hard it would be to provide good error messages for desugared cases in doing so.  

Also, am I recalling correctly that sometimes we use incomplete patterns as a shorthand where we wish there to be a "don't-care" value?  If that is the case, would we want incomplete match warnings to be enabled by default?

quark17, 2025-07-06:
It would be awesome if you can tackle this!

I honestly don't know the right answers.  There is the question of whether you want to report (1) completeness of the case arm expressions in isolation or (2) completeness knowing the context (other conditionals up to and including the rule condition).  The former could maybe be done in ISyntax at the point when the case is evaluated; the latter might require leaving some breadcrumbs in the evaluation monad so that BSC can come back and decide completeness after elaboration has finished (or at least when the rule elaboration is finished and the implicit condition is finally known).  I'm not sure if this can be done in the CSyntax, because you may need to evaluate the arm expressions, to really know what they are, so it at least has to be done during elaboration, if not later.

There may also be a difference between those case constructs that return values (pure value or ActionValue) and those that are pure Action; for Action, the default is chosen as `noAction`, whereas values actually produce a don't-care.  I guess that module monads are another situation where the default is somewhat benign (an empty module).  In BSV, people do use if-statements without else-clauses to conditionally instantiate modules in module monads or to conditionally perform actions in Action (ActionValue) monads.  I don't know how many people might use incomplete BSV case-statements in a similar way -- which I think is what you were asking about.  That would certainly only be for Action or Module situations (or maybe monads in general?), but I think it is entirely reasonable to warn by default on these and require users to explicitly state that they know it's an incomplete case (if nothing else, by adding a default clause with `noAction` or `return`).  But this also brings up that BSV and BH may likely be handled differently.  BSV has a syntactic `default` clause, which can immediately indicate that the case-statement is complete (and no need to analyze it); but I guess BH can also check for underscore or a variable name as a similar indicator.