B-Lang-org/bsc #230: Replace -unsafe-always-ready with -demote-errors G0006

bpfoley, 2020-08-03:
Most bluespec errors can be demoted to warnings using the `-demote-errors` flag. One exception to this is G0006, which produces an error like the following

```
bsc -demote-errors G0006 -verilog RespectAlwaysReady.bs
Warning: Command line: (S0094)
  Cannot demote the following errors:
    G0006
Error: "RespectAlwaysReady.bs", line 33, column 0: (G0006)
  Method(s) not always ready: _read
```

Instead you have to use `-unsafe-always-ready`

```
bsc -unsafe-always-ready  -verilog RespectAlwaysReady.bs
Warning: "RespectAlwaysReady.bs", line 33, column 0: (G0006)
  Method(s) not always ready: _read
```

We should allow `-demote-errors G0006` and make `-unsafe-always-ready` an alias for it.

quark17, 2020-08-17:
In some cases, BSC can't figure out that a method is always ready, because it lacks some knowledge that we as designers know.  For example, we might know that an input will never equal some value, or if two signals will always have the same (or different) values.  I'd feel safer if the user could specify that knowledge in the source, so that -- instead of entirely disabling the always-ready check -- BSC could still do the SMT analysis but using the additional properties provided by the user (in case BSC can prove that the method still isn't always ready, even with that additional info).

quark17, 2020-08-17:
It might also be interesting to generate a simulation-time check of this assertion in the Bluesim/Verilog, if the user wants.