B-Lang-org/bsc #236: Add method args for non-firing methods and and enable signals to VCD dumps

bpfoley, 2020-08-04:
Bluesim VCD dumps only include signals and method calls that occur due to methods that fire on a given cycle. If a method doesn't fire, nothing is logged.

In many cases it would be useful to have a detailed dump option that would
* dump the enable signal (as the verilog sims do)
* dump the arguments to each method even if it doesn't fire

quark17, 2020-08-18:
For methods that aren't called, the enable signal should certainly be dumped as 0.  If you say that Bluesim isn't dumping that, then that should be fixed.

If a method isn't called, the value of the signal driving the Verilog port is not guaranteed to have any meaning.  The generated Verilog is allowed to be optimized in any way, as long as the values are correct when the method is called, but it's free to be anything when the method is not called.  If there are multiple calls to that method in the parent module, the signal is usually the result of a priority mux choosing among the enabled rules that call that method -- implemented as a case-statement.  If none of the calling rules are firing, then the result of that mux is undefined.  That mux is choosing among values from each caller but, again, for any individual caller, if that rule is not firing then the logic for its method-call arguments is not guaranteed to produce a sensible result.  For example, the logic to produce those arguments might use the output of other method calls -- but the rule may not be firing because there are conflicts that prevent using those methods ... and conflicts often represent the fact that the output one method is no longer legal once another method has been called ... so those values (those inputs to the logic computing the method arguments) may be bogus.

Can you say more about what kind of debugging you're trying to do, that you want to see the arguments?  Maybe there's a better way to implement this?  (In Bluesim we have the opportunity to do so much more than is possible in generated Verilog.)

If you're just interested in making Bluesim and Verilog give the same value in these cases (when the value is undetermined), that would still require some work.  There's a "backend split" in BSC where the Bluesim and Verilog backends diverge, and can start optimizing the design in different ways.  The default behavior of BSC is to try to make both backends produce matching designs, for the visible behavior.  If a don't-care value ever appears in the visible behavior, BSC picks a value for it prior to the backend split, and hinders the backends to use that same value.  This limits the amount of optimizing that the Verilog backend can do, although there is a flag that users can give to enable the Verilog backend to diverge (and thereby perhaps achieve more optimized hardware).  In theory, another way to handle it would have been to allow the Verilog backend to always optimize, and make Bluesim match by first running the Verilog backend to find out what values are picked and then running the Bluesim backend knowing those choices (and making Bluesim match them).  That seems like a lot of work, but in theory it's possible.  But someday BSC might have more backends (implementing the design in other other HDLs or in other native simulators or other synthesis methods or technologies) and it's less clear how we'd want to keep all of these in sync.

This default behavior of BSC (to make Bluesim and Verilog match) doesn't apply to non-visible behavior, like the arguments to methods that aren't called.  If we wanted to make sure that both Verilog and Bluesim generated the same values for these undetermined signals, we have two options: either BSC would have to further hinder the Verilog backend from optimizing these signals (which I think would be significant) or BSC would have to run the Verilog backend first (to get the expressions) before then running the Bluesim backend.

quark17, 2020-08-18:
Oh, wait, is it possible that you didn't accurately submit @nanavati 's request?  I looked at his original comments and I think he was just asking for the arguments to have the value X in the VCD dump, when the method is not called.  That is entirely reasonable: in cycles when a method is not called, the VCD should have 0 for the enable and X for any arguments, and X for any result.