B-Lang-org/bsc #238: Add registers that support simulation initial values

bpfoley, 2020-08-04:
We're simulating a design inside a simulator, which has a notion of start of time; one of the things that we're simulating in this world is reset signals; but the simulator may also have meta-level constructs (like assertions) that need their state initialized at the start of time, separate from whatever is happening with reset signals within the simulation.

One possible implementation of this is a way to declare a simulation state element?  Such as:
```
  module mkSimReg (ty init_val) (Reg#(ty));
```
Which has an initial block that sets the initial value, rather than `AAAA`?  And perhaps even has a "translate_off" attribute on the register, to ensure that it's simulation only?

A sketch of how this would be used
```
haveWeReset <- mkSimReg False
haveWeChecked <- mkRegA False
inReset <- isResetAssertedDirect

let exitingReset = not inReset && haveWeReset && not haveWeChecked
when (exitingReset === True) ==> -- Use === to handle X resets.
  do_check;
  haveWeChecked := True
  -- possibly also haveWeReset := False if we want to be able to check future (soft) resets

when inReset ==> haveWeReset := True
```

There are side questions that come to mind.

For example, supporting reset signals as well as initial values would be useful (I think I could collapse haveWeChecked and haveWeReset above if I could combine an asynchronous reset with a custom initial value). I could even imagine supporting an initial value different from the reset value (but that's also straightforward to build out of a pile of registers, so maybe not).

I'm not sure if I'd want a `translate_off `on the register itself (beyond the standard macro/translate_off on the initial block). I can easily imagine wanting a real register with a special initial value for some reason (Verilog people do this, after all). What I probably really want is that if bsc can detect that a register is only used inside `translate_off blocks` to `translate_off` it too. I did a mini version of that with the foreign function result regs bsc inserts. Doing that with user registers is trickier, though. It might be simpler to stick with our current solution of putting a tag in the register name so that downstream tools know that this is a thing they can ignore.

It's possible that the right abstraction is `mkInitialReg` which takes an initial value and a reset value (and `mkInitialRegI` which just takes a reset value, and `mkInitialRegU` which just takes an initial value).

quark17, 2020-08-18:
There's certainly been interest in these kinds of ideas before, and probably some discussion.  I'll have to search through Bluespec's records to see if there was any thoughts or results worth attaching here.

wlott, 2020-10-31:
I've found myself wishing for the same thing, but for a different purpose.  I'm targeting FPGAs and there are cases where I want to explicitly control the just-out-of-configuration state of a register independently from the value assigned at "reset".  And as this is not for simulation, all the `translate_off` stuff gets in the way.