B-Lang-org/bsc #243: Literal '1 not accepted in case match pattern

arjenroodselaar, 2020-08-05:
I am trying to use the literal '1 as a case match pattern but am presented with the following error:
```
Error: "Schmitt.bsv", line 17, column 17: (P0097)
  Numeric literals with X or Z not supported: `'1'
```

For completeness a minimal-ish repro:
```
package Schmitt;

interface Schmitt #(numeric type sz);
    method ActionValue#(Bit#(1)) filter(Bit#(1) x);
endinterface

module mkSchmitt (Foo#(sz))
        provisos (Add#(msb, 1, TLog#(sz)));
    Reg#(Bit#(sz)) xs <- mkReg('1);
    Reg#(Bit#(1)) x <- mkReg(1'b1);

    method filter(x);
        let _xs = Bit#(Bit#(sz))'{xs[msb:1], x};
        let _x =
            case (_xs) matches
                '0: return 1'b0;
                '1: return 1'b1;
                default: return x;
            endcase;

        xs <= _xs;
        x <= _x;
        return _x;
    endmethod
endmodule : mkSchmitt

endpackage : Schmitt
```

The use case above where this came up is in a schmitt trigger like filter with a polymorphic sampling window size. In this case I want to match the "all zeros" and "all ones" cases, and according to paragraph 2.3.1 in the Reference Guide these literals would do just that. I can use the '1 literal in other places, as demonstrated by the initialization of "xs". Am I doing this wrong? And if so, what's the appropriate way to implement this functionality?

bpfoley, 2020-08-11:
The BSV parser explicitly disallows `'1` and `'0` in pattern literals at the moment. You can see the code that forbids it in [CVParser.lhs:2723](https://github.com/B-Lang-org/bsc/blob/master/src/comp/Parser/BSV/CVParser.lhs#L2723).

A workaround for now is to use pattern guards.

```
import Vector::*

      rule do_update;
          let _xs = Bit#(sz)'{xs[valueOf(msb):1], _x0};
          xs <= _xs;
          Vector#(sz,Bit#(1)) vz = replicate(0);
          Vector#(sz,Bit#(1)) vo = replicate(1);
          let zeros = pack(vz);
          let ones  = pack(vo);
          x <=
             case (_xs) matches
                 .* &&& (_xs == zeros): return 1'b0;
                 .* &&& (_xs == ones):  return 1'b1;
                 default: return x;
              endcase;
      endrule
```
```

arjenroodselaar, 2020-08-11:
Thank you for the workaround, that indeed does the trick and the generated Verilog looks like what I would expect the compiler to do with the literals. @quark17, purely my curiosity; any idea why the '0 literal is accepted (at least by the parser) but the '1 is not?

quark17, 2020-08-11:
Your original code works if you remove the `matches` keyword.  BSV has two kinds of case-statements, matching and non-matching, indicated by the presence or absence of the `matches` keyword.  Since you're not pattern matching, you can use the non-matching case-statement (which is the kind original found in Verilog/SystemVerilog) which accepts any expression for the case arms (and does an equality check to find the first one that is equal).

I will have to look into why the pattern-matching case-statement is behaving this way.  I've been slowly working through all the open tickets, and I apologize that I hadn't gotten to this one yet.

arjenroodselaar, 2020-08-11:
Oh wow.. I completely missed there are two different case-statements, thank you for the pointer @quark17! For anyone following after me; the Reference Guide explains the differences in Sections 8.6 and 10. And yes, using the none matched case works for my use case. And please don't apologize, it's all good.