B-Lang-org/bsc #261: Force Rules to Fire In Conjunction (Forward Implicit Conditions)

dz333, 2020-08-28:
As far as I can tell, there's no easy way to forward the implicit conditions for one rule to another.
I'd like to write down rules A, B and C such that:

A fires => C fires
and
B fires => C fires

To achieve this I'd like to ensure that the rule conditions for A and B  both imply the rule conditions for C; however the rule conditions for C contain implicit conditions (e.g., b/c it writes to a BRAMServer).

Additionally, A and B both write to a Wire that C reads, so I can't force C to execute first and write some Wire value that indicates 'did fire'.

I would appreciate a feature (or workaround) that told the compiler to add the implicit conditions for rule C to rules A and B.
Any suggestions?

-------------
N.B. I'm doing this, rather than copying the contents of C into rules A and B, to avoid the overhead of duplicating the combinational logic in C. If that isn't actually a concern, then I could do that instead. I can provide small code examples if necessary for clarity.

quark17, 2020-09-01:
I suggest writing this as just rules A and B (each containing the code from C).  I would not expect this result in duplicated logic.  BSC will generally share common expressions, and the generated Verilog is written in a state-centric way that also leads to sharing.  I have also found that synthesis tools do a good job of optimizing the Verilog they are given; so extra work by BSC to clean up the output Verilog often hasn't improved the QoR or run time of synthesis.  Do you have an example where the generated Verilog is not doing enough sharing? (or where later Verilog tools are not able to discover that sharing is possible?)

If you have logic that for some reason you need to force to be shared, I would suggest doing that by encapsulating it in a module boundary and instantiating that once -- thereby creating a single, limited resource.  If writing this out explicitly is tedious, then we can consider ways to make it easier to specify in the source.  (There is a "noinline" attribute that can be used on functions, but that actually forces the function to be duplicated each time, rather than shared.)

Rules in BSC are supposed to be things that can be executed one at a time and still be functionally correct, but that we can improve the performance by executing more together in a clock cycle.  If you have a design where rule C has to execute with rule A, for correctness, that seems to me to be working against the abstraction, and I would want to avoid writing a design like that.

If you did need to hand off data from one rule to another, I would prefer to use a state element that can bypass the value to C, when C fires in the same cycle as A, but which registers the value (for a later clock cycle) when C can't fire in this cycle.  (See, for example, mkBypassFIFO.)  That at least guarantees that the rules are functionally correct, regardless of how they are scheduled.  Of course, if BSC has generated a design in which C does always fire in the same clock as A, this design would have an unnecssary state element for registering the value (which is never needed because C always takes the value).  In that case, we could look at improving BSC to detect (or confirm) that case and to only generate wires.  (Though, again, synthesis tools might also be able to discover that the state element is never used.)

dz333, 2020-09-01:
Thanks for the detailed response!
I haven't explicitly run into a case where synthesis tools haven't been able to optimize; however I'm writing a tool that generates BSV as output and therefore I'm trying to consider what cases might end up causing issues in the future. However, since I'm going to be generating nearly identical expressions in both rules, your answer assures me that it's not likely to cause problems.

If they do end up being issues, then the bypass option (which I'd also been considering) seems promising.

-----------------------------------------------------------------------------

I will say that the whole need for multiple rules at all (originally, I had wanted to generate rules A, B and C as a single rule) is a result of the imprecision of Bluespec's implicit condition generation. Rules A and B also do almost the same thing, except they have mutually exclusive conditions and read inputs from different FIFOs.

As an example:
```
rule ABC;
  let cond = cq.first;
  cq.deq();
  let x;
  if (cond)
    begin
      x = q1.first;
      q1.deq();
    end
  else
    begin
      x = q2.first;
      q2.deq();
    end
  //compute on x and write to output state
endrule
```
As I said, I originally wanted to generate a single rule of this form, but then the implicit condition for it always contains `q1.notEmpty && q2.notEmpty` and therefore will only execute when all input FIFOs have data even though it can run more often than that.

Even if I re-write this to be able to use *split*, it interestingly still requires all of the input FIFOs to be non empty on both of the generated rules. This, in particular, seems like a lack of precision that Bluespec can definitely solve.

```
rule ABCSplit;
 let cond = cq.first;
  cq.deq();
  (*split*)
  if (cond)
    begin
      let x = q1.first;
      q1.deq();
      //compute on x and write to output state
    end
  else
    begin
      let x = q2.first;
      q2.deq();
      //compute on x and write to output state
    end
endrule
```

dz333, 2020-09-02:
Ah I realize this problem is solved by the `-aggressive-conditions` flag. I didn't realize that flag existed and recently stumbled across it.

closed 2020-09-02 by dz333

quark17, 2020-09-02:
Yes.  In fact, I thought we'd made this flag be ON by default, but I see that it's still not.  (I think the only reason not to have it on by default is the lack of a pragma for turning it off on an individual rule or conditional, in the rare case where the additional logic isn't wanted.)  I'll file an issue for this enhancement.