B-Lang-org/bsc #266: Assertions for function parameters

kmeinhar, 2020-09-30:
I was not able to find a forum for general questions for the bluespec language. If there is one please point me to it.

I have a function and I want to assert that the input parameters are valid.

In the following example there are there valid inputs to the function `get_value`: 1, 2 and 3.
How can I assert at simulation or compile time that the input is a valid UInt?
```
function UInt#(8) get_value (UInt#(8) input);
  case (input)
    1: return 4;
    2: return 2;
    3: return 9;
    default: 
      // This should never happen!
      return 0;
  endcase
endfunction
```

It is not possible to insert something like `$display` or `$fatal` into the default case since this is only allowed in an action and not in a function.

I was not able to find a compiler attribute that checks the input of this function like this:
```
(* assert = "input == 1 || input == 2 || input == 3" *)
function UInt#(8) get_value (UInt#(8) input);
  ...
endfunction
```

arjenroodselaar, 2020-11-30:
I would love a forum for general questions as well. There used to be something hosted on the Bluespec.com domain, but that seems to be gone. There's a Bluespec category on SO, but it seems pretty dormant as well.

Anyway, to your question. I believe if you use this function in a static context I think you can ask the compiler to halt if this returns 0 using the error(...) method. Search for "Compile-time Messages" in the Reference Guide. But this is really limited and I've only used it to test constructor parameters for modules.

What you're after is what's referred to as "refinement types" in for example F*. To implement this a SAT solver would be required to test the (compile time) constraints of a design. This is definitely implemented today. If you're interested in this you may want to take a peek at Kami (https://plv.csail.mit.edu/kami/). But like anything from Chlipala's team this is not for the faint of heart :)

quark17, 2020-11-30:
Would a list on groups.io work as a forum?  I just need to settle on what lists and what names and then I'll create them.  I've opened #281 for discussing that.

kmeinhar, 2020-12-01:
Thanks @arjenroodselaar for bringing back life into this discussion! 

The project you linked sounds really interesting and I hope something like this will eventually find it's way into the Bluespec language. But for now this is a bit out of scope for my purpose.

I would close this issue now since the groups.io forum would be a better fit for those questions. Thank you @quark17 !

closed 2020-12-01 by kmeinhar

quark17, 2020-12-23:
I meant to respond to your question when you posted it, but it fell through the cracks -- I apologize!

It sounds like you're asking about errors that are checked while the hardware design is running (that I usually refer to as "dynamic" error checking).  Though, if a dynamic error can be determined statically, BSC could certainly report that during compile time.

Dynamic error checking is certainly a place where BSC could be doing more.  Bluespec Inc maintains a private issue database and there are a number of suggestions there on this topic.  I'm hoping to migrate these entries over to GitHub, but I apologize that they aren't here yet.

It's been suggested that BSC could automatically insert dynamic checks for situations exactly like yours: failing to match any arm of a case statement, or indexing out of range of a list or vector.  There are some ways that one can manually implement this checking.  It's worthing noting, though, that there are some limitations to what can be checked.

As you say, you could insert system tasks to do the checking if the function were returning an ActionValue, not just a value.  A slightly more flexible approach would be to have the function return a pair: the result plus additional assertions to be checked (for example a Bool that must be true and a String error message if not).  Then, the caller can use that to construct an assertion check however it wants.  You could even hide this machinery inside of a Monad type, so that the plumbing of assertions is less explicit.  The checks can be collected up and put into a rule (or rules) that is always executing, just to check the dynamic assertions.

The problem is that HW logic always exists, even when it's not being used.  So checking the inputs to the case-statement is only valid when the result of the logic is being used; at other times, the wires are allowed to have any values.  We don't want the error message to trigger at those other times.

This is easy to see if the logic is called within a conditional in a rule:
```
rule r (c1);
   if (c2)
     rg <= get_value(...);
   ...
endfule
```
The dynamic check should only happen when `c2` is true.  Further, it should only happen when the rule is executing, so it should only happen when `c1` is true.  Things get a little trickier, though, when the function is used inside a rule condition:
```
rule r (c1 && get_value(...));
```
If the scheduler had picked another rule`r2` to execute, that may cause the wires for rule `r`'s condition to no longer be valid, and the scheduler will choose to disable rule `r` when `r2` fires, to account for this.  (For example, a FIFO module has a `notFull` output, which might be true at the start of a clock cycle, but rule `r2` enqueues into the FIFO; BSC knows that after picking `r2` to fire, that the `notFull` output can no longer be trusted and therefore any rule that uses this value is now blocked until the next clock cycle.)  Any dynamic check of the case-statement in `get_value` would only be valid when rule `r` is not blocked.

There's also the question of how we want to deal with `&&` and `||` expressions, and whether to check the assertions within a term if the term's value is ultimately not needed.

The check would also only be valid when the rules that use the function are not in reset.

For these reasons, it would be helpful for BSC to automatically collect the conditions for dynamic assertions.  Although there is some amount that you can do manually.

Going back to the original statement about not being able to put `$display` in a pure function: there is actually a way to do it.  If you're familiar with Haskell/GHC, there is a function `unsafePerformIO` that will let you perform I/O inside a pure function.  It's unsafe because that I/O is not predictably sequenced with other I/O, so you should only use it in situations where that's OK.  BSC has a hidden primitive that it uses to add module elements when not in a module block (such as in a pure function like this).  It's called `primBuildModule` and it can be used to create an unsafe module function:
```
unsafePerformModule :: Module a -> a
unsafePerformModule = primBuildModule (primGetName _) noClock noReset
```
This function could be used in the default branch of your case-statement to create a rule that calls `$display` when the input is not valid.  However, you wouldn't have the conditions for when the function is called, so the rule might report false errors.

So it's probably best left to BSC to automate.  BSC does currently collect up the conditions of some expressions -- and it uses a SAT solver (or ad hoc boolean solvers) to prune unreachable arms -- so BSC could certainly be extended to collect up assertions (noting all the conditions), report any that fail statically, and turn any remaining assertions into dynamic checks to put into the generated module.