B-Lang-org/bsc #271: FloatingPoint::mkFloatingPointDivider deadlocks when dividing zero by something.

wlott, 2020-10-31:
FloatingPoint::mkFloatingPointDivider deadlocks when dividing zero by something.  Looking at the code and the VCD file, it appears that `s1_stage` detects the zero and sets `out` to a valid value.  `s2_stage` checks for the valid `out` and forwards it without triggering the integer divider.  But then `s3_stage` never fires for reasons I've been unable to unearth.  I'm new enough to Bluespec SystemVerilog that my ability to debug runs out there.  

```
package DivTB;

import FloatingPoint::*;
import Divide::*;
import ClientServer::*;
import GetPut::*;

(* synthesize *)
module mkDivTB();

   Server#(Tuple2#(UInt#(56),UInt#(28)),Tuple2#(UInt#(28),UInt#(28))) int_divider <- mkDivider(1);

   Server#(Tuple3#(Float, Float, RoundMode), Tuple2#(Float,Exception)) fp_divider <- mkFloatingPointDivider(int_divider);

   Reg#(Bool) issued <- mkReg(False);

   rule rl_issue(!issued);
      // works:
      //fp_divider.request.put(tuple3(1, 1, defaultValue));
      // deadlocks:
      fp_divider.request.put(tuple3(0, 1, defaultValue));
      issued <= True;
   endrule

   rule rl_retire;
      let r <- fp_divider.response.get();
      $display("result:", fshow(r));
      $finish;
   endrule

   Reg#(UInt#(12)) watchdog <- mkReg(0);
   rule rl_watchdog;
      watchdog <= watchdog + 1;
      if (watchdog == maxBound) begin
         $display("watchdog hit");
         $finish;
      end
   endrule

endmodule

endpackage
```

quark17, 2022-01-25:
Sorry that there has been no response on this.  I wonder if the changes in #410 have improved things.

quark17, 2022-01-27:
I had a look and the problem is that the flag `-aggressive-conditions` need to be given.

I notice that the `mkFloatingPointDivider` module has an `options` pragma in the source code:
```
(* options = "-aggressive-conditions" *)
```
but this pragma is ignored when the module is not a synthesis boundary.

The way the module is written, stage2 conditionally enqueues to the integer divider and stage3 conditionally takes a response.  Taking a response from the integer divider carries an implicit condition that the divider has a response ready to give.  Without aggressive implicit condition lifting, the stage3 rule will always require the divider to have a response; so in those cases when no request was enqueued, the stage will never be able to fire.

The `mkFloatingPointDivider` could probably be re-written to not require aggressive conditions.  Or the ignored `options` pragma could be fixed in BSC somehow, so that the module can actually add to the options, or at least check if the option was used and error if not.  But otherwise, users of this module will need to know to add `-aggressive-conditions` to the flags when compiling.

PR #448 (2022-02-02): Fix FloatingPoint modules to not require aggressive conditions
This fixes #271.

The divide and square root modules, if compiled with conservative implicit condition lifting (the default), would deadlock when given invalid inputs: the first stage would detect the error and determine the result; such that the second stage didn't need to dispatch a request to the integer divider/squarerooter; but the third stage would expect a response, as part of its implicit conditions.  The fix here is to manually split the third stage into two rules: one expecting a response and one not.  It would have been  [... cut]
closed 2022-02-02 by quark17