B-Lang-org/bsc #281: Create mailing lists for BSC users and/or developers

quark17, 2020-11-30:
What lists do we need and what should they be called?  I think we need:

1. An anything-goes list for BSC questions
2. A low-traffic announcement list
3. A list for BSC developers

I'm proposing to create lists at groups.io and call them:

1. `bsc-discuss`
2. `b-lang-announce`
3. `bsc-develop` (or `bsc-developers`?)

The second list could be called `bsc-announce`, but BSC isn't the only repo under the B-Lang org's umbrella -- we'd need a list for announcing the new BDW repo, for example -- and there won't be enough traffic to justify having a bsc-only announce list, so I figure just an overall B-Lang announce list will do.

We could have a separate list for beginner questions (`bsc-beginners`?), to avoid cluttering `bsc-discuss`, but I think one list is probably ok for now?

If you have preferences for names or other types of lists, please comment here.  I hope to create these lists soon, and add a mention of them to the README -- perhaps under a new "Community" heading, where links to the IRC channel etc can be included as well.

nanavati, 2020-12-01:
These lists sound like a good starting point.

I'd call the bsc developer list "bsc-dev" because that's the convention I
expect to see from open-source projects. bsc is a bit of a misnomer for the
discussion list, but I'm not sure that, for example, bluespec-discuss would
be an improvement.

quark17, 2020-12-01:
Ah, you're right, that questions about BSV/BH are not specific to BSC.  However, there are also questions about using BSC, like flags or specific implementation issues, etc.  But I don't think we need separate lists yet.  How about `b-lang-discuss` for the name? as a place to discuss anything related to B-Lang, BSV/BH, or BSC that's not covered by another list.

arjenroodselaar, 2020-12-01:
```b-lang-announce```, ```b-lang-discuss``` and ```bsc-dev``` sounds reasonable to me. Until traffic of lists becomes a problem I would try and avoid splintering so as to grow a bit of a community. But this split makes sense and should be sufficient for a long time. Thank you for setting this up!

kmeinhar, 2020-12-01:
Dedicated groups for those topics sounds great to me!

Maybe we could also use group.io or another tool for a community build Wiki? I know there is http://wiki.bluespec.com/ but I don't think it is possible to suggest changes currently. It also looks stale, for example has it some dead links.

quark17, 2020-12-09:
GitHub repos can now add a Discussions tab.  It's still in beta, but now publicly available.  I have not used it.  Here is a blurb from the announcement in May:

> Discussions live in your project repository, so they’re accessible where your community is already working together. Their threaded format makes it easy to start, respond to, and organize unstructured conversations. Questions can be marked as answered, so over time a community’s knowledge base grows naturally. And because discussions aren’t closed the way issues are, they can easily serve as a place for maintaining FAQs and other collaborative documentation. We recognize that community discussion is as much a part of development as coding, so discussion contributions appear in users’ contribution graphs.

I can turn on this feature and we can try it out.

arjenroodselaar, 2020-12-09:
I'd be happy with that. I already have GH notifications etc. taken care of, so it should be pretty seamless for me.

quark17, 2020-12-22:
I've turned on the Discussions tab, created the three mailing lists, and updated the README (commit 7d25cde).  Please give them a try!  Here are the lists/groups, for convenience:

* https://groups.io/g/b-lang-announce
* https://groups.io/g/b-lang-discuss
* https://groups.io/g/bsc-dev

closed 2020-12-23 by quark17