B-Lang-org/bsc #285: Coordinate 'bsc' PRs with 'bsc-testsuite' PRs

quark17, 2020-12-22:
When someone submits a PR to `bsc`, they generally also submit a PR to `bsc-testsuite` to reflect changes in behavior.  It would be nice if the automatic CI for the `bsc` PR could notice that there's an associated `bsc-testsuite` branch and run CI on that branch.  One way to do this would be if the two PRs are made with the same branch name, then the `bsc` CI could look for a branch on `bsc-testsuite` with the same name.

Would we only run the CI on the new testsuite branch? or run on both the old and new, to give reviewers more info?

It's also worth noting that sometimes there's two PRs to the testsuite: a first PR to add new tests to the testsuite (but expected to fail with the current BSC) and the a second PR that updates the old and new tests to account for the change to BSC.  If we're developing a convention for how PRs are submitted and coordinated, we'll want to keep this case in mind.

thoughtpolice, 2021-01-16:
I know this might not be the point of this issue, but speaking from experience working on the Glasgow Haskell Compiler, having the testsuite sitting outside of the compiler repository was a clear error at a certain point in the project lifecycle. It caused countless issues with human and automaton error due to synchronization problems, and had almost no advantages &mdash; the reason it was separate was a quirk of history; originally there was one testsuite used among several Haskell compilers, so having things separate made sense. Over time, other compilers fell to the wayside, and it made no sense to keep it separate from GHC when we were the only developers. We later merged the repositories, and this was a very good decision in hindsight that should have been done sooner.

It seems the Bluespec testsuite was released after the main compiler due to some minor hurdles resulting in a small source code delay. It may also be the case that the bsc testsuite was a historically separate codebase from `bsc` when it was proprietary, but I don't think either of these reasons should stop the new open source bsc from integrating the two, and I think this is a good idea.

So, put another way: if we put the testsuite inside the repository, all these problems go away. Can't we just do that?

quark17, 2021-01-20:
Can you say more about the disadvantages when they were in separate repos? or the advantages after they were merged?

BSC's testsuite started out in the same repo (back when it was an SVN repo).  What I miss from that time is, when I'm looking at the commit history for lines of code, I can see what testsuite files were added or adjusted at that time (usually in the same commit, or nearby in the log).  However, it was also a pain to have to checkout the entire testsuite, when all you wanted the BSC source.  (That was in SVN, and on a network filesystem and hardware from years ago, so maybe this isn't a concern now. And the public testsuite isn't as many files, but it's still a ton of tiny files.)  When Bluespec transitioned its SVN repos to GIT, there was a sense that it was better to have lots of separate repos, since GIT made that lightweight and you could could connect them as submodules etc.  So at that time the testsuite was separated into its own repo.  We've been operating like that for some time, and the only thing I miss is the ease of finding related tests in the testsuite (for a block of source code), although I feel that can be improved with good use of commit messages and/or submodules.  But maybe it will be harder to coordinate now, with more developers (and more dispersed developers).  And maybe I'm not a good metric -- maybe people less familiar with the two repos might still find benefit from having them together.

I'm open to discussing the possibility of moving the testsuite back in, but I think I'd prefer to keep them separate.  I'd like to understand if having the testsuite as a submodule wouldn't solve the main issues.  Obviously, that only associates commits in one direction; if someone is looking at code in the testsuite and wants to see what code in bsc is related to it, they'd only have the commit messages to go by -- but I don't know how often that direction is a concern.

czeck, 2021-01-21:
I would agree with Austin and prefer having 1 repo containing the code and
testsuite. This will guarantee that they stay in sync and commits and PRs
can be easily tracked.
Ed.

thoughtpolice, 2021-01-24:
So I guess the most simple way to state it is that the cost of adding external dependencies at the version control level (external repos, submodules, etc) is very non-linear in terms of complexity and effort, and it tends to complicates many workflows people are already used to when interacting with the repository. If you split the testsuite to another repository, the difficulty of adding a feature with tests &mdash; compared to a single monorepo &mdash; isn't like, marginally harder, or 2x harder. It's closer to 5-x to 10x harder, and anything beyond that trivial case is even further down the road. That's a low bound, I think.

Here's a very simple example of a problem that is enormously complicated with many repositories, but trivial with one: bisecting to find which commit introduced a regression in a testsuite. In the event where you *aren't* using something like submodules, you need to either bite the bullet and be prepared to baby your `git bisect` run as you try to find compatible repository states, or have some kind of complex protocol using git notes, commit message syntax, etc (with automation to support it) that will help you "synchronize" the repositories. You're just recreating submodules. (GHC when it used Darcs was in this territory, and it wasn't great when you needed to revert.)

"But then I could use submodules, wouldn't that help?" And yes, they mostly fix the literal problem of "revision X refers to revision Y of submodule foo", but they come with a whole host of issues of their own, like just committing things *correctly*, every time. I can't count the number of times this exact scenario would occur working on GHC: a change needs to be made, like a new feature. You must have a change in both the main repo and a subrepo. Someone takes on this challenge and wants to commit it. One of the following often happens at this point:

- They push the parent repo change, with a submodule update, but forget to push the change to the submodule. Builds now fail because the submodule ref doesn't exist.
- They push the submodule change, and then push the parent repo change, but forget to update the parent repo to point to the new sub repo version. Builds fail, but can at least be fixed.
- They push the submodule change, and push a parent repo change *including* a submodule update. Except: the submodule commit doesn't exist. In fact it exists nowhere at all. The committer simply had some kind of error when using `git` and didn't catch it before a push, and just messed up the update. This could happen for any number of reasons from "Build system cached too much and didn't re-run, giving them false hope" to "They just ran random git commands repeatedly, trying to fix something they thought was wrong."

We ultimately fixed most of these by inevitably adding scripts in our git servers that would scan commit messages for specific "Update submodule ref: blah" lines, and reject commits without it. This effectively required you to both commit the change *and* sign it off in a 2-step process before proceeding. This reduced the number of errors we had to fix (often a handful a week) to effectively zero, overnight.

---

As an interlude here, I'll also say that Git submodules also just in general aren't very good, even if they fix that one exact issue of tracking an exact revision of some other repository. They make merging, committing, and even updating things more complicated. Merging or back-porting for example can quickly become very annoying because every cherry pick is invalidated, every cherry pick requires now not one, but two picks, as well as a constant amount of effort to amend the submodule pointers, etc. I would estimate something like 95/100 users forget to run `git submodule update` after running `git pull origin master`, as another example, and this kind of thing almost always results in weird anomalies like build failures if you're lucky, or other red herrings like test suite "failures" otherwise.

Hell, the fact they actually *don't* cause any kind of visible UX error if they aren't synchronized is basically the *entire* problem in a nutshell. It just gives people too much rope to hang themselves with, unless they're coupled with automation, but at some point you're just avoiding the real problem.

Again, the cost of these solutions isn't like, "Oh, I'm tracking two repositories, so now it's only a little or 2x as hard as before." It's like 5x as hard and 5x as error prone in practice, in my experience.

---

And after that, we started consolidating many repos into GHC itself to simply reduce the number of times people even had to think about any of this at all. High-traffic repositories are especially good candidates for this, because simply having to do all this often introduces scope for errors, regularly. This alone was huge: just getting rid of all these concerns for 90% of developers was huge.

The testsuite is far too commonly trafficked by developers to warrant anything less than first-class treatment, basically.

Finally, I would also note that I would apply this same standard to most of the other "front matter" that comes with the compiler.

For instance, I would argue the `bsc` manual and library reference (currently located with `.pdf` files in another repo) should 100% be co-located in the repository, along with the testsuite, no submodules involved. The build system should build and check the manual on every commit, just like it were testing the compiler, etc etc. So it's not just the testsuite that should get the first-class treatment: everything that's crucial to `bsc` as a user and developer should live as close as possible, IMHO.

quark17, 2021-01-24:
It's definitely intended that the documentation will exist as source in this repo and be built as part of building everything else.  I'm hoping to get to that shortly (as mentioned in another issue entry about documentation).

I'm willing to try merging the testsuite in.

quark17, 2021-02-05:
Do you have advice on merging two repos?

The bsc-testsuite has some history; it'd be nice to keep that, but also not the end of the world if we didn't.  I'm concerned though: the commit messages may reference PR #s in bsc-testsuite, and those numbers will not be correct when inside "bsc".  Plus, those PRs won't be copied over.  Unless GitHub offers way to merge repos that migrates things?

Also, both repos have a history prior to going open.  Someday we would like to make that pre-history public, but first we'd need to sanitize the commit messages (removing mentions of company names or people, etc).  The possibility exists of pre-pending that history into the GitHub repo.  It would be good if the merge does not further hinder that possibility.  However, if the pre-history could not be prepended and had to be made public as a separate archive, off to the side, that people can browse, that would still be OK.

I'm investigating ways to pull the commits of one repo in as a remote and then merging, perhaps with the "allow-unrelated-histories" flag.  I would first make edits to the testsuite, to move everything to a subdirectory, while still working; so that the merge can happen with no conflicting files.  Then after the merge, I'd unify the CI config files (removing/moving the testsuite CI files and updating the top-level CI files).

But if you have advice on merging, that would be welcome.

thoughtpolice, 2021-02-14:
> The bsc-testsuite has some history; it'd be nice to keep that, but also not the end of the world if we didn't.

There is a way to splice Git repositories together, in fact various ways. A quick Google reveals one: https://stackoverflow.com/a/10548919

You basically want the `git-filter-repo` approach, I believe (we used this approach for GHC, IIRC, but don't quote me on that, it was ~8 years ago...)

> the commit messages may reference PR #s in bsc-testsuite, and those numbers will not be correct when inside "bsc". Plus, those PRs won't be copied over. Unless GitHub offers way to merge repos that migrates things?

GitHub doesn't offer this feature, but since the history is only on the order of 30-ish commits, you could just manually rewrite the messages when merging them. GitHub also allows inter-repository links too, so you could rewrite the commit message "Incorporates test ref: #XYZ" (where #XYZ is in the bsc-testsuite repo, today) into "Incorporates test ref: b-lang-org/bsc-testsuite#XYZ" instead, and the hyperlinking automatically happens. That way you don't have to migrate bugs or anything, and you can archive the `bsc-testsuite` and leave it there.

> Also, both repos have a history prior to going open. Someday we would like to make that pre-history public, but first we'd need to sanitize the commit messages (removing mentions of company names or [...]

It's *probably* better to just go the archival route, honestly. There's a lot of potential scope for error when "re-introducing" history in this manner, and the longer you wait on it, the more difficult it will become. This was probably something better done before the FOSS release, but honestly it just takes insane amounts of time to get right. :(

An archival release would be nice, though, since cross referencing it, etc would still be useful. But I wouldn't try rewriting the existing history to match the proprietary development history.

> I'm investigating ways to pull the commits of one repo in as a remote and then merging, perhaps with the "allow-unrelated-histories" flag. I would first make edits to the testsuite, to move [...]

Yep, that basically sounds like the right approach; like I said that SO answer may provide something a little nicer. You can then go back and clean up some of the history using an interactive rebase.

Alternatively to all of this, the history for bsc-testsuite is only 40 commits. It's easy enough to just recreate this manually by hand with no automation; I suspect it would probably take an hour or two.

quark17, 2021-03-16:
I've merged the `bsc-testsuite` into this one.  The files are in a `testsuite` subdirectory and I've merged the copyright info into BSC's copyright info and I've cleaned up the references between the two README files.  But the subdirectory still has a `.github` directory containing its CI, which will need to be merged into BSC's CI -- which I will work on next.

In hindsight, I should have submitted the merge as a PR.  In my clone of `bsc`, every looked fine: the testsuite commits appeared as a branch that was merged into the main branch; but when I pushed to GitHub, the commits appear to be rebased onto the main branch, without a merge commit.

I archived the `bsc-testsuite` repo, putting the commits on an `archive` branch and removing everything from the default branch except a README pointing here.  That README is unable to point to a merge commit here (or a PR) since there isn't one.

thoughtpolice, 2021-03-17:
Amazing work! This will also be great for systems like NixOS, where ideally we'd like to run the testsuite when we build it. I think we can't do that now (for various NixOS-related reasons) but it's something that's much easier to fix, now...

quark17, 2021-03-18:
With PR #336, the testsuite is now being run in the bsc CI.  So I think the merge of bsc-testsuite into bsc is complete and I think that closes this issue.

closed 2021-03-18 by quark17