B-Lang-org/bsc #288: new release tag request: no fixed (unique) release tag yet

vowstar, 2021-01-05:
This may brings problems to many Linux distributions that want to use the unique version of the package.
Could an updated release tag be issued if it is convenient? Thank you!

quark17, 2021-01-05:
Yes, I agree!  Until a point in the repo is tagged as a particular version (for example BSC v4.0.0), packaging BSC for Unix distros is difficult.  I would also like this GitHub repo to have pre-built tar-files for download, and that can't happen either until we tag a version.

This is discussed in issue #137.  As I said in a [comment](https://github.com/B-Lang-org/bsc/issues/137#issuecomment-613107744_), one thing is holding up the process.

It was suggested that we change the directory layout of the BSC installation, and doing that would be a significant change, and so it was suggested that we do this before making a first release (rather than make such a disruptive change later).  This is issue #85 , where I have also just added a comment, to hopefully move that discussion along.

closed 2021-01-05 by quark17

vowstar, 2021-01-06:
Thanks for reply! I'm looking forward to the release of the new version.