B-Lang-org/bsc #316: Export rule schedules via JSON

thoughtpolice, 2021-02-04:
The flag `-show-schedule` shows a very useful textual dump of the computed schedule for any synthesized module. I'd like to extract this information programmatically, in a form like JSON. This would allow me to always export this info with every build, then write a utility that (for example) exported HTML pages containing the schedule information as part of the CI process, so they can easily be recorded and viewed over time.

I suspect this is not very difficult, and thankfully, just emitting valid JSON without any extra 3rd party dependencies isn't *that* hard.

The first step would be to write and agree on rough schema of what the output JSON should look like, I think.

quark17, 2021-02-04:
Are you proposing that BSC should emit this format, or are you okay using a script to generate this after the fact, from the `.ba` file generated by BSC?

Bluetcl has commands for getting schedule information.  (See the User Guide in the `doc` directory, which I've just merged in, for the section on bluetcl commands.)   The [BDW](https://github.com/B-Lang-org/bdw), for example, has a a Schedule window with several tabs for viewing the information that bluetcl reports (specifically: any warnings; the flattened rule order; the submodule methods and which rule predicates, top methods, or instantiation arguments used them; and the rule relationship info for pairs of methods, which you can filter for viewing only the pairs that you care about).

thoughtpolice, 2021-02-05:
I have an old copy of the users guide lying around but often haven't referred to it much (since I'm not a heavy bluetcl user and didn't use BDW in the past, anyway.) I'm using bluetcl for small bits of functionality already though, so if it was possible to get a the schedule from the compiled files that way (and dump it however I need), that would fit the bill!

quark17, 2021-02-06:
The User Guide is now in the `doc` directory.  Appendix B describes the bluetcl libraries, and the commands you want are probably in B.4.2.  The `Bluetcl::schedule` and `Bluetcl::rule` commands probably have what you want.