B-Lang-org/bsc #317: BitExtend argument count doesn't match "BSV by Example"

elliottvillars, 2021-02-08:
On pg. 97 of the BSV by Example PDF, the BitExtend typeclass appears to take a single argument, t. 
This is conflicts with the error generated by ContextErrors.hs which expects BitExtend to take three arguments n,m and t.

quark17, 2021-02-09:
The `BitExtend` typeclass does take three arguments.  (And has since before "BSV By Example" was written.  So I think that's a mistake in the document.)  It may have been written this way because of limitations in BSC's typechecker at the time, that may no longer exist, so it's _possible_ that we could redefine `BitExtend` now to just take one argument.  However, at best, the one argument would be `Bit` and not `Bit#(n)`.  At best, you'd have to write your module like this:
```
module accumulator (AccIFC#(valCon#(n))
provisos (
  Bits#(valCon#(n),valTypeSz),
  Arith#(valCon#(n)),
  BitExtend#(valCon)
);
```
You could also add the following proviso, which would define `valType` as a simpler alias, if you prefer:
```
Alias#(valType, valCon#(n))
```

The `BitExtend` proviso should only be needed if, inside the module body, you are using an extend or truncate function.  You can avoid the need for the proviso by calling extend/truncate on a `Bit` value, rather than on `valType` -- by calling pack/unpack as necessary to convert between the two types.  (Or, you could define your own `Extend` typeclass, that passes size parameters as arguments to the functions, and therefore doesn't require the type to be of form `c#(n)` with the size in its static type.)

elliottvillars, 2021-02-09:
Thank you. So something like this would be a workaround:
```
foo <= extend(pack(valType))
```

quark17, 2021-02-10:
I apologize, I take back what I said about pack/unpack.  This would only work if the `extend` for `valType` is the same for `Bit`.  This isn't true for `Int`, which uses signed extension, because extending `Bit` is unsigned.

If I can back up: Why did you want to put a `BitExtend` proviso on the function?  Did you get a message from BSC that a proviso was needed?  If so, that proviso is probably shown in the form that you should use, in the provisos clause on the module.

I would need to understand more about what you're trying to write, to be able to recommend the right way to resolve the issue.  I'd need to understand why you are extending values.  Maybe your module is being given inputs of size `n` and summing them together, but the storage for the sum needs to be larger, so you need to extend the inputs from size `n` to some size `m` (say) that is larger?

I'm not sure that this is possible to write using the interface type that you've shown (`AccIFC#(valType)`).  I think that you would need to at least include the extended type, like this:
```
module accumulator (AccIFC#(valCon#(n), valCon#(m)))
provisos (
  Bits#(valCon#(n),valTypeSz),
  Arith#(valCon#(n)),
  BitExtend#(n, m, valCon)
);

// Where the interface might look like this?
interface AccIFC #(type inTy, type outTy);
  method Action input_value (inTy val);
  method outTy output_sum();
endinterface
```
Or specify the constructor and the two sizes (similar to how `BitExtend` is written):
```
module accumulator (AccIFC#(valCon, n, m)) ...;

interface AccIFC #(type con, numeric type inSz, numeric type outSz);
  method Action input_value (con#(inSz) val);
  method con#(outSz) output_value();
endinterface
```
In both of these, if you want to restrict `m` to be some proportion based on `n`, then you could do that by adding provisos -- for example, `Add n n m` to specify that `m` is twice as big.

If your interface is not returning the sum as a `valCon` type, but are returning it as a fixed type (like `Bit`), and the extended size is a fixed proportion (for example, twice the size of `n`), then you could probably make the original interface work, with something like this:
```
module accumulator (AccIFC#(valCon#(n))
provisos (
  Bits#(valCon#(n),valTypeSz),
  Arith#(valCon#(n)),
  BitExtend#(n, m, valCon),  // this will not be needed, if you use "extend(pack(val))"
  Add#(n, n, m) // The extended size is twice 'n'
);

interface AccIFC #(type t);
  method Action input_value (t val);
  // the output value is twice the size -- ah, but we can't return a larger size `t` here, so just use Bit
  method Bit#(TAdd#(SizeOf#(t),SizeOf#(t))) output_sum();
endinterface
```
As you can see, we can't construct `con#(m)` so we only have `Bit` and `TAdd` and `SizeOf` to construct the return type.  The `BitExtend` proviso may not be needed in this case (I mentioned in the comment): Using `extend(pack(valType))` would be okay here, because you're storing everything as `Bit` type anyway, and not converting back to the original (but extended) type.

elliottvillars, 2021-02-11:
The main module I am trying to build is an (n+1)-bit accumulator (for a UDP checksum) whose MSB functions as an overflow flag that can be acknowledged and reset independently.
I was trying to keep the whole module polymorphic and not adjust the original interface. 
I think the example you gave where you specified the constructor and the two sizes was helpful.
Is the "Best "Practice" for polymorphic interfaces to define the constructor and the size?

quark17, 2021-02-13:
> Is the "Best "Practice" for polymorphic interfaces to define the constructor and the size?

I don't recall ever seeing a module that was polymorphic with `BitExtend`, which is the only reason why you'd need to specify the constructor to the interface.

I do think it's common to have numeric types as arguments to an interface.  But the numeric variable is typically used with `Vector` or `Bit` or some other concrete type:
```
interface Ifc1 #(numeric type inSz, numeric type outSz);
  method Action input_value (Bit#(inSz) val);
  method Bit#(outSz) output_value ();
endinterface
```
Other times, the interface is just parameterized on the full type:
```
interface Ifc2 #(type inTy, type outTy);
  method Action input_value (inTy val);
  method outTy output_value ();
endinterface
```
I don't think I've seen the constructor given as a type, but presumably that's what you'd do here, if you really wanted to be polymorphic over types of that form.  I'd probably just use `Bit` (or `UInt`) for this situation, and only make it polymorphic over the size.

I might not even make it polymorphic over the size.  Some projects use `typedef` to define size configurations used throughout the project.  The project sizes can be reconfigured by just changing that one place, but the module definitions aren't technically polymorphic.  (Which can be a benefit because BSC currently can't separately synthesize polymorphic modules, so this `typedef` technique can let you put `synthesize` attributes on modules, while still being re-configurable.)  For example:
```
typedef 8 InputSize;
typedef 9 outputSize;

typedef Bit#(InputSize) InputType;
typedef Bit#(OutputSize) OutputType;

interface Ifc3;
  method Action input_value (InputType val);
  method OutputType output_value ();
endinterface
```

elliottvillars, 2021-02-14:
Got it. Thank you for all your help.

closed 2021-02-14 by elliottvillars