B-Lang-org/bsc #326: Current status of SVA support

thoughtpolice, 2021-02-25:
Although it's undocumented, `bsc` does support SVA syntax/properties of the form:

```
import SVA::*;

module t(Empty);
  let fooBool <- mkReg(False);

  property testingProp();
    fooBool |-> True;
  endproperty

  always assert property(testingProp())
    $display("Test");
  else
    $display("Failure");
endmodule;
```

However, when compiled, `bsc` warns `Warning: ... Support for SV assertions in this release is not complete or has not been tested`

The good news is that the assertion defined here actually *does* make it into the resulting RTL and is handled seemingly correctly, but it doesn't actually get emitted into an `assert property` statement or anything, just a regular clause handled and checked on the clock edge, so the best you'll get is some warning violation in simulation.

Generally, I've found the need for SVA in Bluespec to be much smaller than traditional Verilog; a significant amount of the errors in traditional RTL control logic comes from the need to manage enable/ready lines manually ("handshake bugs"), so the scheduler naturally handles the edge cases that arise here when your entire design is written in Bluespec.

That said, SVA assertions would still be use for extra levels of confidence, and I think they would also be useful when manually interoperating with external RTL and importing it into your design &mdash; especially given that you can write properties in terms of interface methods and then model check against specific implementations.

Is there any known list/summary of the current SVA support, and what's left for it to be considered a usable feature?

quark17, 2021-02-25:
SVA is supported in two modes: (1) a "pass-through" mode where the assertions appear in the output Verilog, to be checked by a Verilog simulator that supports VSA; or (2) the assertions are replaced with generated logic that does the checking (as long the assertion is finite).

The second mode is the default.  To get the first mode, you'll need to add the hidden flag `-passthrough-assertions`.

I will see if @joestoy can provide the current status.  I would guess that the pass-through assertions are more robust, since BSC isn't doing much.  The generated (non-passthrough) SVA was implemented first, and I believe that we have a user-guide document for that.  The testsuite has a directory `bsc.assertions` which, again, was probably added with the generated SVA support -- I notice that the `sequence` tests are currently commented out, which might point to a place where the feature is not fully working.

joestoy, 2021-02-26:
@quark17 and the compiler are both correct: the SVA feature is still experimental and not completely tested.  The chief reason has been the lack of realistic examples -- so if anyone can contribute some, or send us some, that would be very helpful in moving things along.

The following example illustrates the current state.  Here is a fragment of an AXI4 protocol checker, ```AxiPC.sv```, released by and (C) ARM Limited.
```
  // INDEX:        - AXI_ERRS_BVALID_STABLE
  // =====
  property AXI_ERRS_BVALID_STABLE;
    @(posedge `AXI_SVA_CLK)
          `AXI_SVA_RSTn & BVALID & !BREADY & !($isunknown({BVALID,BREADY}))
      |-> BVALID;
  endproperty
  axi_errs_bvalid_stable: assert property (AXI_ERRS_BVALID_STABLE) else
   $error("AXI_ERRS_BVALID_STABLE. Once BVALID is asserted, it must remain asserted until BREADY is high. Spec: section 3.1.3 on page 3-4.");

  // INDEX:        - AXI_RECM_BREADY_MAX_WAIT
  // =====
  // Note: this rule does not error if VALID goes low (breaking VALID_STABLE rule)
  property   AXI_RECM_BREADY_MAX_WAIT;
    @(posedge `AXI_SVA_CLK)
      `AXI_SVA_RSTn & !($isunknown({BVALID,BREADY})) &
      RecommendOn                              & // Parameter that can disable all AXI_REC*_* rules
      RecMaxWaitOn                             & // Parameter that can disable just AXI_REC*_MAX_WAIT rules
                         ( BVALID & !BREADY)
      |-> ##[1:MAXWAITS] (!BVALID |  BREADY);  // READY=1 within MAXWAITS cycles (or VALID=0)
  endproperty
  axi_recm_bready_max_wait: assert property (AXI_RECM_BREADY_MAX_WAIT) else
   $warning("AXI_RECM_BREADY_MAX_WAIT. BREADY should be asserted within MAXWAITS cycles of BVALID being asserted.");
```
When this is transliterated into BSV (with minimal changes) it becomes:
```
  // INDEX:        - prAXI_ERRS_bvalid_STABLE
  // =====
  property prAXI_ERRS_bvalid_STABLE;
      `AXI_SVA_RSTn && bvalid && !bready
      |-> bvalid;
  endproperty
/*  axi_errs_bvalid_stable:*/ assert property (prAXI_ERRS_bvalid_STABLE) else
   $error("prAXI_ERRS_bvalid_STABLE. Once bvalid is asserted, it must remain asserted until bready is high. Spec: section 3.1.3 on page 3-4.");

  // INDEX:        - prAXI_RECM_bready_MAX_WAIT
  // =====
  // Note: this rule does not error if VALID goes low (breaking VALID_STABLE rule)
  property   prAXI_RECM_bready_MAX_WAIT;
      recommendOn                              && // Parameter that can disable all prAXI_REC*_* rules
      recMaxWaitOn                             && // Parameter that can disable just prAXI_REC*_MAX_WAIT rules
                         ( bvalid && !bready)
      |-> ##[1:maxWAITS] (!bvalid ||  bready);  // READY=1 within MAXWAITS cycles (or VALID=0)
  endproperty
/*  axi_recm_bready_max_wait:*/ assert property (prAXI_RECM_bready_MAX_WAIT) else
   $warning("prAXI_RECM_bready_MAX_WAIT. bready should be asserted within MAXWAITS cycles of bvalid being asserted.");
```
If this is compiled with the ```-passthrough-assertions``` flag, the generated Verilog RTL contains the following, which can be handled by any simulator supporting SVAs:
```
  // synopsys translate_off
  svaAssertion_L1157C1: assert property (@(posedge CLK)
  ((!ecs_rstBoolM$VAL && snd_b_valid_value && !snd_b_ready_value
    ))|->(((snd_b_valid_value
	    )))) ecs_svaBool_L1157C1=1; else ecs_svaBool_L1157C1=0;
  // synopsys translate_on

  // synopsys translate_off
  svaAssertion_L1172C1: assert property (@(posedge CLK)
  ((snd_b_valid_value && !snd_b_ready_value
    ))|->((((((1'd1
	       ))[*32'd1
		   :32'd16
		    ]))##1((!snd_b_valid_value || snd_b_ready_value
			    ))))) ecs_svaBool_L1172C1=1; else ecs_svaBool_L1172C1=0;
  // synopsys translate_on
```
[... 16 lines cut ...]
  endproperty
let axi_errs_bvalid_stable <-
mkSVAReporter("prAXI_ERRS_bvalid_STABLE. Once bvalid is asserted, it must remain asserted until bready is high. Spec: section 3.1.3 on page 3-4.");
assert property (prAXI_ERRS_bvalid_STABLE) else axi_errs_bvalid_stable.reportError;

  // INDEX:        - prAXI_RECM_bready_MAX_WAIT
  // =====
  // Note: this rule does not error if VALID goes low (breaking VALID_STABLE rule)
  property   prAXI_RECM_bready_MAX_WAIT;
      recommendOn                              && // Parameter that can disable all prAXI_REC*_* rules
      recMaxWaitOn                             && // Parameter that can disable just prAXI_REC*_MAX_WAIT rules
                         ( bvalid && !bready)
      |-> ##[1:maxWAITS] (!bvalid ||  bready);  // READY=1 within MAXWAITS cycles (or VALID=0)
  endproperty
let axi_recm_bready_max_wait <-
mkSVAReporter("prAXI_RECM_bready_MAX_WAIT. bready should be asserted within MAXWAITS cycles of bvalid being asserted.");
assert property (prAXI_RECM_bready_MAX_WAIT) else axi_recm_bready_max_wait.reportError;
```
At compile time, ```mkSVAReporter``` associates a unique ```UInt``` with each instantiation, and prints a list of these ids with their associated strings.  Then, if when running in the FPGA the ```reportError``` method is called, an ```SVAReport``` gets transmitted to the host.

I attach a User Guide, dating from 2004, written by Michael Pellauer, who did the heavy lifting for the FSM implementation as a summer intern with Bluespec.  Note that the ```-passthrough-assertions``` mode came considerably later than this.  The later mode relaxes some of the restrictions imposed by the synthesis flow.  For example, ```CVParser.lhs``` says that "cover" is allowed as well as "assert".  Actually the parser has code to deal with the others ("assume", "expect" etc.), but these are currently disallowed.  I do not know what would happen if attempts were made to use them.

(Please also note that there might be some errors in the above, as I have not touched this for a couple of years: so please do report any errors or inconsistencies.)

[userguide.pdf](https://github.com/B-Lang-org/bsc/files/6050152/userguide.pdf)