B-Lang-org/bsc #331: Verilog Miscompilation

AsmPrgmC3, 2021-03-13:
Compiling the following program produces Verilog with behavior different to the Bluesim simulation.

```bsv
interface Bug;
    method ActionValue#(Bool) calculate(Bool change);
endinterface

module mkBug(Bug);
    let state <- mkReg(False);
    
    method ActionValue#(Bool) calculate(Bool change);
        if (change) begin
            let newState = !state;
            state <= newState;
            return newState; // (a) produces: `assign calculate = calculate_change ^ state ;`
            //return state; // (b) produces:  `assign calculate = state ;`
        end else begin
            return state;
        end
    endmethod
endmodule
```

The Bluesim simulation of variant (a) has the changes to `state` visible the same clock cycle as the change happened, as expected.
The Verilog however contains `assign calculate = calculate_change ^ state ;` meaning the output changes will happen independent of clock and enable signals (internal changes still happen at clock cycles).

The Bluesim simulation of variant (b) has the changes to `state` happen immediately but only visible the next clock cycle, again as expected.
The Verilog meanwhile contains `assign calculate = state ;` making those changes visible the same clock cycle (same as the Bluesim simulation with variant (a)).

quark17, 2021-03-14:
BSC's output looks fine to me.  Let me try to explain.  I'll have to start at the beginning, so I apologize if I'm saying things that are too basic:

BSC is a language for designing digital synchronous hardware.  Designs can be generated for multiple environments -- at the moment, that is just two: Verilog and Bluesim.  (Although saying just "Verilog" isn't precise enough, as I will explain below.)

When you write a BSV design, the meaning in that source language is that state values are updated atomically on clock edges.  BSC will take this source design and attempt to implement that design's meaning in one of the "backend" environments.

It is, indeed, important to ask whether the generated code (in a particular environment) implements the meaning of the original design.  (And, by transitivity, that different backends generate code that matches each other.)  But we have to be precise about what it is that we are observing in the output environment.

For example: We say that BSC has a "Bluesim" backend, but what BSC outputs is C++, because Bluesim is implemented in C++.  If you look at the generated C++ code in `mkBug.cxx`, you will see C++ assignment statements.  The semantics of the C++ language give a meaning to when these assignment statements execute and when the updated value is visible -- but these semantics are not what we refer to when we talk about assignment and visibility in Bluesim.  That's because Bluesim is an abstraction built on top of C++.

A similar perspective can be applied to the "Verilog backend".  BSC outputs Verilog code, but it's not just any Verilog code -- BSC is generating a particular kind of Verilog, for a higher-level purpose.

Verilog is a language that has been pulled in different directions for different purposes.  For one, Verilog has its own timing semantics that are used by simulators.  But Verilog is also used as the input language for synthesis tools that generate gate-level designs (that ultimately become chips).  And Verilog is used as the input language by synthesis tools that generate bit-files for programming FPGAs.

At the moment, BSC has only one backend that generates Verilog code -- but it would be very possible for BSC to have separate backends for ASICs-via-Verilog, FPGA-via-Verilog, and even native-Verilog-sim.

At the moment, BSC's Verilog backend is specifically for synthesis.  This was originally for ASIC-via-Verilog, and more specifically for Synopsys' Design Compiler.  Fortunately, the same style of Verilog can be used for various ASIC-via-Verilog tools; and can be used (for the most part) for various FPGA-via-Verilog tools.  So (at the moment) BSC is able to have one "Verilog" backend for a variety of synthesis tools.

All of this to say: When we talk about whether BSC's generated Verilog implements a particular input design, we aren't looking at the Verilog semantics, we're looking at a particular abstraction written in Verilog.

For synthesis to hardware, our mental model is that state values will update when clock signals tick.  But silicon chips aren't digital, they are merely emulating this mental model.  For latches to work, values have to be stable for a certain time before the clock time ("setup time") and be held for a certain time after ("hold time").  So we use a slightly more nuanced mental model, where combinational values are computed immediately and are held until the end of the clock period.  This is why the generated Verilog has signals that update immediately.

The other thing about hardware is that wires and logic always exist and always have values, even when those values aren't being used.

When we write an ActionValue in BSV, we should have in our mind the following abstract protocol: the method indicates whether it is "ready"; a rule that calls the method will only fire when all its methods are ready, and it will inform all the methods to execute by "enabling" them; when the method receives the "enable" signal, it will return a value and update state.  In our mental model, the returned value is not available until the method is enabled; but in hardware that has setup and hold times, if that value is itself being registered, then the hardware will need that value to be available before the clock edge -- and this is expressed in Verilog with assignments that are not clock-based.

And there's a further complication: The meaning of a BSV design is that each rule executes atomically, in sequence; but BSC attempts to generate hardware that executes many rules in one clock period.  It would be possible to write a design in Verilog, using Verilog's simulation semantics, where each rule executes in sequence throughout a clock period; but this Verilog design would not be accepted by synthesis tools.  BSC has to generate a Verilog design in the style accepted by synthesis tools, where all values are computed early in a clock cycle and all state latches their new values at the end of the clock cycle.

Does any of this answer your questions?

> The Bluesim simulation of variant (a) has the changes to `state` visible the same clock cycle as the change happened, as expected.
> The Verilog however contains `assign calculate = calculate_change ^ [...]

You talk about `state` for Bluesim but then talk about `calculate` for Verilog.  These are not the same thing.  The signal `calculate` is the return value of the method, while `state` is the register inside the module.  In both Verilog and Bluesim, the `state` will only update on the c [... cut ...] egister that is latching the value.

> The Bluesim simulation of variant (b) has the changes to `state` happen immediately but only visible the next clock cycle, again as expected.
> The Verilog meanwhile contains `assign calculate = state [...]

The behavior here is the same as for (a).  The Verilog statement `calculate = state` is a continuous assignment, just like the earlier assignment (`calculate = calculate_change ^ state`).  Why do you think that one is treated differently than the other?

AsmPrgmC3, 2021-03-14:
Thanks for laying things out this way. Even though all of it was known to me (except for the Verilog specialization part) I got a bit confused about what I wanted to happen versus was should happen.

> The behavior here is the same as for (a). The Verilog statement `calculate = state` is a continuous assignment, just like the earlier assignment (`calculate = calculate_change ^ state`). Why do you think that one is treated differently than the other?

It's not that I think it's treated differently, but that the value of `state` can only change at clock cycles instead of continuously.

Now what I got wrong: You are right that the value needs to be computed continuously to be available for further computations in the same clock, so it does work as intended. I was just trying to wire the output (`calculate`) directly up to an output pin, which should be clocked, and was somehow expecting this to be done by the method (as it is semantically only executed at clock edges).
Again, in hindsight, that's obviously wrong.

Though now I have the question: can i somehow specify this behavior (output only changes at clock cycles) in BSV? (I can obviously do it in Verilog, but is it possible in BSV?)

closed 2021-03-14 by AsmPrgmC3

quark17, 2021-03-14:
> Though now I have the question: can i somehow specify this behavior (output only changes at clock cycles) in BSV? (I can obviously do it in Verilog, but is it possible in BSV?)

I'm not sure if I understand.

BSC's output in situation (a) is this continuous assignment:
```
assign calculate = calculate_change ^ state
```
This signal only changes when the signals on the right side change.  The `state` signal changes at clock edges, so that's fine.  The `calculate_change` signal comes from an input -- is that what you're concerned about?  If that input is also coming from a register (and therefore only changing on clock edges) then `calculate` will only change on clock edges.

You could write your interface like this:
```
interface Bug;
    method Action calculate(Bool change);
    method Bool result();
endinterface

module mkBug(Bug);
    let state <- mkReg(False);
    
    method Action calculate(Bool change);
        if (change) begin
            let newState = !state;
            state <= newState;
        end
    endmethod

    method Bool result();
        return state;
    endmethod
endmodule
```
In this code, the `calculate` method is called in one clock cycle and then the `result` becomes available in the next.  Is that what you're asking for?

AsmPrgmC3, 2021-03-15:
Yes/Kinda. I wanted to have the output only change at clock cycles but also be able to feed the output back in as input without the one clock cycle delay that comes from `in <= bug.calculate(in)`. And if possible in a single method.
The solution is of course using your suggestion and then doing `bug.calculate(bug.result)`.

Thanks for all your help!