B-Lang-org/bsc #334: BSC is slower after merging Generics feature

quark17, 2021-03-16:
After the merge of PR #284, the test suite CI took nearly twice as long to complete (Ubuntu and macOS).  I have also observed 50% to 100% increase in testsuite runtime on my own Debian and macOS machines.  I believe that @krame505 is investigating.

kenta2, 2021-03-25:
@krame505 presented performance measurements via teleconference.   These were the commits measured, for future reference

c085ecd807d85f31d102d8bec71f5c28dc96b31d pre generics PR merge
e4c119687d532181cc03728908c5bbc3fddf5d94 post generics PR merge
bfd29c0c2553ae039bcd1c20ec27efe7ee8c3030 getQInsts optimization

Note that bfd29c0c2553ae039bcd1c20ec27efe7ee8c3030 belongs to PR https://github.com/B-Lang-org/bsc/pull/341 , so is not on the trunk.  38713153ad8b698d8e7d9974bdb6f2f0cc66d745 is the commit on the trunk.

@krame505 did measurements on both the pre and post testsuite.  These are some tags for the pre generics testsuite:

https://github.com/krame505/bsc/releases/tag/pre-generics
https://github.com/krame505/bsc-testsuite/releases/tag/pre-generics

quark17, 2023-05-07:
This was discussed in Discussion #414. GitHub puts a note in an issue when another issue references it, but GitHub doesn't seem to do that when a Discussion mentions an issues -- as there's no note here about the Discussion page with a link back to it.  So I'm commenting, to add that connection.  (Maybe we should consider closing the Discussion feature and just doing everything as Issues.  If someone does a search from the Issues page, they won't get any results from Discussions unless they remember to delete `is:issue` from the search command.  I was trying to find the latest on this issue and almost didn't find the Discussion page!)

To repeat the Discussion status here:

This is a significant regression, but unfortunately no one has time to work on it.  Our suspicion is that it has to do with the fact that, for an enum with N tags, there are N nested pattern/constructor applications, each with a type containing up to N `Either` constructors. This would result in an O(n^2) overhead in type checking for the `Generic` instance.

One conceivable fix would be to "flatten" the chain of `Either`s into a balanced binary tree, giving O(n log n) type checking overhead for the instance. This would require changing some generic instances that rely on all `Left` type parameters being `Conc` types. However having the representations structured in this way could actually be useful for doing serialization, as the chain of `Left`/`Right` constructors would directly correspond to 1s and 0s in the tag.

krame505, 2024-02-05:
I'm taking a stab at this - making bsc derive a balanced tree of `Either`s for the Generic instance looks easier than I initially thought.

closed 2024-02-06 by quark17