B-Lang-org/bsc #337: Better error message for -> instead of => in type signature

nanavati, 2021-03-19:
We've seen all sorts of hilarity including things like:

 Type error at:
   d

 Expected type:
   t

 Inferred type:
   (Prelude.IsModule m mType, Prelude.Bits t st)
(ironically one of the easier-to-diagnose variants).

I suspect some sort of up-front check that classes aren't where they aren't supposed to be in a type signature would help here.
Add comment

quark17, 2021-03-24:
Having a concrete example would help, particularly since I totally misread this and thought you were talking about code that mixes up left arrows (`<-` instead of `<=`)!  But I see now that you mean code like this:
```
mkMod :: (IsModule m mType, Bits t st) -> m t
```
where the user intended to write:
```
mkMod :: (IsModule m mType, Bits t st) => m t
```
This code can cause errors in two ways: Mismatches between the declared type and the inferred type from the body of the definitions; and (for top-level definitions) mismatches between the declared type and contexts where this top-level def is used in other top-level defs.

If I define the module as `mkMod = _`, then this definition typechecks without error!  If I define the module using a `module` block, such as `mkMod = module { return _; }`, then I get an error like this:
```
Error: "Test.bs", line 4, column 17: (T0031)
  The contexts for this expression could not be resolved because there are no
  instances of the form:
    Prelude.Monad ((->) (Prelude.IsModule m mType, Prelude.Bits t st))
```
If I try to use this top-level def in code elsewhere, such as:
```
mkTop :: (IsModule m mType) => m Empty
mkTop = module
          m :: Bool
          m <- mkMod
```
then I get this error:
```
Error: "Test.bs", line 9, column 15: (T0084)
  The following function expects 1 arguments, but it was used without any
  arguments:
    mkMod

  The expected type is:
    (Prelude.IsModule a__ b__, Prelude.Bits c__ d__) -> a__ c__

  But it was used with the following type:
    e__ Prelude.Bool
```
I don't know what code gave the error that @nanavati showed; if there are examples, they would be good to include.

quark17, 2021-03-24:
Ah, if the definition has an argument, then I see an error similar to what @nanavati reports:
```
mkMod :: (IsModule m mType, Bits t st) -> t -> m t
mkMod x = module { return x; }
```
gives:
```
Error: "Test.bs", line 4, column 26: (T0020)
  Type error at:
    x

  Expected type:
    m t

  Inferred type:
    (Prelude.IsModule m mType, Prelude.Bits t st)
```

quark17, 2021-03-24:
This code produces the message that @nanavati posted:
```
mkMod :: (IsModule m mType, Bits t st) -> t -> m t
mkMod x = module
            let val :: t = x
            return _
```
And that error arises on the assignment:
```
Error: "Test.bs", line 5, column 27: (T0020)
  Type error at:
    x

  Expected type:
    t

  Inferred type:
    (Prelude.IsModule m mType, Prelude.Bits t st)
```
This code also gives a second error on the return statement (the one we saw about no instances of monad for arrow).  If I change the code to return `val` instead of don't-care, the return statement causes this error:
```
Error: "Test.bs", line 6, column 19: (T0020)
  Type error at:
    val

  Expected type:
    m t

  Inferred type:
    t
```
So a single definition might produce multiple type errors, resulting from the mistaken use of `<-` instead of `<=`.  So reporting this better may not be just a matter of changing how one error is reported, but catching several errors (and replacing them with a new error, or just printing a new error to start and then allow the other errors to follow?).

In this example, if I change the use of `x` from an assignment to inside a register instantiation:
```
mkMod x = module
            val :: t <- mkReg x
            return val
```
then I see a slightly different message (and no second message):
```
Error: "Test.bs", line 5, column 30: (T0020)
  Type error at:
    x

  Expected type:
    a__

  Inferred type:
    (Prelude.IsModule m mType, Prelude.Bits (Prelude.Reg a__) st)
```