B-Lang-org/bsc #339: Support methods with multiple output ports

nanavati, 2021-03-19:
An alternative (probably better) way to resolve the underlying issue from #219 is supporting methods with multiple output ports.

The large numbers of conflict-free methods weren't logically distinct. They were just hack to get different output ports for different fields of a structure / different elements of vectors. 

If there were a way to tell the compiler to create separate output ports for struct fields, vector elements and so on *without* making separate methods we could avoid all of the unnecessary pressure on the scheduler.

This probably needs to be done in two phases:
1. Extending import "BVI" to support multiple outputs ports.
2. Giving users attribute or other ways to indicate they want multiple output ports for synthesized method.

quark17, 2021-03-19:
It's worth noting that there's another workaround that people currently use: post-process with a bluetcl script that can lookup the types and split the ports.  I believe that this is what the `expandPorts.tcl` script in the `util` directory is for.  This script is mostly for adjusting the ports at the outer boundary of a BH/BSV design (more would need to be done for internal boundaries); and obviously it would be better for BSC to generate the Verilog this way itself.

jahagirdar, 2024-03-13:
@quark17 ,
The expandPorts.tcl seems to be broken. There are hardcoded paths in the script for things like portUtils etc. which do not exist in the installation.