B-Lang-org/bsc #354: Internal Bluespec Compiler Error

rsnikhil, 2021-04-26:
This is what I'm doing:

    $ git clone https://github.com/bluespec/Piccolo
    $ cd Piccolo/builds/RV64ACDFIMSU_Piccolo_bluesim/
    $ make all

During the bsc linking phase I get:

    Internal Bluespec Compiler Error:
    Please report this failure to your Bluespec technical contact.
    If you do not know your contact, you may email support@bluespec.com.
    The following internal error message should be included in your
    correspondence along with any other relevant details:
    Bluespec Compiler (build d05342e3)
    cycle in combsched: [Exec soc_top.core.cpu.csr_regfile.RL_rl_reset_start,
     Exec soc_top.RL_rl_connect_external_interrupt_requests,
     Exec soc_top.core.cpu.RL_rl_stage1_SFENCE_VMA,
     Exec soc_top.core.cpu.near_mem.icache.RL_rl_rereq,
     Sched soc_top.core.cpu.near_mem.icache.RL_rl_rereq,
     Exec soc_top.core.cpu.RL_imem_c_rl_fetch_next_32b,
     Exec soc_top.core.cpu.RL_rl_reset_start,
     Exec soc_top.core.cpu.csr_regfile.RL_rl_reset_start]

I have attached a full log of 'make all' (which includes the
above error message near the end).

The bsc version I'm using was freshly cloned and built:

    $ bsc -v
    Bluespec Compiler (build d05342e3)

[log_make_all.txt](https://github.com/B-Lang-org/bsc/files/6373734/log_make_all.txt)

quark17, 2021-04-26:
I am seeing the same behavior, and will look into it.

I notice that Bluespec's bug database has a ticket (bug 1911) for a similar combsched error in an early version of Flute.  That was going through the data cache in the write back.  It was never diagnosed.

I also see (in my email archive) that I later helped you with a combsched error in the SoC which goes through the icache (and so might be more similar to this one).  That was when the SoC was still in Bluespec's private repo, not on GutHub yet.  That situation _was_ diagnosed, and here's what I said at the time:
> What ought to happen is that you should get an error G0101 about module "mkCPU", that it requires "dynamic scheduling" to properly implement, which Bluesim does not support.  As the message would explain:
>
>"This is because the rules use methods which have a rule that executes between them in the static execution order of the separately synthesized submodule, but some pairs must execute in one order and some pairs must execute in the opposite order. See entry 30 in the KPNS document for more information and possible solutions."
>
> However, bug 1410 is preventing BSC from detecting this situation before trying to combine the schedules.  Bug 1410 is that the "has rule(s) between" info for pairs of methods is not propagated from synthesized submodules to synthesized parents.
>
>The module "mkCPU" instantiates "mkNear_Mem" which instantiates "mkICache".  The "mkICache" module has rules between methods, and these methods are exported by "mkNear_Mem".  But BSC does not correctly compute the "has rule(s) between" info for the exported interface.  If you remove the "synthesize" attribute on "mkNear_Mem", removing the intermediate boundary, you will get the G0101 errors from BSC.
>
> My suggestion is to remove the attribute, resolve the G0101 errors, and then put the synthesize attribute back.

I will see if I can remove synthesize attributes (on the near_mem) to reveal a G0101 error.  If so, this can be resolved either by removing the "rule in between" the two methods or by changing the caller of the methods (to break up the before and after calls, so that they can be ordered separately in the schedule).

I'm not sure why bug 1410 was not addressed.  I will file that here on GitHub and look into it.

quark17, 2021-04-26:
Removing the synthesize attribute leads to the design compiling.  So it's not the same issue that I mentioned above.

It might be related to Bluespec bug 1706, which is about a combsched that results when methods are used in predicates: BSC needs to treat execution order requirements as urgency order requirements, in that case; because it's not, the problem isn't detected until the schedule is combined.  That the example compiles when you remove the synthesize attribute is consistent with this issue.  I will file 1706 on GitHub, and Bluespec bug 1363 that is mentioned in it and seems possibly related.

Do you need a workaround in the meantime?  Removing the synthesize is one workaround, but there may also be workarounds while keeping the attribute.  I can look closer into the exact cycle, to suggest ways to break it.  I notice offhand that the cycle includes some execution-orders for reset rules.  That suggests that you might break the cycle by having the reset rules conflict, rather than merely have an order.  (When I compile, I notice, for example, that there are execution order warnings about reset rules appearing to execute after other rules.  If you were OK with suppressing those other rules, when the reset is applicable, then there'd be no execution ordering.)

quark17, 2021-05-13:
Some updates:

A workaround for Piccolo is to add the following scheduling attributes to `src_Core/Near_Mem_VM/MMU_Cache.bsv`:
```
(* mutually_exclusive = "rl_start_cache_refill, req" *)
(* mutually_exclusive = "rl_cache_refill_rsps_loop, req" *)
(* mutually_exclusive = "rl_rereq, req" *)
```
The design already expects that `req` will not be called again until the previous call has finished and returned a result (as indicated by a comment on the method, that it does not have flow control).  So we can also tell BSC that the rules for processing a request will never be executing at the same time as a new request.  This eliminates scheduling dependencies between the rules and the method, which are part of the cycle that BSC has detected -- but has detected late in its process, so an internal error has been reported, since such cycles are expected to have been detected earlier.

I don't believe that the issue here is the same as what's reported in Bug 1706.  I have a prototype fix for Bug 1706, and it doesn't not change the behavior in this case.  (BSC determines execution-order relationships between rules by noting which methods a rule calls.  When the urgency and execution orders are merged into a single graph, containing separate Sched and Exec nodes for each rule, the execution-order requirements become Exec-to-Exec edges.  But when methods are used in the predicate, the edge should connect the Sched node.)

What I think is happening for Piccolo is that BSC is not fully detecting "rule between methods" situations.  A number of examples of this are in `testsuite/bsc.scheduler/relax-schedule/`.  In particular, the examples that test for error `G0100` about "dynamic scheduling" being not yet supported for Bluesim.  The situation is that two methods might have an execution order relationship because the submodule has a rule that executes between the two; if the methods are called in two rules in the parent module, this implies an ordering on those rules; however, if the rules have exclusive predicates (or if they conflict, such that one blocks the other from firing), then the scheduling in the parent module can ignore any ordering, because the rules won't execute together; and, so, if there's another rule in the parent module that needs to execute between the two methods, forcing the opposite ordering, then that's OK; but, for the moment, Bluesim needs to be able to order all rules in the design, and there's no way to order these four rules.  Bluesim could probably be made to support this, if we expressed the schedule not as a single ordering but as an if-else-tree of alternate orders (allowing one of the exclusive rules to fire before the other in some cases and to fire after the other in some cases). I called this "dynamic scheduling".  The "rule between" in Piccolo is only in the combined graph (it goes through urgency edges between Sched nodes); I'm guessing that BSC is not looking for this kind of path, and that needs to be fixed.

I also want to make sure that BSC is not unnecessarily creating loops that are rejected as "dynamic scheduling".  That is: a path between the two exclusive rules, in the parent module, could be created by arbitrary choices taken by BSC (such as when two rules write the same register, so the later one overwrites the earlier one's value).  We want BSC to bias its choices based on the rule-between info that it has.  (This may already be happening, but I want to check.)  I also notice that BSC is not warning about arbitrary shadowing choices when two rules write the same value to a register, but does seem to still create an ordering edge (that can lead to a dynamic-scheduling error).  If it's not going to warn, then it shouldn't create an edge (or at least should warn if it's going to create an edge -- because I was trying to debug a small example and couldn't figure out where an ordering edge was coming from, since I didn't see a warning).

quark17, 2021-05-14:
My original comment here suggested that maybe we're seeing Bug 1410: rule-between info from `mkMMU_Cache` is not being propagate up to `mkNear_Mem`. I decided against that because BSC doesn't report a dynamic-scheduling error when the `synthesize` attribute on `mkNear_Mem` is removed.  However, I think that's because the dynamic scheduling was only triggered by the arbitrary choice, which could be picked in the opposite direction and not cause a need for dynamic scheduling; so when the `synthesize` attribute is removed, BSC picks the right direction (because it is not longer arbitrary).  So it could be that Bug 1410 is the issue here after all.

I have a boiled-down example of the Piccolo bug here: [CPU.bsv](https://github.com/B-Lang-org/bsc/files/6475590/CPU.bsv.txt).  (It is perhaps a bit _too_ boiled down, because, as I was removing code and compiling to confirm that the removed code didn't change the behavior, I only checked that a `combsched` error was still reported; I did not check that the same path was still reported.  So this path is slight shorter, but I think that's OK because the relevant part is the path between the methods of `mkMMU_Cache`, which is still there.)  In this example, the rule `imem_c_rl_fetch_next_32b` did not originally have an `execution_order` attribute, but BSC would pick an arbitrary order and the `combsched` would occur.  Without the attribute, if you remove `synthesize` on `mkNear_Mem`, the error goes away -- as on the full Piccolo.  But if you include the attribute (which I've done in the attached code), then removing the `synthesize` does lead to a dynamic-scheduling error:
```
Error: "CPU.bsv", line 30, column 9: (G0100)
  The rules `RL_rl_stage1_SFENCE_VMA' and `RL_imem_c_rl_fetch_next_32b'
  require dynamic scheduling, which is not supported by Bluesim. This is
  because the rules use methods which have a rule that executes between them
  in the static execution order of the separately synthesized submodule, but
  the rules must execute in the opposite order according to the current
  module's schedule. See entry #30 in the KPNS document for more information
  and possible solutions.
  The methods and the rules between them are as follows:
    (near_mem_icache.req, near_mem_icache.tlb_flush)
      RL_rl_cache_refill_rsps_loop
  The execution order path is as follows:
    `RL_rl_stage1_SFENCE_VMA' -> `RL_rl_reset_start' ->
    `RL_imem_c_rl_fetch_next_32b'
  The relationships were introduced for the following reasons:
    (RL_rl_stage1_SFENCE_VMA, RL_rl_reset_start)
    execution order because of
    calls to rg_cache_b16.write vs. rg_cache_b16.read
    (RL_rl_reset_start, RL_imem_c_rl_fetch_next_32b)
    execution order because of
    earliness attribute at "CPU.bsv", line 35, column 26
```
This suggests that Piccolo might start compiling (no `combsched` error) if BSC bug 1410 is fixed.  It also suggests that we could workaround the bug in BSC by inserting an `execution_order` attribute in an appropriate place (where BSC is currently making a poor arbitrary choice in the full Piccolo).  And (as I said earlier) BSC may not currently be warning about this arbitrary choice because it is for register write shadowing where the value is the same.  (If there's a flag to turn on these silenced shadow warnings, we might turn them on, to see if any are reported along the `combsched` path.)

closed 2021-05-17 by quark17