B-Lang-org/bsc #356: Compile time error with sprintf and dynamicAssert

chiraag, 2021-05-04:
The following code fragment generates a surprising compile time error.
```bsv
dynamicAssert(in_d == ref_d, sprintf("FAIL: %d != %d", in_d, ref_d));
```

Error Message:
```sh
Compile time expression did not evaluate:
Type: Prelude.Bit 32
.FIFOF_.first �32 �0 stim_in_f

During elaboration of the body of rule `action_l27c21' at "src/Stim.bsv",
```

The following fragment without the sprintf works fine:
```
dynamicAssert(in_d == ref_d, "FAIL");
```

The code was compiled with `-check-assert`

quark17, 2021-05-04:
The `String` argument to `dynamicAssert` has to be constructed at compile time.  That means that you can't print dynamic values, such as the contents of a FIFO.  That's what the error message is telling you (although maybe the message could be made friendlier).

The `Assert` package is extremely old.  If I were writing a package like that today, I might change the `String` argument to something like `Fmt`, so that dynamicly constructed messages could be printed.  The `Fmt` argument could be constructed with `$format` (and `fshow`).  Or, if necessary, the argument could be `Action`, allowing you to pass in arbitrary set of system tasks (for displaying or printing to a file).

You might try writing your own version of `dynamicAssert` (along those lines) instead of using the `Assert` package.

quark17, 2022-12-05:
> If change the `String` argument of `dynamicAssert` to `Fmt`, how to get the position of the `Fmt` argument in the source file?

That's a good question!  In the worst case, you can require a `String` argument that's only used for its position.  However, you could try using `primGetEvalPosition` to get the position of the `Fmt` value.  This is used in the `Prelude` in a couple places, such as:
```
digitToInteger :: Char -> Integer
digitToInteger c =
    if (isDigit c)
    then (primCharOrd c) - (primCharOrd '0')
    else let pos = primGetEvalPosition c
         in  primError pos $
                 "digitToInteger: not a digit: " +++ quote (charToString c)
```
The declaration for `primGetEvalPosition` is also given there:
```
primitive primGetEvalPosition :: a -> Position__
```
However, I notice it is not exported from the Prelude.  You can add an export or possibly workaround it by repeating the primitive declaration in your file.  If this function is useful, we could consider exporting it and also consider giving it a better name (or defining and exporting a function based on it).

pwang7, 2022-12-05:
> That's a good question! In the worst case, you can require a `String` argument that's only used for its position. However, you could try using `primGetEvalPosition` to get the position of the `Fmt` [...]

Thanks a lot @quark17, let me try~

One more question is if I have a case statement in a regular function (function not return Action/ActionValue), the case statement will return a value, and I'd like assert at default case. If using `staticAssert` as following, it gives compilation error instead of runtime assertion.

What's the best way to assert in a case statement with return value?
```Bluespec
function Bool simpleFunc(Bits#(8) inputValue);
    let ret = case (inputValue)
        validCase1 : True;
        validCase2 : False;
        validCase3 : False;
        validCase4 : True;
        default : staticAssert(...);
    endcase;
    return ret;
endfunction
```

quark17, 2022-12-06:
You could change the function to return `Maybe#(Bool)` and then the place that calls the function can check for `Invalid` values and give an error there.  Or, change the function to be a module (that returns `Bool` when instantiated), and then the arms of the case-statement can call `staticAssert`.  If you want a runtime assertion, you could change the function to return `ActionValue#(Bool)`, and then the arms of the case-statement can call actions like `dynamicAssert`.

pwang7, 2022-12-06:
> You could change the function to return `Maybe#(Bool)` and then the place that calls the function can check for `Invalid` values and give an error there. Or, change the function to be a module (that [...]

Thanks a lot! @quark17