B-Lang-org/bsc #364: Incomplete conflict spec in module verilog causes an uninformative ICE

mieszko, 2021-05-21:
Not specifying conflict info for all method-method pairs in `module verilog` fails with a very uninformative ICE. This arises when compiling, for example, the following slightly altered extract from `Libraries/Base2/SPSRAM.bs`:

```
package BugVModInfo where

interface VSyncSRAM adrs dtas =
    exec  :: Bit adrs -> Bit dtas -> Bit 1 -> Bit 1 -> PrimAction
    rdata :: Bit dtas

mkSPSRAM_V :: Integer -> Module (VSyncSRAM adrs dtas)
mkSPSRAM_V nwords = do
    let name = "RRSPSRAM_" +++ integerToString nwords +++ "_" +++
                integerToString (valueOf dtas) +++ "_bussed"
    messageM ("Used SRAM: " +++ name)
    module verilog name "CLK" {
        exec  = "A" "DI" "WE" "ENABLE" "?"{inhigh};
        rdata = "DO";
    } [ exec <> rdata ]
```

Compiling with `bsc` causes the following ICE:

```
Internal Bluespec Compiler Error:
Please report this failure to your Bluespec technical contact.
If you do not know your contact, you may email support@bluespec.com.
The following internal error message should be included in your
correspondence along with any other relevant details:
Bluespec Compiler (build 8d454e41)
VModInfo: (VModInfo
 ???
 clock _clk__1(CLK, {-unused-});
 [clockarg _clk__1;]
 [method exec((A, []), (DI, []), (WE, []), (ENABLE, [])) enable ((?,
                                                                  [inhigh])) clocked_by (_clk__1) reset_by (no_reset);,
  method (DO, [])rdata clocked_by (_clk__1) reset_by (no_reset);]
 SchedInfo [exec CF rdata] [] [] []
 [])
[(exec, rdata)]
[(exec, rdata), (exec, exec), (rdata, rdata)]
```

This is especially unhelpful as the module name has been hardcoded to `???` in `IConv.hs`.

The ICE shows comes from `mkVModuleInfo`, and is triggered because conflict information was specified only for exec vs rdata, but not for exec with itself or rdata with itself (the original annotation was `[ [exec,rdata] <> [exec,rdata] ]`).

Possible fixes:

- hack to add a better error description / position info to the ICE (at least the fields are Ids so they have position info from the same `module verilog` clause)
- factor out checking from `mkVModuleInfo` into an IO wrapper that can be called from IConv
- do the checking elsewhere

quark17, 2021-07-22:
In BSC's internal representation (CSyntax) there is an expression construct, `CmoduleVerilog`.  In BSV, users can't directly create one of these expressions.  Instead, there is a special import-BVI syntax that has to appear at the top-level; and the BSV parser converts this into a wrapper module that instantiates a `CmoduleVerilog` inside and then does other wrapping as necessary.  For example, `CmoduleVerilog` can only import limited interfaces (with `Bit` types), but import-BVI can declare an imported module with user types and nested interfaces etc -- the wrapper module includes the packing/unpacking between the raw `CmoduleVerilog` and the types as visible to the user.

The BSV parser also checks the scheduling attributes.  If there are unspecified pairs (in the same clock family), then the BSV parser warns and inserts that pair with a best guess at a conflict type.  The kind of internal error that you report here can't occur with BSV, because the parser has ensured that the conflict info is complete.  See `src/comp/Parser/BSV/CVParserImperative.lhs`, where it generates the warning `WNoSchedulingAnnotation`.

In BH/Classic, there is no equivalent to import-BVI that can be specified at the top level.  Instead, users have direct access to create `CmoduleVerilog` expressions (by writing `module verilog` in any expression).  Well, there _is_ a small wrapper -- the function `xClassicModuleVerilog` in `CSyntax` is called by the Classic parser (`src/comp/Parser/Classic/CParser.hs`) and it does some basic wrapping.

The user-friendliness of Classic `module verilog` had not been a concern, while BSV was the only supported syntax. I'm not surprised that we're now finding these situations which need better checking and messages.  There are probably also many features in import-BVI that are now available in BH/Classic.  We may want to support a new construct in BH/Classic that is like import-BVI.  But I don't think we can remove `module verilog`, so we'll still need to fix up these internal-error issues.

My suggestion (in place of the three possibilities that you mentioned) is that we add checking code to `xClassicModuleVerilog`, so that we warning the user as soon as possible.  Hopefully we can move the checking code from the BSV parser into a common location and then call into that (and not duplicate the checking code). Maybe there's some way to do the checking in one place, after parsing by either pass, but I don't see a problem with just calling into shared checking code from both parsers.  (If we decide to support an import-BVI construct in both BH and BSV, then we could probably defer a lot of the processing currently done in the BSV parser, to be done in a stage just after, that's common to both parsers.  But that would still be separate from how we support module-verilog.)

Hrm.  But I guess the BH/Classic parser doesn't support reporting warnings (even if the function `xClassicModuleVerilog` were extended to return warning information)?  A second pass that is called after the BH/Classic parser, that just looks for `CmoduleVerilog` and does any necessary checking ... would be fine too.  (If the BH/Classic parser can't be adjusted to accumulate warnings.)

mieszko, 2021-08-06:
IIRC the parser combinators used for BH only support errors of the type "expected _x_, found _y_"? I suppose one could retrofit the state to carry errors, but I'm not sure it's worth the hacking effort.

But in the short term I think it would be fine to detect in/after parsing as you suggest, and even just error out via an ICE that is more informative and has a location. I don't think it's a big deal to add all the required pairs to `module verilog` if one knows what to do, but a beginning user would probably have a heart attack :)

As you said, there is a separate issue of the Classic `module verilog` not having all the features of the BSV variant. I think the most glaring issue in practice is probably support for specifying multiple clocks/resets. For example, if you have a multi-ported SRAM it's likely that the ports have their own clocks, and really most designs of any complexity will have multiple clock domains. I think these annotations can be retrofitted into the Classic `module verilog`, but perhaps it's worth thinking about multiple clock domain support in BH as a separate issue, and solve `module verilog` as part of that.

(Incidentally, one nice feature the BH version does have is that the Verilog module name is passed in as a string, which is very useful if you are instantiating many SRAMs with different dimensions for example, while in BSV you have to declare a BVI import for every. single. instance...)