B-Lang-org/bsc #368: Cannot synthesize real

samuelUniversal, 2021-06-11:
```
package RealOperation where

-- Interface definition

interface RealOperation_IFC =
    addition       :: Real -> Real -> Real
    multiplication :: Real -> Real -> Real

{-# verilog mkRealOperation #-}

mkRealOperation :: Module RealOperation_IFC
mkRealOperation =
    module
        interface
            addition       a1 a2 = a1 + a2
            multiplication p1 p2 = p1 * p2
```
Cannot synthesize `mkRealOperation': The interface method `addition' uses
  type `Prelude.Real' which is not in the Bits class.
I want to use Real to realize floating-point addition and multiplication, how to resolve Real in static elaboration?
Thanks!

quark17, 2021-06-13:
At the moment, you will need to replace `Real` with a type that can be synthesized, such as `FixedPoint` and `FloatingPoint`.  These are available in libraries by those names.

It's not an unreasonable feature request to ask that BSC allow `Real` values to remain in a synthesized design, and perhaps be natively implemented with Verilog's `real` type.  This hasn't been a priority, because designs for real hardware have usually wanted to have more control over the implementation of fixed/floating operations.  The use of `real` has seemed like more of a simulation feature, although there may be a simulation speed advantage to using `real` (and maybe synthesis tools can do more with it these days?).

For anyone interested in working on this feature, I'm attaching a small patch ([bitstoreal_patch.txt](https://github.com/B-Lang-org/bsc/files/6644813/bitstoreal_patch.txt)) that Don Baltus wrote to allow `$bitstoreal` to exist after elaboration and appear in the Verilog -- he was specifically interesting in using it within `$display` statements, so I don't know if it's been tested for other uses.  For example, I don't know if this code would work:
```
module mkTest ();
  Reg#(Bit#(64)) rg1 <- mkRegU;
  Reg#(Bit#(64)) rg2 <- mkRegU;
  Reg#(Bit#(64)) rg3 <- mkRegU;

  rule r;
    rg3 <= $realtobits( $bitstoreal(rg1) * $bitstoreal(rg2) );
  endrule
endmodule
```
This patch also doesn't add `Real` to the `Bits` class, so it doesn't yet allow synthesizing modules with `Real` types in the interface, or allow instantiating registers of `Real` values.  That might be significantly more work.

samuelUniversal, 2021-06-15:
> At the moment, you will need to replace `Real` with a type that can be synthesized, such as `FixedPoint` and `FloatingPoint`. These are available in libraries by those names.
>
> It's not an [...]

Thanks for your reply!
I don't know that there is a `FloatingPoint` module in libraries. I search all materials I can find on Google, but those materials don't mention the `FloatingPoint` module. That is all I needed! Thank you very much!

And for those code:
```haskell
package Bits2Real where

interface Bits2Real_IFC =
    set :: Bit 64 -> Bit 64 -> Action
    get :: ActionValue (Bit 64)

{-# verilog mkBits2Real #-}
mkBits2Real :: Module Bits2Real_IFC
mkBits2Real =
  module
    rg1 :: Reg (Bit 64) <- mkRegU
    rg2 :: Reg (Bit 64) <- mkRegU
    rg3 :: Reg (Bit 64) <- mkRegU

    interface
        set x y = do
                    rg1 := x
                    rg2 := y

        get = do
                rg3 := $realtobits( $bitstoreal(rg1) * $bitstoreal(rg2) )
                return rg3
```
it can be passed when I am using the command `bsc Bits2real.bs`, but it also cannot be synthesized when I am using a command like `bsc -verilog Bits2real.bs`, and the error message are:
```shell
Error: "Bits2Real.bs", line 21, column 35: (G0013)
  Compile time expression did not evaluate:
  Type: Prelude.Real
  PrimBitsToReal (.Prelude.read ?64 rg1)

  During elaboration of the interface method `get' at "Bits2Real.bs", line 8,
  column 0.
  During elaboration of `mkBits2Real' at "Bits2Real.bs", line 8, column 0.
```

quark17, 2021-06-20:
I would like to keep this issue open, because it is a legitimate feature request that someone else may be interested in later.