B-Lang-org/bsc #372: /bin/sh: 1: pkg-config: not found

kenta2, 2021-06-17:
Compiling on Ubuntu 18.04 I see the following warning in several places (not an error, compile continues past the point and `make check` passes).  For example, twice in the excerpt below:

```
make[2]: Entering directory '/tmp/ken/bsc/src/vendor/htcl'
/bin/sh: 1: pkg-config: not found
ghc -Wall -I/usr/include/tcl -c haskell.c
ar -r libhtcl.a haskell.o
ar: creating libhtcl.a
make[2]: Leaving directory '/tmp/ken/bsc/src/vendor/htcl'
# we need to build targets from here sequentially, as they operate in the same workspace
make  -C comp -j1   PREFIX=/tmp/ken/bsc/inst  install
make[2]: Entering directory '/tmp/ken/bsc/src/comp'
----- Normal build options -----
./update-build-version.sh
./update-build-system.sh
/bin/sh: 1: pkg-config: not found
bsc start Thu Jun 17 18:04:53 EDT 2021
./update-build-version.sh
```

It looks like `pkg-config` is being invoked in `src/comp/Makefile`:
```
TCL_INC_FLAGS = $(shell pkg-config --silence-errors --cflags-only-I tcl || echo -I/usr/include/tcl)
```

Also `src/vendor/htcl/Makefile`.

The straightforward suggestion is to add a package containing `/usr/bin/pkg-config` to the list of recommended packages.

However `/usr/bin/pkg-config` is provided by two competing packages on Debian and Ubuntu:
1. `pkg-config`
2. `pkgconf`

It looks like `pkg-config` is in `main` on Ubuntu but `pkgconf` is in `universe` so pkg-config might be marginally better supported.

I confirm that both "work" for the command line `pkg-config --silence-errors --cflags-only-I tcl` .

I put "work" in scare quotes, because `tcl` not part of the pkg-config database on Ubuntu 18.04, so the command actually fails and falls back to `echo -I/usr/include/tcl`, which is fine: the correct include files are at that location.

So the suggestion for Debian and Ubuntu might be along the lines of, "to suppress a spurious warning of `pkg-config: not found`, install either the `pkg-config` or `pkgconf` packages."

quark17, 2021-06-17:
Thanks for noting this!

Our CI builds on GitHub's Ubuntu VM, which has a number of packages pre-installed ([listed here](https://github.com/actions/virtual-environments/blob/main/images/linux/Ubuntu1804-README.md)) and we install further packages with the commands in the script at [.github/workflows/install_dependencies_ubuntu.sh](https://github.com/B-Lang-org/bsc/blob/main/.github/workflows/install_dependencies_ubuntu.sh).

One of the packages that we explicitly install is `build-essential` and I don't know if that pulls in one of the pkgconfig packages.  But I note that GitHub's VM has `pkg-config` pre-installed (and does not have `pkgconf` installed).

I will see about adding `pkg-config` to the list of required packages in the README files.

I believe that defaulting to `/usr/include/tcl` when not found in the pkg-config database is what's currently expected (as introduced in 3952141 and 53f6b36, as a slight improvement on hardcoding `/usr/include/tcl` for all systems).

quark17, 2021-06-17:
I assume that we could also insert a redirect of `stderr` to `/dev/null`, to avoid warnings when `pkg-config` is not found

quark17, 2021-09-21:
The README now mentions pkg-config.  And PR #260 includes a cleanup to how the Tcl include flags are computed (moving it into `platform.sh`), including checking whether pkg-config exists before using it.  That should close this issue when merged.

closed 2023-07-29 by quark17