B-Lang-org/bsc #377: HTML based documentation

1ntEgr8, 2021-07-07:
It would be nice to have a website that hosts the Bluespec user guide and libraries reference. We can use [`pandoc`](https://pandoc.org/) to do the conversion, and host it through Github Pages.

The goal is to make documentation more accessible. We could extend this in the future to perhaps have a web-based playground for writing Bluespec modules.

Thoughts?

quark17, 2021-07-24:
Are you specifically looking for HTML-based documentation, or would a link a pre-built PDF work?  My plan is for this repo to have pre-built releases and PDFs that can be downloaded (using the "Releases" link that's on the right side of the main page).  I'm hoping to make that happen any day now.

I am also open to using Github Pages to host a website for B-Lang-org.  And I'm open to using this to host HTML versions of the docs.  But it would help if the concrete process for doing that could be spelled out more.  (Would you use Github Actions to automatically commit new versions whenever edits are pushed to `main` here?  Or would that be a manual process, and only done at certain times, such as for tagged release points?)

In issue #36, Read the Docs was mentioned as an option for hosting documentation.  I believe that it can generate HTML versions, and it can host multiple versions including up-to-date versions that sync with new commits.  However, again, I would need someone to walk me though the concrete steps for making that happen.  I briefly had a look, and it was not immediately obvious to me how to configure it to work with BSC's documentation.

ThePerfectComputer, 2023-01-11:
I played around with converting the libraries guide via panda some time ago and it didn't go very well...

A lot of the docs are scraped from bluespec source files before being built.
Also, there are a number of latex commands in the docs pandoc doesn't recognize.

There are also multiple sources of truth W.R.T the various documentation PDFs that get built. For example, FIFOs are documented in Bluespec Classic docs, BSV Libs docs, and BSV by example I believe.

If a user wants to learn about FIFOs, where should they go?
These ,multiple sources of truth are usually pretty close, but sometimes, one PDF contains more info than the others.

I've found the bluespec classic documentation not all that helpful, but the BSC *.bs test suite files have been extremely informative.

Anyways to sum up, you could do a quick tex -> html a la pandoc or even with tex4ht, but the result you'll get probably won't be worth using.

The best path is probably a few hundred hours worth of effort(at least this is how long it would take me) consolidating docs and porting them manually to AsciiDoc. 

I know this isn't the most encouraging answer - but it reflects the reality of my to-date efforts trying to bring docs online.

ThePerfectComputer, 2023-02-10:
I've begun to work on a manual port as I get the time, here: https://github.com/ThePerfectComputer/bluespec-docs