B-Lang-org/bsc #379: Bluesim not able to handle BRAMs where the address field is longer than 64 bits

ivanmgribeiro, 2021-07-15:
Hello,
I was trying to use a BRAMCore and came across a compilation error when trying to simulate using bluesim. It appears that bluesim is not able to handle BRAMCores which have an address type that is longer than 63 bits. The following example demonstrates the issue I was having:

```
import BRAMCore :: *;

(* synthesize *)
module mkTestBRAMSize (Empty);
   BRAM_DUAL_PORT #(Bit #(64), Bit #(128)) mybram <- mkBRAMCore2 (1, False);
endmodule
```

It looks like when the address type is less than 64 bits, the generated cxx file for the `mkTestBRAMSize` module uses a literal integer for the memory size, but for address types greater than or equal to 64 bits it instead uses a `tUWide` variable and this is causing issues.

Let me know if there is anything more that I should provide.
Thanks.

quark17, 2021-07-24:
Yes, I confirm that the templated class `MOD_BRAM` (defined in `bs_prim_mod_bram.h` in `src/bluesim/`) only has constructors where the `mem_size` argument is of type `unsigned long long`.  If the size of the BRAM is greater than can be represented in 64 bits, then the instantiation provides the value as type `tUWide`, which doesn't match any of the constructors, so you get an error like this:
```
mkTestBRAMSize.cxx:20:5: error: no matching constructor for initialization of 'MOD_BRAM<tUInt64, tUWide, tUInt8>' (aka 'MOD_BRAM<unsigned long long, WideData, unsigned char>')
    INST_mybram(simHdl, "mybram", this, (tUInt8)0u, 64u, 128u, UWide_literal_65_h1, 2u),
```
followed by several lines listing the candidate constructors and rejecting them as "not viable".

I see that the `mem_size` argument is only ever used to assign the private variable `hi_addr` which is of type `AT`:
```
hi_addr(mem_size-1)
```
where `AT` is the template type for the address type (`tUInt64` for 64-bit, or `tUWide` for larger widths).

I suspect, then, that we can remove the `mem_size` argument and instead just pass in the high address directly -- and this value can be of type `AT`.  That requires changing both the class in `bs_prim_mod_bram.h` and the codegen in BSC, which is handed in `src/comp/SimPrimitiveModules.hs`.  (The translation between import-BVI parameters and the actual C++ class arguments happens in the table `primMap`, where you'll notice that functions like `bramType` inspect the Verilog instantiation args to produce the C++ instantiation.)  So this might be simple to fix.

That said, do you really mean to create a BRAM of this size?  Are there FPGAs that have this much block ram?

ivanmgribeiro, 2021-07-24:
I do not want to create a BRAM with 2^64 words, but at the time I did want to use the address type that the BRAM's parent module had as the index into the BRAM without having to truncate it - I assumed that the size of the BRAM would be determined by the `mem_size` argument, and hence the maximum address size would be determined from that and any input addresses would be truncated to that size.

I don't have a need for this to be fixed (I did end up just truncating the size), I just wanted to bring it up in case anyone in the future tries to use larger address sizes for BRAMs and ends up with that (IMO) quite opaque error.

I think removing the `mem_size` argument would probably be the "right" fix but given how few people would try to instantiate a BRAM with addresses > 63 bits, and that removing the `mem_size` argument would affect any code which instantiates BRAMs now, it might be better to just make the error clearer if possible, or get BSC to either issue a warning or error when the address type is larger than 63 bits.