B-Lang-org/bsc #390: deleted verilog files

neelgala, 2021-08-05:
[This](https://github.com/B-Lang-org/bsc/pull/388/files) PR seems to have deleted some of the verilog files. While most of the changes in the PR were related to removing the ifdefs for icarus, I was wondering if the files like ASSIGN1.v and the gray counter had real issues to be removed completely ?

We had been using the ASSIGN1.v in as a bvi import in our [project](https://gitlab.com/incoresemi/blocks/common_bsv/-/blob/master/clock_divider.bsv#L78-87) which now fails with the latest bsc. Any chance if we could push those back in ?

quark17, 2021-08-05:
BSC doesn't provide Verilog files just to provide useful Verilog files; Verilog files exist where necessary to implement modules in BSC's library packages.  The ASSIGN1 and gray counter modules were not being used by any of BSC's packages, so they were removed.

If you have libraries that import these Verilog files and you think that those libraries deserve to be in BSC, then you're welcome to submit a PR to contribute those libraries (and then the ASSIGN1 and gray counter Verilog files can exist here to implement those libraries).  Or submit the PR to the bsc-contrib repo, if the libraries are better suited to be there than in the bsc repo.

Otherwise, I would suggest that you just keep local copies of these Verilog files in your project, alongside their import-BVIs.

quark17, 2021-08-05:
I see that there is a library in `bsc-contrib` that imports ASSIGN1 and ClockGater.  So I will need to add those two Verilog files back, in that repository.

neelgala, 2021-08-05:
@quark17 Thanks. You can close this now or once the files are pushed.

quark17, 2021-08-22:
I have added the files to `bsc-contrib` in PR B-Lang-org/bsc-contrib#18

closed 2021-08-22 by quark17