B-Lang-org/bsc #391: 'Internal Bluespec Compiler Error" on cover property

danielkasza, 2021-08-09:
I get the error below every time I try to use `cover property`.

Error message:

```
Internal Bluespec Compiler Error:
Please report this failure to the BSC developers, by opening a ticket
in the issue database: https://github.com/B-Lang-org/bsc/issues
The following internal error message should be included in your
correspondence along with any other relevant details:
Bluespec Compiler, version untagged-g5f8fb782 (build 5f8fb782)
CVParserImperative:transAssertStmtPT at "../Example_Programs/Eg02a_HelloWorld/src_BSV/Testbench.bsv", line 4, column 4
```

Minimal test case:

```
module mkTestbench (Empty);
   cover property (True);
endmodule
```

quark17, 2021-08-09:
I want to link this to Issue #326, to keep track of the status of SVA.  I note that the example gives the same error even when the `-passthrough-assertions` flag is provided.

danielkasza, 2021-08-09:
Correct, I should have mentioned that I was using the `-passthrough-assertions` flag.